openwrt/packages #21464: golang: opkg install golang tries to pull ipk but then returns response "killed" and crashes router

akademsubotnik, 2023-06-28:
Maintainer: @jefferyto
Environment: arch: armv7l, model: mikrotik hap2, OpenWrt version: OpenWrt 22.03.5 r20134-5f15225c1e

Description:
I am trying to install go on my openwrt router because it's a dependency for an application I want to install, it seems like it tries to get the ipk but then the next line says killed and my router crashes, slows down and wifi turns off.

jefferyto, 2023-06-29:
Not really sure which device you have - is your router in this list: https://openwrt.org/toh/mikrotik/start

Are you sure you have enough disk space? The Go compiler package is very large (and very slow on low-powered devices).

If you only need to compile a Go program, it may be easier to create a source package for your program, compile the target package using the sdk, then you would only need to install the program and not the Go compiler.

akademsubotnik, 2023-06-29:
Sorry, I said wrong name of router.  My router https://openwrt.org/toh/mikrotik/hap_ac2 .  I install packages on a usb drive that is connected to the router, but maybe weak cpu?

> If you only need to compile a Go program, it may be easier to create a source package for your program, compile the target package using the sdk, then you would only need to install the program and not the Go compiler.

Thank you for the advice, I will give this a try.
~Greg

akademsubotnik, 2023-06-29:
Low resources of router installing on

closed 2023-06-29 by akademsubotnik

jefferyto, 2023-06-30:
> Sorry, I said wrong name of router. My router https://openwrt.org/toh/mikrotik/hap_ac2 . I install packages on a usb drive that is connected to the router, but maybe weak cpu?

My guess is the issue is RAM, I believe opkg downloads the ipk files to /tmp before installing. When I tried this with qemu (with 128MB RAM) opkg exits with a "failed to download" error though, it doesn't crash the image. If you want you could try downloading the ipk files manually (both golang and golang-src), transferring them to the USB drive and installing them from there.

> Thank you for the advice, I will give this a try.

Another alternative is to cross-compile the program(s) on your computer, since every Go compiler is capable of cross-compilation.