openwrt/packages #21508: AdguardHome 0.107.21-1 won't start on reboot

aliekiddo, 2023-07-07:
Maintainer: @dobo90
Environment: (armv7 ?, Linksys WRT3200ACM-UK, OpenWrt 21.02.3 r16554-1d4dea6d4f )

Description:

AdguardHome doesn't start on reboot.
I've had this issue for a while and tried to investigate a fix for it periodically. I am running 0.107.21-1 installed via the software page (I can't see an updates)

I have this in local startup:

```
# Put your custom commands here that should be executed once
# the system init finished. By default this file does nothing.
service adguardhome enable
service adguardhome start
exit 0
```

and I've tried changing /etc/init.d/adguardhome from this:

```
# starts just after network starts to avoid some network race conditions
START=21
# stops before networking stops
STOP=89
```

to this as suggested in other threads

```
# starts just after network starts to avoid some network race conditions
START=99
# stops before networking stops
STOP=120
```

but it still does not start automatically.

brada4, 2023-07-07:
There have been package upgrades in v22 since v21 was retired.
After upgrade please check `logread -e adguard`

brada4, 2023-07-12:
init.d scripts start sutomatically without rc.local
Try to space manual restart minutes from bootup.

aliekiddo, 2023-07-12:
It worked!
I added sleep 60 into the local startup:

> # Put your custom commands here that should be executed once
> # the system init finished. By default this file does nothing.
> sleep 60
> service adguardhome enable
> service adguardhome start
> exit 0

closed 2023-07-12 by aliekiddo

brada4, 2023-07-12:
It should not be necessary ideally.

aliekiddo, 2023-07-12:
Yes, this has been irritating me for 2+ years

brada4, 2023-07-12:
The  numeric start does not mean you have 'net in granma's terms. If the service executable fails it needs to poll 0/0 route idk via ubus or so. Another is ntp time setup - most ddvices come up with rtm time, then get / atime, then do ntp. before that no ssl certs are valid.