openwrt/packages #21519: DHCP server issues IP addresses outside the specified range

bro2020, 2023-07-08:
### Specification
Model: TP-Link Archer C6 v2 (EU/RU/JP)
Architecture: Qualcomm Atheros QCA956X ver 1 rev 0
Target Platform: ath79/generic
Firmware Version: OpenWrt 23.05.0-rc1 r23069-e2701e0f33 / LuCI openwrt-23.05 branch git-23.118.79121-6fb185f
(https://downloads.openwrt.org/releases/23.05.0-rc2/targets/ath79/generic/openwrt-23.05.0-rc2-ath79-generic-tplink_archer-c6-v2-squashfs-sysupgrade.bin)
Kernel Version: 5.15.114

### Playback procedure
Go to interface settings:
http://192.168.1.1/cgi-bin/luci/admin/network/network
Click on `Edit` lan interface:
[image]

In the window that opens, go to the `DHCP Server` tab and set the following values and click on `Save`:
[image]

Now apply the saved settings by clicking on `Save & Apply`
[image]

Go to the overview page:
http://192.168.1.1/cgi-bin/luci/admin/status/overview
And you will see that devices receive IP outside the specified range upwards.
[image]

For purity of playback, you can also reboot the router and the devices that connect to it.
### Expected Behavior
All IPs that devices connected to the router will receive are in the specified range (In this example, 40-90).
### Clarification
Correct information is entered in the router configuration:
```
root@ArcherC6:~# cat /etc/config/dhcp 

config dnsmasq
        option domainneeded '1'
        option localise_queries '1'
        option rebind_protection '1'
        option rebind_localhost '1'
        option local '/lan/'
        option domain 'lan'
        option expandhosts '1'
        option cachesize '1000'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
        option localservice '1'
        option ednspacket_max '1232'

config dhcp 'lan'
        option interface 'lan'
        option start '40'
        option limit '90'
        option leasetime '1h'
        option dhcpv4 'server'
        list dhcp_option '6,192.168.1.1,1.1.1.1'
...
```
and uci
```
root@ArcherC6:~# uci show dhcp
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].cachesize='1000'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].ednspacket_max='1232'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='40'
dhcp.lan.limit='90'
dhcp.lan.leasetime='1h'
dhcp.lan.dhcpv4='server'
dhcp.lan.dhcp_option='6,192.168.1.1,1.1.1.1'
...
```
Exactly the same behavior on the current stable version of OpenWRT 22.03.5
(https://downloads.openwrt.org/releases/22.03.5/targets/ath79/generic/openwrt-22.03.5-ath79-generic-tplink_archer-c6-v2-squashfs-sysupgrade.bin)

stangri, 2023-07-09:
90 is the maximum number of leases, 40+90=130, so the 120 falls within allowed range.

bro2020, 2023-07-09:
Thanks, it seems the pool was set differently before. Now, based on your explanation, everything converges.

closed 2023-07-09 by bro2020