openwrt/packages #21527: nginx: QUIC/http3 doesn't work

SopaDeMacaco-UmaDelicia, 2023-07-10:
Maintainer: Thomas Heil [heil@terminal-consulting.de](mailto:heil@terminal-consulting.de) & Ansuel Smith [ansuelsmth@gmail.com](mailto:ansuelsmth@gmail.com)
Environment: (ipq40xx/generic, ASUS RT-AC58U, SNAPSHOT (r23545-98e3f82c3f) )

Description:
So I was playing a bit with nginx+luci and trying to set it up to use QUIC, but sadly it didn't like my QUIC-enabling lines
```
2023/07/10 05:51:13 [emerg] 3714#0: the "quic" parameter requires ngx_http_v3_module in /etc/nginx/uci.conf:37
```
Looks like quic module was not compiled? Here's `nginx -V`
```
nginx version: nginx/1.25.1 (x86_64-pc-linux-gnu)
built with OpenSSL 3.0.9 30 May 2023
TLS SNI support enabled
configure arguments: --target=arm-openwrt-linux --host=arm-openwrt-linux --build=x86_64-pc-linux-gnu --disable-dependency-tracking --program-prefix= --program-suffix= --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls --crossbuild=Linux::arm --prefix=/usr --conf-path=/etc/nginx/nginx.conf --modules-path=/usr/lib/nginx/modules --error-log-path=stderr --pid-path=/var/run/nginx.pid --lock-path=/var/lock/nginx.lock --http-log-path=/var/log/nginx/access.log --http-client-body-temp-path=/var/lib/nginx/body --http-proxy-temp-path=/var/lib/nginx/proxy --http-fastcgi-temp-path=/var/lib/nginx/fastcgi --with-cc=arm-openwrt-linux-muslgnueabi-gcc --with-cc-opt='-I/builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.3.0_musl_eabi/usr/include -I/builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.3.0_musl_eabi/include/fortify -I/builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.3.0_musl_eabi/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -mfloat-abi=hard -ffile-prefix-map=/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-ssl/nginx-1.25.1=nginx-1.25.1 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNGX_LUA_NO_BY_LUA_BLOCK' --with-ld-opt='-L/builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.3.0_musl_eabi/usr/lib -L/builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.3.0_musl_eabi/lib -fuse-ld=bfd -Wl,--gc-sections -znow -zrelro' --without-http_upstream_zone_module --without-pcre2 --with-compat --with-http_ssl_module --with-ipv6 --with-http_gzip_static_module --with-http_dav_module --with-http_v2_module --with-stream=dynamic --with-stream_ssl_module --with-stream_ssl_preread_module --add-dynamic-module=/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-ssl/nginx-1.25.1/nginx-mod-naxsi/naxsi_src --add-dynamic-module=/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-ssl/nginx-1.25.1/nginx-mod-geoip2 --add-dynamic-module=/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-ssl/nginx-1.25.1/nginx-mod-lua --add-dynamic-module=/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-ssl/nginx-1.25.1/nginx-mod-rtmp --add-dynamic-module=/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-ssl/nginx-1.25.1/nginx-mod-dav-ext --add-dynamic-module=/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-ssl/nginx-1.25.1/nginx-mod-brotli --add-dynamic-module=/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-ssl/nginx-1.25.1/nginx-mod-headers-more --add-dynamic-module=/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-ssl/nginx-1.25.1/nginx-mod-ts --add-dynamic-module=/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-ssl/nginx-1.25.1/nginx-mod-ubus

```
also cc @tiagogaspar8 as original author

Ansuel, 2023-07-10:
Are you using nginx-full?

SopaDeMacaco-UmaDelicia, 2023-07-10:
So I tried nginx-full and was able to enable http3 in the config. Though, I though It should be available in the minimal nginx setup, just like http2.

Ansuel, 2023-07-10:
> So I tried nginx-full and was able to enable http3 in the config. Though, I though It should be available in the minimal nginx setup, just like http2.

no the feature is experimental and increase the package for something that is not stable and should be used only by advanced users.

SopaDeMacaco-UmaDelicia, 2023-07-10:
Ok, thanks for clarifying.

closed 2023-07-10 by SopaDeMacaco-UmaDelicia

sbwml, 2023-08-19:
I use quictls and nginx quic is working fine at the moment.

```
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 23.05.0-rc2, r23228-cd17d8df2a
 -----------------------------------------------------
[root@OpenWrt ~]# nginx -V
nginx version: nginx/1.25.1 (OpenWrt)
built with OpenSSL 3.0.10+quic 1 Aug 2023
TLS SNI support enabled
configure arguments: --target=aarch64-openwrt-linux --host=aarch64-openwrt-linux --build=x86_64-pc-linux-gnu --disable-dependency-tracking --program-prefix= --program-suffix= --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls --crossbuild=Linux::aarch64 --prefix=/usr --build=OpenWrt --conf-path=/etc/nginx/nginx.conf --with-http_ssl_module --with-file-aio --with-threads --with-ipv6 --with-http_stub_status_module --with-http_flv_module --with-http_dav_module --with-http_auth_request_module --with-http_v2_module --with-http_v3_module --with-http_realip_module --with-http_secure_link_module --with-http_sub_module --with-http_random_index_module --with-stream --with-stream_ssl_module --with-stream_ssl_preread_module --with-openssl-opt=enable-ktls --add-module=/home/runner/openwrt/build_dir/target-aarch64_generic_musl/nginx-all-module/nginx-1.25.1/nginx-headers-more --add-module=/home/runner/openwrt/build_dir/target-aarch64_generic_musl/nginx-all-module/nginx-1.25.1/nginx-naxsi/naxsi_src --add-module=/home/runner/openwrt/build_dir/target-aarch64_generic_musl/nginx-all-module/nginx-1.25.1/nginx-dav-ext-module --add-module=/home/runner/openwrt/build_dir/target-aarch64_generic_musl/nginx-all-module/nginx-1.25.1/nginx-brotli --add-module=/home/runner/openwrt/build_dir/target-aarch64_generic_musl/nginx-all-module/nginx-1.25.1/nginx-rtmp --add-module=/home/runner/openwrt/build_dir/target-aarch64_generic_musl/nginx-all-module/nginx-1.25.1/nginx-ts --add-module=/home/runner/openwrt/build_dir/target-aarch64_generic_musl/nginx-all-module/nginx-1.25.1/nginx-ubus-module --error-log-path=stderr --pid-path=/var/run/nginx.pid --lock-path=/var/lock/nginx.lock --http-log-path=/var/log/nginx/access.log --http-client-body-temp-path=/var/lib/nginx/body --http-proxy-temp-path=/var/lib/nginx/proxy --http-fastcgi-temp-path=/var/lib/nginx/fastcgi --with-cc=aarch64-openwrt-linux-musl-gcc --with-cc-opt='-I/home/runner/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/runner/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/runner/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include -O3 -Wl,--gc-sections -pipe -march=armv8-a+crc+crypto -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/openwrt/build_dir/target-aarch64_generic_musl/nginx-all-module/nginx-1.25.1=nginx-1.25.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fvisibility=hidden -ffunction-sections -fdata-sections -DNGX_LUA_NO_BY_LUA_BLOCK' --with-ld-opt='-L/home/runner/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/lib -L/home/runner/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib -znow -zrelro -Wl,--gc-sections' --without-http_upstream_zone_module
```

[image]