openwrt/packages #21530: rust: can't find crate for `core` (when using multiple targets using ./scripts/env)

brocaar, 2023-07-10:
Maintainer: @1715173329 @lu-zero
Environment: Raspberry Pi 4 (`bcm27xx/bcm2709`) + Raspberry Pi Zero W (`bcm27xx/bcm2708`) 

### Summary

I'm creating custom OpenWrt based images for several Raspberry Pi models. To manage the configuration I'm using `./scripts/env` to switch between configuration. The first target (`bcm27xx/bcm2709`) compiled fine. After switching my configuration to the `bcm27xx/bcm2708` target, I got the following error:

```
error[E0463]: can't find crate for `core`
  |
  = note: the `arm-unknown-linux-musleabihf` target may not be installed
  = help: consider downloading the target with `rustup target add arm-unknown-linux-musleabihf`
```

I see that `rustc` is invoked with `--target arm-unknown-linux-musleabihf`.

Looking at the `cargo/config`, I find the following:

```
❯ cat ./staging_dir/hostpkg/cargo/config
[target.armv7-unknown-linux-musleabihf]
linker = "arm-openwrt-linux-muslgnueabi-gcc"
rustflags = ["-Ctarget-feature=-crt-static", "-Clink-args=-lssp_nonshared"]

[profile.stripped]
inherits = "release"
opt-level = "s"
strip = true
```

### Assumption

I think this issue is related to the fact that I'm using `./scripts/env` for working with multiple environments. Compiling my Rust source-code for the first target works fine because this will compile the `rust/host` package. My assumption is that if `rust/host` compiles, that this will include the toolchain for the target that is present in the OpenWrt configuration (in my case `bcm27xx/bcm2709`).

Once I switch my configuration to `bcm27xx/bcm2708`, then this does not invalidate the `rust/host` package and thus the `--target` flag points to a target which is not present in the already compiled `rust/host`, causing the above error.

### Possible solution

Currently the `cargo` binary is installed in:

```
./staging_dir/hostpkg/cargo/bin/cargo
```

Wouldn't it be better to prefix this with the OpenWrt target, like for example the `gcc` binaries:

```
./staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.3.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-gcc
./staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-12.3.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-gcc
```

lu-zero, 2023-07-10:
No, it does not. Ideally you can build just the target and then rustc and cargo will pick what's available. 

You may change the Makefile to build all the targets you care about instead of just one.

brocaar, 2023-07-10:
I was hoping that supporting multiple targets within one build environment would work without any modifications to the Makefile(s).

E.g. I run:

```
# This works fine
./scripts/env switch raspberrypi_bcm27xx_bcm2709
make

# This fails
./scripts/env switch raspberrypi_bcm27xx_bcm2708
make
```

This works fine with packages that are cross-compiled using the GCC toolchain. Is this expected to work too using the Rust compiler too?

In case it is relevant, this is one of my Rust Makefiles:

https://github.com/chirpstack/chirpstack-openwrt-feed/blob/master/chirpstack/chirpstack-udp-forwarder/Makefile

brocaar, 2023-07-10:
I'm now trying to re-compile my Rust source after a:

```
make package/feeds/packages/rust/host/clean
```

Rust is re-compiling and I can see that Rust is now being configured with `arm-unknown-linux-musleabihf` target. However, I assume I will get a missing target issue once I switch back to my `raspberrypi_bcm27xx_bcm2709` configuration, which requires a different Rust target.

```
configure: processing command line
configure: 
configure: build.configure-args := ['--build=x86_64-unknown-linux-gnu', '--target ...
configure: build.build          := x86_64-unknown-linux-gnu
configure: build.target         := ['arm-unknown-linux-musleabihf', 'x86_64-unkno ...
configure: build.host           := ['x86_64-unknown-linux-gnu']
configure: install.prefix       := /workdir/staging_dir/hostpkg/cargo
configure: install.bindir       := /workdir/staging_dir/hostpkg/cargo/bin
configure: install.libdir       := /workdir/staging_dir/hostpkg/cargo/lib
configure: install.sysconfdir   := /workdir/staging_dir/hostpkg/cargo/etc
configure: install.datadir      := /workdir/staging_dir/hostpkg/cargo/share
configure: install.mandir       := /workdir/staging_dir/hostpkg/cargo/man
configure: dist.compression-formats := ['xz']
configure: dist.missing-tools   := True
configure: build.sanitizers     := False
configure: rust.channel         := stable
configure: build.cargo-native-static := True
configure: llvm.download-ci-llvm := True
configure: target.arm-unknown-linux-musleabihf.ar := arm-openwrt-linux-muslgnueab ...
configure: target.arm-unknown-linux-musleabihf.cc := arm-openwrt-linux-muslgnueab ...
configure: target.arm-unknown-linux-musleabihf.cxx := arm-openwrt-linux-muslgnuea ...
configure: target.arm-unknown-linux-musleabihf.linker := arm-openwrt-linux-muslgn ...
configure: target.arm-unknown-linux-musleabihf.ranlib := arm-openwrt-linux-muslgn ...
configure: target.arm-unknown-linux-musleabihf.musl-root := /workdir/staging_dir/ ...
configure: 
configure: writing `config.toml` in current directory
```

lu-zero, 2023-07-10:
Something I suggest to explore is to have clang as C/C++ compiler to reduce further the build time but there are many caveats to tackle...

brocaar, 2023-07-10:
Ok, that is a pitty. Because I was hoping to make this work without patching `packages` repository / Rust Makefile. I assume in order to follow your suggestion, I would need to.

What would happen if I change `CARGO_HOME` (https://github.com/openwrt/packages/blob/master/lang/rust/rust-values.mk):

```
CARGO_HOME:=$(STAGING_DIR_HOSTPKG)/cargo
```

to for example

```
CARGO_HOME:=$(STAGING_DIR_HOSTPKG)/cargo-$(RUSTC_HOST_ARCH)
```

Wouldn't that cause `make` to automatically compile Rust if it hasn't been compiled for that target?

If that works, would you be against such change? Because the way Rust / Cargo host package is being built is that it is executable on the host architecture, but with a specific target (https://github.com/openwrt/packages/blob/master/lang/rust/Makefile#L60). This is different than for example the `protobuf/host` package which is target agnostic.

I'm happy to do some testing at my end to see if this would solve my issue.

lu-zero, 2023-07-10:
I mean, pass to `--target` `arm-unknown-linux-musleabihf,armv7-unknown-linux-musleabihf` or as many targets you want to support and you are done with that. You have to do the same for llvm.

brocaar, 2023-07-10:
Just to make sure I understand you correctly, you mean overwriting:

`--target=$(RUSTC_TARGET_ARCH),$(RUSTC_HOST_ARCH)` in:
https://github.com/openwrt/packages/blob/5362b272cbe6124519616052bf61a6faee60cd2b/lang/rust/Makefile#L60C13-L60C30

to include all additional targets that must be included?

If that is the case, it would mean I need to hardcode the `TARGET_` values, as these are not known at compile time for all the targets that I would add by hand?:

```
TARGET_CONFIGURE_ARGS = \
	--set=target.$(RUSTC_TARGET_ARCH).ar=$(TARGET_AR) \
	--set=target.$(RUSTC_TARGET_ARCH).cc=$(TARGET_CC_NOCACHE) \
	--set=target.$(RUSTC_TARGET_ARCH).cxx=$(TARGET_CXX_NOCACHE) \
	--set=target.$(RUSTC_TARGET_ARCH).linker=$(TARGET_CC_NOCACHE) \
	--set=target.$(RUSTC_TARGET_ARCH).ranlib=$(TARGET_RANLIB) \
	$(if $(CONFIG_USE_MUSL),--set=target.$(RUSTC_TARGET_ARCH).musl-root=$(TOOLCHAIN_DIR))
```

https://github.com/openwrt/packages/blob/5362b272cbe6124519616052bf61a6faee60cd2b/lang/rust/Makefile#L46C1-L46C22

If I'm mis-understanding, could you give a code example?

lu-zero, 2023-07-10:
You need to:
- set `--target=$(RUSTC_TARGET_ARCH),$(RUSTC_HOST_ARCH),$(RUSTC_TARGET_ARCH_OTHER)`
- make 2 sets of `TARGET_CONFIGURE_ARGS` for both of the targets you care about and use both.

Alternatively we'd have to split building the host toolchain and the targets so the env trick could work (maybe).

brocaar, 2023-07-10:
Thanks. However, I'm afraid that this might become hard to maintain. The `TARGET_` variable values are all computed and the values held by these variables are only valid for the selected OpenWrt target. This will be very prone to errors. I'm currently thinking of 4 different OpenWrt targets for my project, and this number will probably increase over time.

As well I believe the current packages in this repository using Rust are affected by this issue too. This means that switching configuration using `./scripts/env` (or changing target configuration using `make menuconfig`) will break for these packages too if selected using `make menuconfig` if I'm not mistaken. I don't think patching the `rust/Makefile` should be a requirement for compiling to multiple OpenWrt targets within the same build context.

I've been making some changes, and this seems to work. Before I create a PR I want to do a bit more testing, but I'm looking forward to your feedback in general:

https://github.com/openwrt/packages/compare/master...brocaar:packages:fix_issue_21530

lu-zero, 2023-07-10:
I wonder how golang packages solve the same problem. if you want to do the additional legwork you can split the targets in a separate package that way you can depend on the target + the host compiler and everything will work for your use case.

brocaar, 2023-07-11:
The difference with Go is that all targets are included in the Go toolchain
by default. Once compiled, you can cross compile to all targets.

brocaar, 2023-07-11:
> you can split the targets in a separate package that way you can depend on the target + the host compiler and everything will work for your use case.

I think the cleanest would be when this works out-of-the box, without the need to package each target separately. This way it works for all packages that are using Rust, including the packages included in this repository. The `./scripts/env` is a nice tool to work with multiple configurations (e.g. if working with multiple Raspberry Pi models) within the same build context and it would be a pitty if Rust-based packages would prevent users from using this tool.

Maybe this will work:

```
Rust based package
    -> rust/host
        -> rust-target/host
```

Then `rust/host` would only build for the host architecture, and `rust-target/host` adds the target (based on current OpenWrt config). Changing the OpenWrt config should then trigger a new `rust-target/host` build to add this target.

https://rustc-dev-guide.rust-lang.org/building/how-to-build-and-run.html#building-targets-for-cross-compilation

But, I think this will require quite some changes, because then you want to have all Rust code (source & binaries) under the same path, but you need to trick OpenWrt to rebuild `rust-target/host` if the OpenWrt target has changed, and the Rust target is not yet available.

As well this will require a per target Cargo configuration to make sure that the build flags are passed and the correct linker is used.

I think the above proposed change is a lot less disruptive as it doesn't change the way Rust is compiled, it only changes the paths such that OpenWrt will rebuild (under a new path) if the target has changed. As well it requires no changes from the package Makefiles.

commit f489e019 (2023-07-17): rust: compile host package per target

While the compiled binaries are intended to run on the host system, the
rust/host package does include the target matching the configured
OpenWrt target.

If using (for example) ./scripts/env to switch between different
OpenWrt configurations, this will cause issues if the different
configuration is for a different target. In such case there will be a
mismatch between the available Rust target and OpenWrt target and the
following error will be printed:

> error[E0463]: can't find crate for `core`
> note: the `XXX` target may not be installed

This fix wil [... cut]
commit 26de69a8 (2023-07-18): rust: compile host package per target

While the compiled binaries are intended to run on the host system, the
rust/host package does include the target matching the configured
OpenWrt target.

If using (for example) ./scripts/env to switch between different
OpenWrt configurations, this will cause issues if the different
configuration is for a different target. In such case there will be a
mismatch between the available Rust target and OpenWrt target and the
following error will be printed:

> error[E0463]: can't find crate for `core`
> note: the `XXX` target may not be installed

This fix wil [... cut]
closed 2023-07-17 by 1715173329