openwrt/packages #21535: simple-adblock: Any interruption in WAN device causes simple-adblock to crash

VA1DER, 2023-07-10:
Maintainer: @stangri 

Environment: OpenWrt Master on Banana Pi BPI-R64, BPI-R2, and BPI-R3.  Suspect problem manifests on all devices.

Description:  When simple-adblock is running on a device in standard home router confiuration (WAN <-> LAN), then any interruption in the WAN device causes simple-adblock to silently fail.

LuCI will show that it is not running, but attempting to start it through LuCI will fail.  It has to be restarted through `/etc/init.d/simple-adblock restart`

It will happen if the WAN device comes down for any reason.  It is especially troublesome for situations where the WAN is provided by WiFi and the upstream provider, for example, changes channels or if the client has to reconnect for any reason.

brada4, 2023-07-11:
Does it work you edit hotplug script to "restart"? The "hotplug" parameter there is completely noop for the script.
https://github.com/openwrt/packages/blob/7709b85f86f7ce2da5a86f2d5d1fcdd77c5089a1/net/simple-adblock/files/simple-adblock.hotplug#L4

stangri, 2023-07-14:
@VA1DER Thank you for the report. 

Can you provide any logs/tests? Like logread for simple-adblock/dnsmasq after the `ifdown wan; ifup wan;` commands? Coupled with the nslookup requests for domains in the block-list (dnsmasq should report last dozen or so in the log) before and after WAN interruption?

stangri, 2023-08-30:
Closing due to lack of replies.

closed 2023-08-30 by stangri