openwrt/packages #21548: coreutils-who: not returning any useful output

psmode, 2023-07-12:
Maintainer: @httpstorm 
Environment: (x86, 64, OpenWrt v 22.03.05)

Description:
coreutils-who v9.0-2 appears to be non-functional in my environment. Executing the command who by itself or with "-a" returns no output at all. Adding -H or -Hl will return the column headings, but no data beyond that. -q returns a blank line, then "# users=0"

httpstorm, 2023-07-26:
@psmode 
I reproduced the behaviour you described on the following versions of `coreutils-who`: `9.3-1` `9.2-1` `9.1-1` `9.0-2` `9.0-1` `8.32-6`. I am not able to go back in time before version `8.32-6` using simple `git revert`.
Platforms tested: x64 and WRT3200ACM. OpenWrt SNAPSHOT r22888-86adc89390.

`coreutils-who-9.3` on macOS Ventura 13.4.1 (c) (22F770820d) works fine, except for `-Hl` which prints only the title row.
`coreutils-who-8.32` on Ubuntu 22.04.2 LTS kernel 5.15.0-76 works fine.

Did any previous version of `coreutils-who` work in the past on OpenWRT?
Try contacting the `coreutils` team. `coreutils at gnu . org`

psmode, 2023-07-26:
Unfortunately, I cannot say which, if any, prior versions worked.

I will reach out to the team via email referencing this thread

httpstorm, 2023-07-26:
Just a thought, might be completely wrong, but I wonder whether the kernel we build offers all necessary features for `who` to work? One more thing to consider: the exit code is 0 (success):

```
who && echo success || echo fail
success
```

psmode, 2023-09-11:
I believe that the problem is not so much with who, but rather with the fact that I do not have a `/var/run/utmp` file being populated for who to read. This seems to be an issue on a number of platforms.

So now the question becomes, it it possible/desirable to have the `/var/run/utmp` populated by the kernel for who to read?

httpstorm, 2023-09-13:
I tried copying `utmp` from Ubuntu and macOS, makes no difference. Though I believe that file should be created by the system or kernel to represent the currently logged on users. And since the Linux kernel in OpenWRT is very tiny, you would need to find and add the proper components to make it work. Hence rebuild should be needed.

M95D, 2024-05-18:
This is a known issue (or feature, depending on your POV).
`w`, `who` and `users` take data from /var/run/utmp. That file is updated using functions from libc (musl or glibc). Musl doesn't support utmp (read their FAQ). Even if built with glibc instead of musl, Dropbear and Busybox `login`, `getty` and `tftpd` don't write utmp data in the default OpenWrt configuration.
Details here: https://github.com/openwrt/openwrt/pull/15417