openwrt/packages #21569: lspci cannot load libkmod resources

graysky2, 2023-07-15:
### Describe the bug

Running `lspci -v >/dev/null` gives partially correct output due to an inability to use libkmod.
```
# lspci -v >/dev/null
lspci: Unable to load libkmod resources: error -2
```

### OpenWrt version

r23228-cd17d8df2a

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi 4 Model B Rev 1.4

### Image kind

Official downloaded image

### Steps to reproduce

Run this command:
```
lspci -v > /dev/null
```

### Actual behaviour

Running the above will print the following error and leave various generics in the actual output of the command whereas running the same command on another distro populates various generic statements about the hardware into vendor-specific data
```
lspci: Unable to load libkmod resources: error -2
```

Example of missing data:
On OpenWrt:
```
# lspci -v
...
03:00.0 Ethernet controller: Intel Corporation Device 125c (rev 04)
  Subsystem: Intel Corporation Device 0000
...
```

On Arch:
```
# lspci -v
...
03:00.0 Ethernet controller: Intel Corporation Ethernet Controller I226-V (rev 04)
  Subsystem: Intel Corporation Ethernet Controller I226-V
...
```

### Expected behaviour

Expectation is for libkmod to work properly with `lspci` and to detect hardware.

### Additional info

I have confirmed this bug on two different machines (`x86/64` machine) and the rpi4 listed above running the official image for 23.05.0-rc2 and a very recent snapshot.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

trippleflux, 2023-07-16:
Me too, somehow in recent OpenWrt version, some of the kernel module failed to load properly?.

Some of the kernel modules was segmentation fault when testing it out with modinfo.

I am also experiencing [this](https://github.com/openwrt/mt76/issues/799) error, which never ever happened before.
Currently I have spent days and hours to track it down and haven't figure it out yet. 

I have tried to negate whether it's my global OpenWrt compilation CFLAGS ("-O3" into "-O2") that is causing trouble or something else upstream.

Quentin-M, 2023-07-29:
Might be related to https://github.com/openwrt/openwrt/issues/13181

scr4tchy, 2023-07-29:
Might be related to https://github.com/openwrt/openwrt/issues/13181