openwrt/packages #21579: tang: /usr/sbin/tangd fails with "Illegal instruction" in openwrt 23.05

lespinasse, 2023-07-15:
Maintainer: @Tiboris @neheb @cotequeiroz

Environment: Openwrt 23.05.0-rc2 running on a raspberry pi zero W

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

I have been setting up a couple tang instances on raspberry pi zero W hardware. With openwrt 22.03, tang seems to work fine.

I wanted to try with openwrt 23.05 (I was interested in the standalone feature from tang 12), but I have not been able to get tang going there.

After setting `option enabled '1'` in /etc/config/tang, the `/etc/init.d/tang start` script fails silently (the tang port is not opened for listening).

Running `/usr/sbin/tangd -p 9090 -l /usr/share/tang/db` manually, I see:

```
# /usr/sbin/tangd -p 9090 -l /usr/share/tang/db
Listening on 0.0.0.0:9090
Listening on [::]:9090
Illegal instruction
#
```

lespinasse, 2023-07-16:
Looking into it a little longer, I was able to get tang to work on openwrt 23.05 by running it with xinetd rather than standalone. So, the issue seems to be with the standalone code that was added in tang version 12. Wild guess, maybe it might be the issue that was fixed by https://github.com/latchset/tang/commit/9b9d14973aa1e5b2290595a6b53e67c2f4e0369f in tang version 13.

neheb, 2023-07-16:
Linked issue sounds like it's for 32-bit.

brada4, 2023-07-16:
ccan you extract relevant lines from logread/dmesg? It should jave backtrace and pinpoint to instruction in a source file.

lespinasse, 2023-07-16:
@brada4 logread / dmesg do not show anything about the crash. Also I don't think java should be involved as tang is written in C ?

@neheb That makes sense, pi zero W is a 32 bit platform.

lespinasse, 2023-07-16:
@neheb I can confirm tangd works fine in a standalone setup on pi 4B, which is a 64 bit platform. So I guess that confirms the theory about the issue being exposed on 32 bit systems only.

samcday, 2023-08-29:
I just ran into this issue with my AVM FRITZ!Box 4040 (ipq4xxx / armv7l) when upgrading from 22.03.3 to 22.03.5, so this doesn't just affect 23.05 RCs, but stable 22.03 releases also.

nmav, 2023-08-29:
I've updated 22.03 to v14. Please try this one whether it fixes the issue.

samcday, 2023-08-30:
`opkg` isn't showing any updates available for the tang package from 22.03 repositories, yet.

I noticed you fixed some hashes a couple of hours ago - how long is it expected to take for packaging updates to make their way to the repos?

samcday, 2023-08-31:
tang `14-8` is available in 22.03.5 package repos and appears to fix this issue (for me on `ipq40xx`/`armv7l` platform, at least). Thanks @nmav!

nmav, 2023-08-31:
Thank you for confirming. I'm closing now this.

closed 2023-08-31 by nmav