openwrt/packages #21626: nginx: dynamic modules are not loaded

TheB1gG, 2023-07-22:
Maintainer: @Ansuel (find it by checking history of the package Makefile)
Environment: target x86_64 master

Description:

I want to build nginx with module brotli and headers more, but even though it build successfull the modules not usable

Choosen nginx pakages in .config
```
CONFIG_PACKAGE_nginx-util=y
CONFIG_PACKAGE_nginx-full=y
CONFIG_PACKAGE_nginx-mod-brotli=y
CONFIG_PACKAGE_nginx-mod-headers-more=y
CONFIG_PACKAGE_nginx-ssl-util=y
```
build command: make -j 20 clean defconfig download world
```
nginx -V
root@OpenWrt:~# nginx -V
nginx version: nginx/1.25.1 (x86_64-pc-linux-gnu)
built with OpenSSL 3.0.9 30 May 2023
TLS SNI support enabled
configure arguments: --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --disable-dependency-tracking --program-prefix= --program-suffix= --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls --crossbuild=Linux::x86_64 --prefix=/usr --conf-path=/etc/nginx/nginx.conf --modules-path=/usr/lib/nginx/modules --error-log-path=stderr --pid-path=/var/run/nginx.pid --lock-path=/var/lock/nginx.lock --http-log-path=/var/log/nginx/access.log --http-client-body-temp-path=/var/lib/nginx/body --http-proxy-temp-path=/var/lib/nginx/proxy --http-fastcgi-temp-path=/var/lib/nginx/fastcgi --with-cc=x86_64-openwrt-linux-musl-gcc --with-cc-opt='-I/home/user/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/usr/include -I/home/user/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/include/fortify -I/home/user/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/user/openwrt/build_dir/target-x86_64_musl/nginx-full/nginx-1.25.1=nginx-1.25.1 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNGX_LUA_NO_BY_LUA_BLOCK' --with-ld-opt='-L/home/user/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/usr/lib -L/home/user/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib -fuse-ld=bfd -Wl,--gc-sections -znow -zrelro' --without-http_upstream_zone_module --without-pcre2 --with-compat --with-http_ssl_module --with-ipv6 --with-http_gzip_static_module --with-http_stub_status_module --with-http_flv_module --with-http_dav_module --with-http_auth_request_module --with-http_v3_module --with-http_v2_module --with-http_realip_module --with-http_secure_link_module --with-http_sub_module --add-dynamic-module=/home/user/openwrt/build_dir/target-x86_64_musl/nginx-full/nginx-1.25.1/nginx-mod-brotli --add-dynamic-module=/home/user/openwrt/build_dir/target-x86_64_musl/nginx-full/nginx-1.25.1/nginx-mod-headers-more```
If I want to use more headers or brotli i get
```
nginx -t
[emerg] 25848#0: unknown directive "more_clear_headers" in /etc/nginx/nginx.conf:22
[emerg] 26154#0: unknown directive "brotli" in /etc/nginx/nginx.conf:89
```

Ansuel, 2023-07-22:
You need to actually load the module. You need to install the extra package and you can follow nginx documentation on how to load dynamic modules

TheB1gG, 2023-07-22:
That was my bad I'm sorry, I got to used to debian handling it with symlinks for me without having to do that manually

closed 2023-07-22 by TheB1gG

Ansuel, 2023-07-22:
Can you give me some hint on how it's done there? The idea was to mimic
similar implementation instead of reinventing the wheel

TheB1gG, 2023-07-22:
They have a
```
include /etc/nginx/modules-enabled/*.conf;
```
in their nginx.conf
and if you install a module via apt you get a symlink placed in /etc/nginx/modules-enabled
eg 50-mod-http-headers-more-filter.conf
```
readlink -f /etc/nginx/modules-enabled/50-mod-http-headers-more-filter.conf
/usr/share/nginx/modules-available/mod-http-headers-more-filter.conf
```
content of it
```
load_module modules/ngx_http_headers_more_filter_module.so;
```