openwrt/packages #21685: miniupnpd: Required key not available - flood every 30 seconds

systemcrash, 2023-07-30:
```logs
Sun Jul 30 00:51:13 2023 daemon.err miniupnpd[14703]: sendto(udp_notify=8, 192.168.1.1): Required key not available
Sun Jul 30 00:51:13 2023 daemon.err miniupnpd[14703]: sendto(udp_notify=8, 192.168.1.1): Required key not available
Sun Jul 30 00:51:13 2023 daemon.err miniupnpd[14703]: sendto(udp_notify=8, 192.168.1.1): Required key not available
Sun Jul 30 00:51:13 2023 daemon.err miniupnpd[14703]: sendto(udp_notify=8, 192.168.1.1): Required key not available
Sun Jul 30 00:51:13 2023 daemon.err miniupnpd[14703]: sendto(udp_notify=8, 192.168.1.1): Required key not available
Sun Jul 30 00:51:13 2023 daemon.err miniupnpd[14703]: sendto(udp_notify=8, 192.168.1.1): Required key not available
Sun Jul 30 00:51:13 2023 daemon.err miniupnpd[14703]: sendto(udp_notify=8, 192.168.1.1): Required key not available
Sun Jul 30 00:51:13 2023 daemon.err miniupnpd[14703]: sendto(udp_notify=8, 192.168.1.1): Required key not available
Sun Jul 30 00:51:13 2023 daemon.err miniupnpd[14703]: sendto(udp_notify=8, 192.168.1.1): Required key not available
Sun Jul 30 00:51:13 2023 daemon.err miniupnpd[14703]: sendto(udp_notify=8, 192.168.1.1): Required key not available
Sun Jul 30 00:51:13 2023 daemon.err miniupnpd[14703]: sendto(udp_notify=8, 192.168.1.1): Required key not available
Sun Jul 30 00:51:13 2023 daemon.err miniupnpd[14703]: sendto(udp_notify=8, 192.168.1.1): Required key not available
Sun Jul 30 00:51:13 2023 daemon.err miniupnpd[14703]: sendto(udp_notify=8, 192.168.1.1): Required key not available
Sun Jul 30 00:51:14 2023 daemon.err miniupnpd[14703]: try_sendto(sock=8, len=467, dest=239.255.255.250:1900): sendto: Required key not available
Sun Jul 30 00:51:14 2023 daemon.err miniupnpd[14703]: try_sendto(sock=8, len=467, dest=239.255.255.250:1900): sendto: Required key not available
Sun Jul 30 00:51:14 2023 daemon.err miniupnpd[14703]: try_sendto(sock=8, len=483, dest=239.255.255.250:1900): sendto: Required key not available
Sun Jul 30 00:51:14 2023 daemon.err miniupnpd[14703]: try_sendto(sock=8, len=479, dest=239.255.255.250:1900): sendto: Required key not available
Sun Jul 30 00:51:14 2023 daemon.err miniupnpd[14703]: try_sendto(sock=8, len=467, dest=239.255.255.250:1900): sendto: Required key not available
Sun Jul 30 00:51:14 2023 daemon.err miniupnpd[14703]: try_sendto(sock=8, len=465, dest=239.255.255.250:1900): sendto: Required key not available
Sun Jul 30 00:51:14 2023 daemon.err miniupnpd[14703]: try_sendto(sock=8, len=412, dest=239.255.255.250:1900): sendto: Required key not available
Sun Jul 30 00:51:14 2023 daemon.err miniupnpd[14703]: try_sendto(sock=8, len=451, dest=239.255.255.250:1900): sendto: Required key not available
Sun Jul 30 00:51:14 2023 daemon.err miniupnpd[14703]: try_sendto(sock=8, len=412, dest=239.255.255.250:1900): sendto: Required key not available
Sun Jul 30 00:51:14 2023 daemon.err miniupnpd[14703]: try_sendto(sock=8, len=471, dest=239.255.255.250:1900): sendto: Required key not available
Sun Jul 30 00:51:14 2023 daemon.err miniupnpd[14703]: try_sendto(sock=8, len=412, dest=239.255.255.250:1900): sendto: Required key not available
Sun Jul 30 00:51:14 2023 daemon.err miniupnpd[14703]: try_sendto(sock=8, len=475, dest=239.255.255.250:1900): sendto: Required key not available
Sun Jul 30 00:51:14 2023 daemon.err miniupnpd[14703]: try_sendto(sock=8, len=403, dest=239.255.255.250:1900): sendto: Required key not available
Sun Jul 30 00:51:14 2023 daemon.err miniupnpd[14703]: try_sendto failed to send 13 packets
```

# Issue template (remove lines from top till here)

Maintainer: @neheb / @stintel (?) 
Environment: 

--
Model | AVM FRITZ!Box 7530
Architecture | ARMv7 Processor rev 5 (v7l)
Target Platform | ipq40xx/generic
Firmware Version | OpenWrt 22.03.5 r20134-5f15225c1e / LuCI openwrt-22.03 branch git-23.093.57104-ce20b4a
Kernel Version | 5.10.176

config is factory default

```config

config upnpd 'config'
	option download '1024'
	option upload '512'
	option internal_iface 'lan'
	option port '5000'
	option upnp_lease_file '/var/run/miniupnpd.leases'
	option igdv1 '1'
	option enabled '1'
	option uuid 'b3eb3b7d-2471-477d-a3e6-a8f3993bf00f'

config perm_rule
	option action 'allow'
	option ext_ports '1024-65535'
	option int_addr '0.0.0.0/0'
	option int_ports '1024-65535'
	option comment 'Allow high ports'

config perm_rule
	option action 'deny'
	option ext_ports '0-65535'
	option int_addr '0.0.0.0/0'
	option int_ports '0-65535'
	option comment 'Default deny'
```

About 24 hours after factory config - haven't tweaked much. Added miniupnpd, wg, dyndns.

systemcrash, 2023-08-01:
Another hint:
```
Tue Aug  1 12:10:51 2023 daemon.err miniupnpd[24162]: PCPSendUnsolicitedAnnounce(sockets[0]) sendto(): Required key not available
```

systemcrash, 2023-08-01:
Apparently, miniupnpd becomes extremely confused and never quite starts up, if a wireguard interface (legal in wg config) shares the same IP as the (e.g. br-lan) LAN interface IP. 

I had 
lan: 192.168.1.1
wg: 192.168.1.1

miniupnpd: waaaah

lan: 192.168.1.1
wg: 192.168.1.2

miniupnpd:

I will now start

systemcrash, 2023-11-24:
So apparently the above errors are more likely to occur if network services restart, and miniupnpd does not also restart. The daemon probably needs to know when network restarts and sync with that to restart or reload itself.

ChernyaevAN, 2024-01-21:
Try to add in the config file of the upnpd this line:
`option external_iface 'wan'`
It usually helps. The problem is that upnpd mixes different interfaces and try to work on the WG tunnel.

systemcrash, 2024-01-22:
Thanks @ChernyaevAN I'll give that a try. Would having an `option ignore_iface` be helpful?

I don't see one in the wiki https://openwrt.org/docs/guide-user/firewall/upnp/miniupnpd

ChernyaevAN, 2024-01-23:
Based on [this](https://openwrt.org/docs/guide-user/firewall/upnp/miniupnpd) page I do not see option `ignore_iface` at all.
I'm not sure this option is applicable to miniupnpd.

systemcrash, 2024-01-23:
Mt question was whether such an option should exist 😅

> Based on this
> <https://openwrt.org/docs/guide-user/firewall/upnp/miniupnpd> page I do
> not see option ignore_iface at all.
> I'm not sure this option is applicable to miniupnpd.
>

closed 2025-12-15 by GeorgeSapkin