openwrt/packages #21686: udp-broadcast-relay-redux-openwrt service does not start

TheTrueRandom, 2023-07-31:
The service udp-broadcast-relay-redux-openwrt does not start besides it says it is running, it does not show up in ps. Interestingly the service starts and works as expected when using `service udp-broadcast-relay-redux trace`.

```
root@OpenWrt:~# service udp-broadcast-relay-redux status
running
root@OpenWrt:~# ps | grep udp 
<nothing>
root@OpenWrt:~# service udp-broadcast-relay-redux restart
root@OpenWrt:~# service udp-broadcast-relay-redux status
running
root@OpenWrt:~# ps | grep udp
<nothing>
root@OpenWrt:~# service udp-broadcast-relay-redux trace
root@OpenWrt:~# ps | grep udp
 5630 root       912 S    /sbin/utrace /usr/sbin/udp-broadcast-relay-redux --id 1 --port 6524 ...
 5632 root       776 S    /usr/sbin/udp-broadcast-relay-redux --id 1 --port 6524 ...
```

Version: OpenWrt 22.03.5 r20134-5f15225c1e / LuCI openwrt-22.03 branch git-23.093.57104-ce20b4a

Anthropohedron, 2023-08-19:
I am seeing the same behavior.

schreibikus, 2023-09-07:
I just removed this line:
procd_add_jail ubr-${PIDCOUNT}
from /etc/init.d/udp-broadcast-relay-redux and everything worked!
Enjoy it! Have a nice day!

chhaugen, 2023-10-05:
> I just removed this line: procd_add_jail ubr-${PIDCOUNT} from /etc/init.d/udp-broadcast-relay-redux and everything worked! Enjoy it! Have a nice day!

Your answer saved our LANparty! (lanmine.no)
Thanks!

slusarz, 2023-10-21:
This seems to be fixed by https://github.com/openwrt/packages/pull/20601

closed 2025-11-15 by GeorgeSapkin