openwrt/packages #21700: rust: add --locked argument to cargo install?

brocaar, 2023-08-02:
Maintainer: @lu-zero 

### Description

I just noticed that one of my packages suddenly (no changes) failed to compile because of a failing dependency, while it would compile successfully before. Looking at the dependency that was being compiled and my `Cargo.lock` file I noticed these were not the same which seemed odd to me as it seemed `cargo` was ignoring the `Cargo.lock` file of my project.

### Cause

Currently Rust code is compiled using the following Makefile rule:

```
# $(1) path to the package (optional)
# $(2) additional arguments to cargo (optional)
define Build/Compile/Cargo
	( \
		cd $(PKG_BUILD_DIR) ; \
		export PATH="$(CARGO_HOME)/bin:$(PATH)" ; \
		CARGO_HOME=$(CARGO_HOME) \
		TARGET_CFLAGS="$(TARGET_CFLAGS) $(RUST_CFLAGS)" \
		TARGET_CC=$(TARGET_CC_NOCACHE) \
		CC=$(HOSTCC_NOCACHE) \
		$(CARGO_VARS) \
		cargo install -v \
			--profile stripped \
			--target $(RUSTC_TARGET_ARCH) \
			$(if $(strip $(RUST_PKG_FEATURES)),--features "$(strip $(RUST_PKG_FEATURES))") \
			--root $(PKG_INSTALL_DIR) \
			--path "$(if $(strip $(1)),$(strip $(1)),.)" \
			$(2) ; \
	)
endef
```

The important thing is that `cargo install` is used and not `cargo build`. The issue I was experiencing is cause by the fact that `cargo install` ignores the lockfile by default:

> By default, the Cargo.lock file that is included with the package will be ignored. This means that Cargo will recompute which versions of dependencies to use, possibly using newer versions that have been released since the package was published.

https://doc.rust-lang.org/cargo/commands/cargo-install.html#dealing-with-the-lockfile

### Proposal

I would propose adding the `--locked` flag by default. While for local installations it might be beneficial to pull the latest and greatest dependencies, within the context of OpenWrt I think reproducibility is more important. A tagged release of a Rust project might show different behavior because between two builds there were new releases of its dependencies. As well the choice for using `cargo install` is an implementation detail that users are not aware of, unless they start diving deeper in the Makefile code.

---

Please let me know what you think. I can make a PR for this in case you think this is a good thing to change.

lu-zero, 2023-08-02:
Adding as an option might be better since there are times in which the Cargo.lock needs an update. 

The best would be to keep a list of dependencies and fetch them and then built the crate using them, if you have time to try.

brocaar, 2023-08-03:
> The best would be to keep a list of dependencies and fetch them and then built the crate using them, if you have time to try.

Could you further elaborate as I'm not fully understand what you mean. Isn't this exactly what the `Cargo.lock` file is doing? It keeps a list of dependencies and exact versions that must be fetched compiled and linked against on build.

The "problem" I am experiencing is that `cargo install` vs `cargo build` have different behavior with regards to respecting the `Cargo.lock` file:

* `cargo build` builds the crate using the exact dependency versions as specified in the `Cargo.lock` file
* `cargo install` build the crate using the latest dependency versions as specified in the `Cargo.toml` file and ignores the `Cargo.lock` file. This is documented behavior.

Using `cargo build` you will therefore always get a reproducible build. Using `cargo install` you might get different results between two builds (without any change to package `Makefile`!).

E.g. my `Cargo.lock` files specifies:

https://github.com/chirpstack/chirpstack-concentratord/blob/master/Cargo.lock#L1196

```toml
[[package]]
name = "rustix"
version = "0.37.20"
...
```

But during the `make packages/..../chirpstack-concentratord/{clean,compile} -j1 V=s` I noticed it was suddenly building rustix `0.38.5`. This build failed because there was an issue with that version. Shortly after the release of rustix `0.38.5`, this version was revoked from Crates.io (see: https://crates.io/crates/rustix/0.38.5) and  `make packages/..../chirpstack-concentratord/{clean,compile} -j1 V=s`  was now building against rustix `0.38.4` which did not fail.

Adding the `--locked` argument to `cargo install` should change this behavior (`cargo install` will then use the `Cargo.lock` file).

In my opinion a build system should always respect the `Cargo.lock` file. If at some point I want to update my dependencies, I should (within my project) run `cargo update`, commit the changed `Cargo.lock` file, tag a new release and then update the OpenWrt `Makefile` for this package.

lu-zero, 2023-08-03:
> Could you further elaborate as I'm not fully understand what you mean. Isn't this exactly what the `Cargo.lock` file is doing? It keeps a list of dependencies and exact versions that must be fetched [...]

See below :)

> The "problem" I am experiencing is that `cargo install` vs `cargo build` have different behavior with regards to respecting the `Cargo.lock` file:
>
>     * `cargo build` builds the crate using the [...]

that's the default behaviour, explicit `--locked`, `--frozen` or even the use of `cargo vendor` work for both commands.

> Using `cargo build` you will therefore always get a reproducible build. Using `cargo install` you might get different results between two builds (without any change to package `Makefile`!).

In general you do want to not build (and distribute) something with dependencies with known bugs, alas sometimes the bugfix release may carry different bugs.
 
> In my opinion a build system should always respect the `Cargo.lock` file. If at some point I want to update my dependencies, I should (within my project) run `cargo update`, commit the changed `Cargo.lock` file, tag a new release and then update the OpenWrt `Makefile` for this package.

There are different opinions on pinning dependencies, in general as long it is easy to update them it is a good idea, Cargo.lock sadly does not work well enough.
 
The alternative is to build a local registry with the versions known to be good and use it via .cargo/config see [here](https://github.com/gentoo/gentoo/blob/master/eclass/cargo.eclass#L329) for an example. `cargo ebuild` could be repurposed using a template that produces a Makefile compatible with OpenWrt.

If you have enough time we could try to come up with a full solution together.

brocaar, 2023-08-03:
> Cargo.lock sadly does not work well enough.

Could you further clarify why it does not work well? Is this from the perspective of an author, or packager? In my case (as author of a piece of software), I want to be in full control of what I ship. If I release version 1.2.3 which contains library x, then I want to know if 1.2.3 was shipped with or without a bug in x. If a customer reports a bug with 1.2.3 which might be related to dependency x, I want to be able to reproduce this error. Now this is impossible.

If you look from it from the perspective of somebody packaging an existing project, then I understand your point since if x has been fixed, but the software author has not update x in its Cargo.lock file then with `--locked` you can not include the fixed x library. The only option then would be creating a patch for `Cargo.lock`, which actually might work.

Note that `cargo ebuild` and `cargo bitbake` (which I have used in the past with Yocto) do use the `Cargo.lock` file versions when creating an `.ebuild` and `.bb` template. Updating these files by hand (e.g. to include a fixed version of x) can be tricky as by updating x, you might also need to update other dependencies (the updated x version might include new or update dependencies).

---

I think even if we add `--locked` as argument, it can still pull in the latest and greatest dependencies if you do not commit a `Cargo.lock` file in the source repository. This way, you leave it up to the author of the software how the dependencies should be handled. See:

https://doc.rust-lang.org/cargo/commands/cargo-install.html#dealing-with-the-lockfile

> By default, the Cargo.lock file that is included with the package will be ignored. This means that Cargo will recompute which versions of dependencies to use, possibly using newer versions that have been released since the package was published. The --locked flag can be used to force Cargo to use the packaged Cargo.lock file **if it is available**. This may be useful for ensuring reproducible builds, to use the exact same set of dependencies that were available when the package was published. It may also be useful if a newer version of a dependency is published that no longer builds on your system, or has other problems. The downside to using --locked is that you will not receive any fixes or updates to any dependency. Note that Cargo did not start publishing Cargo.lock files until version 1.37, which means packages published with prior versions will not have a Cargo.lock file available.

---

As an alternative, would you be open for a `CARGO_LOCKED` like configuration option, which if set would add the `--locked` argument to `cargo install`?

brocaar, 2023-08-04:
Instead of implementing a `CARGO_LOCKED` flag, I have changed my Makefile to:

Workspace with multiple crates:

```
define Build/Compile
	$(call Build/Compile/Cargo,crate-a,--locked)
	$(call Build/Compile/Cargo,crate-b,--locked)
	$(call Build/Compile/Cargo,crate-c,--locked)
endef
```

Single crate:

```
define Build/Compile
	$(call Build/Compile/Cargo,.,--locked)
endef
```

This works as well without the need to make changes to the Rust Makefiles.

@oskarirauta @FacundoAcevedo since you are maintaining one or multiple Rust based OpenWrt packages in this repository, you might be interested in this discussion as well.

There is also a `cargo build` vs `cargo install` GitHub issue in the `cargo` repository about this: https://github.com/rust-lang/cargo/issues/7169.

feckert, 2024-02-02:
I think I hit an error with netavark, that is related to this issue. I have **not** updated the Openwrt only made a `make clean` and recompiled the OpenWrt distribution. And then I can no longer compile netavark!

As described in this issue I think that a dependency has changed the version that no longer use the same API as before.

Excerpt from the compile log:
```
   Compiling netavark v1.9.0 (/home/bbworker/bbworker/stable_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/netavark-1.9.0)
     Running `rustc --crate-name build_script_build --edition=2021 build.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debug-assertions=off -C overflow-checks=on --cfg 'feature="default"' --cfg 'feature="deps-serde"' --cfg 'feature="serde"' -C metadata=b43a4ddf9c613042 -C extra-filename=-b43a4ddf9c613042 --out-dir /home/bbworker/bbworker/stable_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/netavark-1.9.0/target/release/build/netavark-b43a4ddf9c613042 -L dependency=/home/bbworker/bbworker/stable_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/netavark-1.9.0/target/release/deps --extern chrono=/home/bbworker/bbworker/stable_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/netavark-1.9.0/target/release/deps/libchrono-7a4caa020c842073.rlib --extern tonic_build=/home/bbworker/bbworker/stable_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/netavark-1.9.0/target/release/deps/libtonic_build-8f16849f421cfec2.rlib`
error[E0308]: mismatched types
    --> /home/bbworker/.openwrt-build-artifacts/cargo/registry/src/index.crates.io-6f17d22bba15001f/nispor-1.2.16/conf/bond.rs:33:33
     |
33   |         mutator.attributes.push(info);
     |                            ---- ^^^^ expected `LinkAttribute`, found a different `LinkAttribute`
     |                            |
     |                            arguments to this method are incorrect
     |
     = note: `LinkAttribute` and `LinkAttribute` have similar names, but are actually distinct types
note: `LinkAttribute` is defined in crate `netlink_packet_route`
    --> /home/bbworker/.openwrt-build-artifacts/cargo/registry/src/index.crates.io-6f17d22bba15001f/netlink-packet-route-0.18.1/src/link/attribute.rs:105:1
     |
105  | pub enum LinkAttribute {
     | ^^^^^^^^^^^^^^^^^^^^^^
note: `LinkAttribute` is defined in crate `netlink_packet_route`
    --> /home/bbworker/.openwrt-build-artifacts/cargo/registry/src/index.crates.io-6f17d22bba15001f/netlink-packet-route-0.19.0/src/link/attribute.rs:105:1
     |
105  | pub enum LinkAttribute {
     | ^^^^^^^^^^^^^^^^^^^^^^
     = note: perhaps two different versions of crate `netlink_packet_route` are being used?
note: method defined here
    --> /home/bbworker/bbworker/stable_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/host/rustc-1.74.0-src/library/alloc/src/vec/mod.rs:1878:12
     |
1878 |     pub fn push(&mut self, value: T) {
     |            ^^^^
error[E0308]: mismatched types
```

lu-zero, 2024-02-02:
try passing `--locked` in the Makefile but also tell upstream please.

feckert, 2024-02-05:
Should I report the problem upstream to the [rust issue](https://github.com/rust-lang/cargo/issues/7169) project?

When I look at the build system in OpenWrt, the repository specifies exactly which versions are used for binaries and libraries during compilation. If I want an update a version because of an security issue, then I have to update the repository. So that on the next build i want to build the binaries or libraries with an other version.

I would expect the same for rust packages, because I want to build **reproducible**. If I don't change anything then the system should behave exactly as before. This is currently **not** the case. The version of an packages will get implicit updated, that is not what I would expect!

Unfortunately, I am not familiar with the rust ecosystem, whether a lock file is always supplied. But if I understand this correctly then we should use `rust build` instead of `rust install`.

lu-zero, 2024-02-05:
> @lu-zero Should I report the problem upstream to the [rust issue](https://github.com/rust-lang/cargo/issues/7169) project?

Nope, `netavark` is the upstream I'm talking to.

> When I look at the build system in OpenWrt, the repository specifies exactly which versions are used for binaries and libraries during compilation. If I need to update a version because of an security issue, then I have to update the repository. So that on the next build the binaries or libraries are build with an fixed version of that.

The problem is what to do for the rust dependencies, either you pass `--locked` to peruse the lockfile provided, or you can pick whichever is the latest dependencies within semver. Everything should works or otherwise it is a bug and should be reported upstream.

> I would expect the same for rust packages, because I want to build **reproducible**. If I don't change anything, then the system should behave exactly as before. This is currently **not** the case. The version of an packages will get implicit updated, that is not what I would expect!

You can use `--locked` as explained above in this issue to pin the dependencies, the "full" solution distro-wise is to keep a custom crate index or to track the dependencies as it is done within Gentoo.

> Unfortunately, I am not familiar with the rust ecosystem, whether a lock file is always supplied. But if I understand this correctly then we should use `rust build` instead of `rust install`.

the crate distribution should always provide one. `cargo install --locked` works as intended.

> But I don't think it's that simple ;-)

It is conceptually simple but it is a bit more work than I'm comfortable to do alone right now :/

brocaar, 2024-02-05:
@feckert @lu-zero this is how I solved it for my project:

```toml
define Build/Compile
	$(call Build/Compile/Cargo,.,--locked)
endef
```

https://github.com/chirpstack/chirpstack-openwrt-feed/blob/master/chirpstack/chirpstack-mqtt-forwarder/Makefile#L53

See also:

> By default, the Cargo.lock file that is included with the package will be ignored. This means that Cargo will recompute which versions of dependencies to use, possibly using newer versions that have [...]

https://doc.rust-lang.org/cargo/commands/cargo-install.html#dealing-with-the-lockfile

---

For all my own OpenWrt packages in the above repo I'm using the `--locked` option, since I want to make sure that compiling the same package at two different times are built against the exact same dependencies.

lu-zero, 2024-05-05:
> For all my own OpenWrt packages in the above repo I'm using the `--locked` option, since I want to make sure that compiling the same package at two different times are built against the exact same dependencies.

Probably we could add an option that sets the default cargo args and/or config.toml so the users have an easy way to override globally.
 
It would allow enterprising users to set their local registry with their pre-vetted set of dependencies.

feckert, 2025-01-22:
@lu-zero Somehow we need a solution for this. I now have the problem again that `netavark` does not compile anymore. I have not made any changes again! From my point of view, I would like to have a reproducible build. 

Cann you pleas elaborate what you mean with that.
```

Probably we could add an option that sets the default cargo args and/or config.toml so the users have an easy way to override globally.

It would allow enterprising users to set their local registry with their pre-vetted set of dependencies.
```

Build error.

```
    ERROR: package/feeds/packages/netavark failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make[1]: *** [/home/bbworker/bbworker/owrt_dev_x86_64/System6/build/openwrt/include/toplevel.mk:233: world] Fehler 1
make[1]: Verzeichnis „/home/bbworker/bbworker/owrt_dev_x86_64/System6/build/openwrt“ wird verlassen
################################################################
Package buildlog '/home/bbworker/bbworker/owrt_dev_x86_64/System6/build/openwrt/logs/package/feeds/packages/netavark/compile.txt'
################################################################
  Installing netavark v1.12.2 (/home/bbworker/bbworker/owrt_dev_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/netavark-1.12.2)
    Updating crates.io index
     Locking 316 packages to latest compatible versions
      Adding async-broadcast v0.5.1 (latest: v0.7.2)
      Adding async-fs v1.6.0 (latest: v2.1.2)
      Adding async-io v1.13.0 (latest: v2.4.0)
      Adding async-lock v2.8.0 (latest: v3.4.0)
      Adding async-process v1.8.1 (latest: v2.3.0)
      Adding axum v0.6.20 (latest: v0.8.2)
      Adding axum-core v0.3.4 (latest: v0.5.1)
      Adding base64 v0.21.7 (latest: v0.22.1)
      Adding bitflags v1.3.2 (latest: v2.8.0)
      Adding clap v4.4.18 (latest: v4.5.27)
      Adding clap_builder v4.4.18 (latest: v4.5.27)
      Adding clap_derive v4.4.7 (latest: v4.5.24)
      Adding clap_lex v0.6.0 (latest: v0.7.4)
      Adding dhcproto v0.9.0 (latest: v0.12.0)
      Adding enum-as-inner v0.5.1 (latest: v0.6.1)
      Adding etherparse v0.13.0 (latest: v0.17.0)
      Adding event-listener v2.5.3 (latest: v5.4.0)
      Adding event-listener v3.1.0 (latest: v5.4.0)
      Adding fastrand v1.9.0 (latest: v2.3.0)
      Adding fixedbitset v0.4.2 (latest: v0.5.7)
      Adding futures-lite v1.13.0 (latest: v2.6.0)
      Adding generic-array v0.14.7 (latest: v1.2.0)
      Adding h2 v0.3.26 (latest: v0.4.7)
      Adding hashbrown v0.12.3 (latest: v0.15.2)
      Adding heck v0.4.1 (latest: v0.5.0)
      Adding hermit-abi v0.3.9 (latest: v0.4.0)
      Adding http v0.2.12 (latest: v1.2.0)
      Adding http-body v0.4.6 (latest: v1.0.1)
      Adding hyper v0.14.32 (latest: v1.5.2)
      Adding hyper-timeout v0.4.1 (latest: v0.5.2)
      Adding idna v0.2.3 (latest: v1.0.3)
      Adding indexmap v1.9.3 (latest: v2.7.1)
      Adding io-lifetimes v1.0.11 (latest: v2.0.4)
      Adding itertools v0.12.1 (latest: v0.14.0)
      Adding linux-raw-sys v0.3.8 (latest: v0.7.0)
      Adding linux-raw-sys v0.4.15 (latest: v0.7.0)
      Adding matchit v0.7.3 (latest: v0.8.6)
      Adding memoffset v0.7.1 (latest: v0.9.1)
      Adding netlink-packet-route v0.19.0 (latest: v0.21.0)
      Adding netlink-packet-route v0.20.1 (latest: v0.21.0)
      Adding nftables v0.4.1 (latest: v0.5.0)
      Adding nix v0.26.4 (latest: v0.29.0)
      Adding nix v0.27.1 (latest: v0.29.0)
      Adding ordered-float v2.10.1 (latest: v4.6.0)
      Adding petgraph v0.6.5 (latest: v0.7.1)
      Adding polling v2.8.0 (latest: v3.7.4)
      Adding proc-macro-crate v1.3.1 (latest: v3.2.0)
[... 257 lines cut ...]
     Running `rustc --crate-name icu_properties --edition=2021 /home/bbworker/dl/cargo/registry/src/index.crates.io-6f17d22bba15001f/icu_properties-1.5.1/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type lib --emit=dep-info,metadata,link -C opt-level=z -C linker-plugin-lto -C codegen-units=1 -C overflow-checks=on --cfg 'feature="compiled_data"' --cfg 'feature="default"' --check-cfg 'cfg(docsrs)' --check-cfg 'cfg(feature, values("bidi", "compiled_data", "datagen", "default", "serde", "std"))' -C metadata=514c1c614f4f908b -C extra-filename=-514c1c614f4f908b --out-dir /home/bbworker/bbworker/owrt_dev_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/netavark-1.12.2/target/x86_64-unknown-linux-musl/release/deps --target x86_64-unknown-linux-musl -C linker=x86_64-openwrt-linux-musl-gcc -C strip=debuginfo -L dependency=/home/bbworker/bbworker/owrt_dev_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/netavark-1.12.2/target/x86_64-unknown-linux-musl/release/deps -L dependency=/home/bbworker/bbworker/owrt_dev_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/netavark-1.12.2/target/release/deps --extern displaydoc=/home/bbworker/bbworker/owrt_dev_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/netavark-1.12.2/target/release/deps/libdisplaydoc-c575d4c43a795daa.so --extern icu_collections=/home/bbworker/bbworker/owrt_dev_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/netavark-1.12.2/target/x86_64-unknown-linux-musl/release/deps/libicu_collections-225b029b397b9a46.rmeta --extern icu_locid_transform=/home/bbworker/bbworker/owrt_dev_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/netavark-1.12.2/target/x86_64-unknown-linux-musl/release/deps/libicu_locid_transform-0a [... cut ...] /build/openwrt/build_dir/target-x86_64_musl/netavark-1.12.2/target`.
To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.
make[4]: *** [Makefile:49: /home/bbworker/bbworker/owrt_dev_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/netavark-1.12.2/.built] Error 101
time: package/feeds/packages/netavark/compile#102.24#6.98#50.88
make: *** [Makefile:44: build] Fehler 2
program finished with exit code 2
elapsedTime=3232.538534
```