openwrt/packages #21731: kmod-i2c-gpio package doesn't  contain a kernel module

MarkTerMors, 2023-08-05:
Maintainer: Sorry, don't know how to find that.
Environment: ath79, gl-inet 6416, 22.03.5

Description:

The package kmod-i2c-gpio doesn't contain any kernel modules,
Just a text file int /etc/modules.d

```
root@mainscontrol:~# opkg list-installed | grep i2c
i2c-tools - 4.3-1
kmod-i2c-algo-bit - 5.10.176-1
kmod-i2c-core - 5.10.176-1
kmod-i2c-gpio - 5.10.176-1
libi2c - 4.3-1

root@mainscontrol:~# opkg files kmod-i2c-gpio
Package kmod-i2c-gpio (5.10.176-1) is installed on root and has the following files:
/etc/modules.d/59-i2c-gpio

brada4, 2023-08-05:
Yes, file misskng in data.tar.gz inside
https://downloads.openwrt.org/releases/22.03.5/targets/ath79/generic/kmods/5.10.176-1-85703b998fa54b6f2c1131f83b0a7d11/kmod-i2c-gpio_5.10.176-1_mips_24kc.ipk

brada4, 2023-08-05:
shall be reported in ../openwrt

KanjiMonster, 2023-08-06:
This is because the i2c-gpio driver is [built into the kernel](https://github.com/openwrt/openwrt/blob/main/target/linux/ath79/generic/config-default#L10), so there is no module to package. The (empty) package exists just to satisfy opkg depdendencies, since other kmod packages would fail to install without it.

closed 2023-08-06 by KanjiMonster

MarkTerMors, 2023-08-08:
That's great, but how is a user supposed to configure it? There's no modinfo, kernel parameters, etc.
The only way seems to be to modify the .dts file for the platform. How exactly to accomplish that is unclear and seems to rely on experimentation. Given a recompile is needed for each attempt, I have no idea how long it will take to get working i2c.

KanjiMonster, 2023-08-08:
There usually is no need to (user available) configuration, unless you are trying to add support for a new board. And in that case you would need to create and modify a dts file anyway. And htis would be the case regardless if it were a module or not.

The expected node properties are described [in its binding documentation](https://www.kernel.org/doc/Documentation/devicetree/bindings/i2c/i2c-gpio.txt).

MarkTerMors, 2023-08-08:
The board is a Gl-inet 6416. Up to 19.x we had a kmod-gpio-i2c-custom which worked with it and could be configured in modules.d.  Since 22.x that seems to be deprecated.
Dmesg shows the driver loading but there is no way to configure which gpio's it uses. I'm still building the image with my modification to the dts.