openwrt/packages #21738: luci-app-upnp: Faulty dependency on miniupnpd-iptables

Ashus, 2023-08-06:
Maintainer: OpenWrt LuCI community
Environment: OpenWrt 23.05.0-rc2

Description:

This package depends on outdated package miniupnpd-iptables and doesn't take miniupnpd-nftables as viable replacement. Obviously iptables has been replaced by nftables in OpenWRT recently and miniupnpd-iptables doesn't work properly anymore. By default luci-app-upnp should come with miniupnpd-nftables.

To temporarily fix this, I need to call:
```
opkg remove miniupnpd-iptables --force-depends
opkg install miniupnpd-nftables
```

brada4, 2023-08-06:
The ptoblem is 'provides' dependencies being selected  based on being 1st in alphabet, not technical merit.
miniupnpd is gotten from -iptables not -nftables variant
then iptables comes from -legacy ipo working -nft kind.
If you instal either non-default provider it goes ahead correctly.

Luci probblems go to ../luci btw, just that i dont see one here.

Ashus, 2023-08-20:
Moved [here](https://github.com/openwrt/luci/issues/5409#issuecomment-1685374865) and closed.

closed 2023-08-20 by Ashus