openwrt/packages #21759: schroot: fails to build with NLS

kesslerm, 2023-08-08:
Maintainer: @jmarcet 
Environment: mvebu, turris-omnia, master/23.05

Description:

The `schroot` package fails to build when NLS is enabled in global settings.

```
[47/55] Linking CXX executable bin/schroot/schroot
FAILED: bin/schroot/schroot 
: && /development/openwrt.org/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-12.3.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -mfloat-abi=hard -fmacro-prefix-map=/development/openwrt.org/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/reschroot-1.6.13/reschroot=reschroot -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/development/openwrt.org/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libiconv-full/include -I/development/openwrt.org/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libintl-full/include -std=c++11 -pedantic -Wall -Wcast-align -Wwrite-strings -Wswitch-default -Wcast-qual -Wunused-variable -Wredundant-decls -Wctor-dtor-privacy -Wnon-virtual-dtor -Wreorder -Wold-style-cast -Woverloaded-virtual -fstrict-aliasing -DNDEBUG -L/development/openwrt.org/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-12.3.0_musl_eabi/usr/lib -L/development/openwrt.org/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-12.3.0_musl_eabi/lib -fuse-ld=bfd -znow -zrelro -L/development/openwrt.org/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libiconv-full/lib -Wl,-rpath-link=/development/openwrt.org/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libiconv-full/lib -L/development/openwrt.org/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libintl-full/lib -Wl,-rpath-link=/development/openwrt.org/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libintl-full/lib   -rdynamic bin/schroot/CMakeFiles/schroot.dir/schroot-main.cc.o bin/schroot/CMakeFiles/schroot.dir/schroot-options.cc.o bin/schroot/CMakeFiles/schroot.dir/schroot.cc.o -o bin/schroot/schroot  sbuild/libsbuild.a  bin/schroot-base/libschroot-base.a  bin/schroot/libschroot-common.a  bin/schroot-base/libschroot-base.a  sbuild/libsbuild.a  /development/openwrt.org/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libboost_iostreams.so.1.82.0  /development/openwrt.org/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libboost_filesystem.so.1.82.0  /development/openwrt.org/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libboost_system.so.1.82.0  /development/openwrt.org/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libboost_program_options.so.1.82.0 && :
/development/openwrt.org/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-12.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/12.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld.bfd: sbuild/libsbuild.a(sbuild-feature.cc.o): in function `sbuild::feature::print_features(std::ostream&)':
sbuild-feature.cc:(.text+0x740): undefined reference to `libintl_gettext'
collect2: error: ld returned 1 exit status
```

A fix for this has already been submitted (but not yet merged) to the upstream project at https://codeberg.org/shelter/reschroot/pulls/7. The same patches, notably https://codeberg.org/kt/reschroot/commit/bcaf4de8f4cd037ced68a2a13323f706161d980f for the NLS issue should be applied here as well.

kesslerm, 2023-08-26:
The cause of the build failure appears to be, that `TARGET_LDFLAGS` is missing `-lint`. 

After adding it to the package Makefile, the package builds successfully even with `CONFIG_BUILD_NLS` set.

```
diff --git a/admin/schroot/Makefile b/admin/schroot/Makefile
index 967efdc07..eabfb9327 100644
--- a/admin/schroot/Makefile
+++ b/admin/schroot/Makefile
@@ -21,6 +21,10 @@ include $(INCLUDE_DIR)/package.mk
 include $(INCLUDE_DIR)/cmake.mk
 include $(INCLUDE_DIR)/nls.mk
 
+ifeq ($(CONFIG_BUILD_NLS),y)
+       TARGET_LDFLAGS += -lintl
+endif
+
 define Package/schroot
   SECTION:=admin
   CATEGORY:=Administration
```

The rest of  the `libintl` settings are added to the `TARGET_*` flags correctly in `$(INCLUDE_DIR)/nls.mk`. It appears that other packages that also include this file don't need to add `-lintl` explicitly.

Is the above the best solution for now, or should the fix be added to `$(INCLUDE_DIR)/nls.mk`?

neheb, 2024-06-19:
This can be closed. I’ve since fixed this.

BKPepe, 2024-06-19:
Any reference? I am reluctant to close this without any more details.

neheb, 2024-06-19:
https://github.com/openwrt/packages/pull/24154

BKPepe, 2024-06-19:
Backported to OpenWrt 23.05 - https://github.com/openwrt/packages/commit/8e683037f089a54db92749d9e4b6d98e8556e032. Closing.

closed 2024-06-19 by BKPepe