openwrt/packages #21764: python3 possible cross compile problems

pmelange, 2023-08-08:
Maintainer: @jefferyto 
Environment: I had this issue on rampis/mt7621 (RB 750GR3) and @PolynomialDivision had a similar issue, but I don't know the arch.  But when I went and made an x86-64 virtual machine, the problem was not reproducible.

Description:
I am opening this issue because of https://github.com/wlanslovenija/tunneldigger/issues/169

Here is the error message from @PolynomialDivision 
```
Sun Jun 11 09:02:54 2023 daemon.err python[4535]: [INFO/tunneldigger.hooks] Running hook 'session.pre-down' via script '/usr/lib/tunneldigger-broker/hooks/teardown'.
Sun Jun 11 09:02:57 2023 daemon.err python[4535]: [INFO/tunneldigger.broker] Creating tunnel (abcd) with id 104.
Sun Jun 11 09:02:57 2023 kern.info kernel: [317719.482982] do_page_fault(): sending SIGSEGV to python for invalid read access from 00000004
Sun Jun 11 09:02:57 2023 kern.info kernel: [317719.500022] epc = 77c2b455 in libpython3.11.so.1.0[77aba000+259000]
Sun Jun 11 09:02:57 2023 kern.info kernel: [317719.512813] ra  = 77c2c16b in libpython3.11.so.1.0[77aba000+259000]
```

and here is the error message from me
```
Sat Jul 29 17:34:11 2023 daemon.err python[5681]: [INFO/tunneldigger.broker] Creating tunnel (b6:01:be:53:f0:d5:71:a2:3a:35) with id 102.
Sat Jul 29 17:34:11 2023 daemon.err python[5681]: [INFO/tunneldigger.tunnel] Set tunnel 102 MTU to 1446.
Sat Jul 29 17:34:11 2023 daemon.err python[5681]: [INFO/tunneldigger.hooks] Running hook 'session.up' via script '/usr/lib/tunneldigger-broker/hooks/setup'.
Sat Jul 29 17:34:11 2023 kern.info kernel: [ 2805.040635] do_page_fault(): sending SIGSEGV to python for invalid read access from 00000004
Sat Jul 29 17:34:11 2023 kern.info kernel: [ 2805.049178] epc = 77b52905 in libpython3.11.so.1.0[779e1000+259000]
Sat Jul 29 17:34:11 2023 kern.info kernel: [ 2805.055614] ra  = 77b5361b in libpython3.11.so.1.0[779e1000+259000]

```

The package both of us were using is tunneldigger-broker from the openwrt-routing repo.

I am assuming there is a cross compiling issue here.  The only reason that I assume this is because it works fine under x86-64.  I do not know how best to test this. If I can help debug this any further, please let me know how.

jefferyto, 2023-08-09:
Is the error reproducible? If so, what are the steps to reproduce?

Do you get this error with snapshots or 23.05, or both?

@PolynomialDivision on what target(s) do you see this error?

> The package both of us were using is tunneldigger-broker from the openwrt-routing repo.

I think you mean the package [here](https://github.com/openwrt/packages/tree/master/net/tunneldigger-broker)? There is no tunneldigger-broker package in the routing repo.

> I am assuming there is a cross compiling issue here.

Not necessarily, it could be a regular coding error.

pmelange, 2023-08-09:
Hi @jefferyto, yes you are right, it's from the packages repo.  I've updated my description.

I got this with both snapshot and 23.05.

To reproduce, you will need two routers.   The server-side router should have tunneldigger-broker installed. You will need about 6.7MB space in /overlay.  My router is an ramips/mt7621 RB750GR3

/etc/config/tunneldigger-broker should look like this:
```
config broker
	list port '8942'
	option interface 'wan'
	option max_cookies '1024'
	option max_tunnels '1024'
	option port_base '20000'
	option tunnel_id_base '100'
	option tunnel_timeout '60'
	option pmtu '0'
	option namespace 'default'
	option connection_rate_limit '1'

config log
	option filename '/dev/null'
	option verbosity 'DEBUG'
	option log_ip_addresses '0'
```
and remember to open up the firewall to allow incoming connections on the wan port.  I have also noticed that if dhcp is enabled on the wan port, that the broker will try to initialize before wan is ready.  So, I recommend a static IP on the broker router.  Otherwise, you might just have to restart the tunneldigger-service.

---
On the client-side router the package tunneldigger should be installed.  My router is a TP-Link WDR4900 V1.

/etc/config/tunneldigger should look like this (the server IP may be different)
```
config broker 'tdclient'
	list address '192.168.1.100:8942'
	option uuid 'test'
	option interface 'tdclient'
	option broker_selection 'usage'
	option bind_interface 'wan'
	option enabled '1'
```
and /etc/config/network should have
```
config interface 'tdclient'
	option proto 'none'
	option device 'tdclient'
	option disabled '0'
```

---

On the client-side router, you may need to restart tunneldigger a few times to reproduce this error.  
```/etc/init.d/tunneldigger restart```.  The error is visible on the server-router (with tunneldigger-broker)

I do not have this error on an x85-64 virtual machine

pmelange, 2023-08-09:
I reran the test with a different server-router. This time it's a mvebu/cortexa9 Linksys WRT1900ACS v1 / Linksys WRT1900ACS v2 / Linksys Shelby.  I'm still getting the crash of the tunneldigger-broker, but the error message is not showing up in the logfile.

So I ran the service on the command line.  Here is the output
```
root@OpenWrt:~# /usr/bin/python -m tunneldigger_broker.main /var/etc/tunneldigger-broker.cfg
[INFO/tunneldigger.broker] Initializing the tunneldigger broker.
[INFO/tunneldigger.broker] Registered script '/usr/lib/tunneldigger-broker/hooks/setup' for hook 'session.up'.
[INFO/tunneldigger.broker] Registered script '/usr/lib/tunneldigger-broker/hooks/teardown' for hook 'session.pre-down'.
[INFO/tunneldigger.broker] Registered script '/usr/lib/tunneldigger-broker/hooks/mtu-changed' for hook 'session.mtu-changed'.
[INFO/tunneldigger.broker] Maximum number of tunnels is 1024.
[INFO/tunneldigger.broker] Tunnel identifier base is 100.
[INFO/tunneldigger.broker] Listening on 192.168.188.22:8942.
[INFO/tunneldigger.broker] Broker initialized.
[INFO/tunneldigger.broker] Creating tunnel (test) with id 100.
[INFO/tunneldigger.tunnel] Set tunnel 100 MTU to 1446.
[INFO/tunneldigger.hooks] Running hook 'session.up' via script '/usr/lib/tunneldigger-broker/hooks/setup'.
[INFO/tunneldigger.hooks] (session.up/3275) Unable to determine which bridge to use for MTU 1446.
[WARNING/tunneldigger.tunnel] Tunnel 100 (test) got error from remote peer
[INFO/tunneldigger.tunnel] Closing tunnel 100 (test) after 7 seconds (reason=0x1)
[INFO/tunneldigger.hooks] Running hook 'session.pre-down' via script '/usr/lib/tunneldigger-broker/hooks/teardown'.
[INFO/tunneldigger.broker] Creating tunnel (test) with id 101.
[INFO/tunneldigger.tunnel] Set tunnel 101 MTU to 1446.
[INFO/tunneldigger.hooks] Running hook 'session.up' via script '/usr/lib/tunneldigger-broker/hooks/setup'.
[INFO/tunneldigger.hooks] (session.up/3300) Unable to determine which bridge to use for MTU 1446.
[WARNING/tunneldigger.tunnel] Tunnel 101 (test) got error from remote peer
[INFO/tunneldigger.tunnel] Closing tunnel 101 (test) after 14 seconds (reason=0x1)
[INFO/tunneldigger.hooks] Running hook 'session.pre-down' via script '/usr/lib/tunneldigger-broker/hooks/teardown'.
[INFO/tunneldigger.broker] Creating tunnel (test) with id 102.
[INFO/tunneldigger.tunnel] Set tunnel 102 MTU to 1446.
[INFO/tunneldigger.hooks] Running hook 'session.up' via script '/usr/lib/tunneldigger-broker/hooks/setup'.
[INFO/tunneldigger.hooks] (session.up/3325) Unable to determine which bridge to use for MTU 1446.
[WARNING/tunneldigger.tunnel] Tunnel 102 (test) got error from remote peer
[INFO/tunneldigger.tunnel] Closing tunnel 102 (test) after 7 seconds (reason=0x1)
[INFO/tunneldigger.hooks] Running hook 'session.pre-down' via script '/usr/lib/tunneldigger-broker/hooks/teardown'.
[INFO/tunneldigger.broker] Creating tunnel (test) with id 103.
Segmentation fault
```

There are no messages in dmesg

If there is anything else I can do to help, please let me know

pmelange, 2023-08-20:
I spent some time working on this issue.  Now I have more to present.  When I run the tunneldigger-broker with the ```-X faulthandler``` command line option, I get a better stack trace when the program crashes.

I would also like to reiterate that I can not reproduce this error on an x86-64 virtual machine.  The error that I'm reporting here happened on an ramips/mt7621 RB 750GR3

```
Fatal Python error: Segmentation fault

Current thread 0x77e91d8c (most recent call first):
  Garbage-collecting
  File "/usr/lib/python3.11/contextlib.py", line 144 in __exit__
  File "/usr/lib/python3.11/subprocess.py", line 1694 in _get_handles
  File "/usr/lib/python3.11/subprocess.py", line 992 in __init__
  File "/usr/lib/python3.11/site-packages/tunneldigger_broker/hooks.py", line 28 in __init__
  File "/usr/lib/python3.11/site-packages/tunneldigger_broker/hooks.py", line 157 in run_hook
  File "/usr/lib/python3.11/site-packages/tunneldigger_broker/tunnel.py", line 154 in setup_tunnel
  File "/usr/lib/python3.11/site-packages/tunneldigger_broker/broker.py", line 144 in create_tunnel
  File "/usr/lib/python3.11/site-packages/tunneldigger_broker/broker.py", line 248 in create_tunnel
  File "/usr/lib/python3.11/site-packages/tunneldigger_broker/protocol.py", line 159 in message
  File "/usr/lib/python3.11/site-packages/tunneldigger_broker/broker.py", line 267 in message
  File "/usr/lib/python3.11/site-packages/tunneldigger_broker/network.py", line 179 in read
  File "/usr/lib/python3.11/site-packages/tunneldigger_broker/eventloop.py", line 60 in start
  File "/usr/lib/python3.11/site-packages/tunneldigger_broker/main.py", line 116 in <module>
  File "<frozen runpy>", line 88 in _run_code
  File "<frozen runpy>", line 198 in _run_module_as_main
Segmentation fault
```

The last bit of code within tunneldigger-broker is https://github.com/wlanslovenija/tunneldigger/blob/51a5e46ad143c92d2867835a563146ec4fbc6211/broker/src/tunneldigger_broker/hooks.py#L28-L32

I had also added an extra print to show what the command and arguments are before the call to Popen, and they look completely fine.  I had also changed the called script to only print the passed args to stdout to make sure that the script isn't doing anything incorrectly.  Yet still this error is happening.

I also switched from subprocess.Popen() to subprocess.run() and ran into the same issue.

This does not happen every time https://github.com/wlanslovenija/tunneldigger/blob/51a5e46ad143c92d2867835a563146ec4fbc6211/broker/src/tunneldigger_broker/hooks.py#L28-L32 is run, only some of the time.  This, and the fact that I can't reproduce it on x86-64 leads me to believe that it's a platform specific problem.

Any further help figuring out what the problem is would be greatly appreciated.

pmelange, 2023-08-30:
Is there any advice on this topic?

pmelange, 2023-09-18:
@jefferyto do you have any advice?

jefferyto, 2023-10-31:
> I had also changed the called script to only print the passed args to stdout to make sure that the script isn't doing anything incorrectly. Yet still this error is happening.

When you changed the script to only print to stdout, does the segfault still occur? If so, does it occur before or after the output is printed?

jefferyto, 2023-11-02:
I've opened an issue upstream for this: https://github.com/python/cpython/issues/111629

pmelange, 2023-11-05:
> When you changed the script to only print to stdout, does the segfault still occur? If so, does it occur before or after the output is printed?

I just moved and have to set up my test env again.  I will get back to you.

pmelange, 2023-11-05:
I just saw your upstream issue. It looks like my input is not needed right now.  Let me know if there is anything I can do.