openwrt/packages #21782: mwan3: mwan3_get_src_ip returns deprecated addresses

l-jonas, 2023-08-10:
Maintainer: @feckert @aaronjg 
Environment: OpenWrt 22.03

Description: ``mwan3_get_src_ip`` in ``/lib/mwan3/common.sh`` uses IPv6 addresses that are marked as deprecated if ``network_get_ipaddr6``reports them (it always reports only one address); if commenting out the call to it, then the fallback using ``ip a``is used and this works as it seems to sort the addresses so that the non-deprecated one is shown first

Issues that result from that:

- ``mwan3 use`` sets a wrong src IP
- ``mwan3track`` uses a wrong src IP

In one environment, this causes connectivity issues (traffic from the deprecated src IP is lost somewhere) so that ``mwan3track`` considers an interface offline

Independent of the environment, it never harms to use the non-deprecated address if it exists

brada4, 2023-08-10:
> traffic from the deprecated src IP is lost somewhere

That is RFC violation. Deprecated addresses should have connectivity.

l-jonas, 2023-08-11:
You are right, that's what [RFC 4862](https://www.rfc-editor.org/rfc/rfc4862) says. I have to check that network.

Additionally, the RFC states that 

> While an address is in a deprecated
> state, its use is discouraged, but not strictly forbidden.  New
> communication (e.g., the opening of a new TCP connection) should use
> a preferred address when possible.  A deprecated address should be
> used only by applications that have been using it and would have
> difficulty switching to another address without a service disruption.

So mwan3 ``should`` not use it.

brada4, 2023-08-11:
You are suggesting more robust approach to address selection by avoiding 'privacy' short-term addresses for normal permanent ones.
Probably worth first using/listing only perpetual addrrsses and using short-lived only in case good ones are absret.

l-jonas, 2023-08-14:
It looks like you are mixing up some few things. The deprecated addresses are not temporary addresses. In case of temporary addresses, after generating a new one, the old one becomes deprecated. However, there is one non-deprecated temporary address when using the privacy extensions.

brada4, 2023-08-15:
It does not make deprecated addresses wrong or non-routable.

l-jonas, 2023-08-15:
@brada4 Please make a statement that is understandable and adds anything new. According to RFC 4862, the deprecated address is valid and thus "may appear as the source or destination address of a packet, and the Internet routing system is expected to deliver packets sent to a valid address to their intended recipients.". It is my problem if my network does not do it this way.

And I repeat myself: The same RFC states "A deprecated address should no longer be used as a source address in new communications". This is no MUST but a SHOULD. It would still be better to implement it this way, for example when using ``mwan3 use`` with wget for a long running download it is very bad idea to use the address that is already deprecated.

brada4, 2023-08-15:
It would be worth submitting few-line PR, in hope to draw attention that `ip a` positions "better" addresses first, in hope that script function could be improved alike to return them first, but relying on omnipresent command would be fine too and essentially fix your internet connection.

l-jonas, 2023-08-15:
If I would be the maintainer, then I would not like trivial merge requests. Moreover, I can not understand why a maintainer would react to a trivial PR but not to a reported issue that contains a workaround.

I see a "risk" (chance) that the maintainer knows things about the package that I don't know like design decisions. So I assume that there is a reason why "ip a" is only used as a fallback. If this is true, then there is the possibility for a better solution than always using the fallback.

brada4, 2023-08-15:
actually you found something that slipped past their radar.
config_get is from common uci functions, it is in other package and works how it works with 100-some dependants.
i dunno if its worth sorting ip addr list for sed that extracts addresses, but the simplest one-liner seems with no unintended consequences until dependencies are made predictable.

feckert, 2023-08-16:
I have not dealt with this topic of deprecated IPv6 addresses before, so this is new to me.

>So I assume that there is a reason why "ip a" is only used as a fallback.

The reason is that, if the netifd does not have a source address, I ask the interface directly. That is all. 
I use the ubus backend from netifd for the source [ipv4](
https://github.com/openwrt/openwrt/blob/master/package/base-files/files/lib/functions/network.sh#L30) or the [ipv6](https://github.com/openwrt/openwrt/blob/master/package/base-files/files/lib/functions/network.sh#L37) addresses. 
As I understand it, the netifd does not know the new source address either. What output do you get from the command `ubus call network.interface dump`?

l-jonas, 2023-08-16:
I will try to create a dump at the right time (of ubus call ... and ip a). I don't have a deprecated address during the whole day, so I cannot do that now.

I assume that luci uses netifd to get the addresses. As far as I remember, luci showed both addresses.

l-jonas, 2023-08-17:
result from ``ip a`` (anonymized)

```
X:  ethX: ...
    link/ether ...
    inet ...
       valid_lft forever preferred_lft forever
    inet6 2000:2002:2003:2004:1:2:3:4/64 scope global dynamic noprefixroute 
       valid_lft 7062sec preferred_lft 3462sec
    inet6 2000:2001:200a:200b:1:2:3:4/64 scope global deprecated dynamic noprefixroute 
       valid_lft 3714sec preferred_lft 0sec
    inet6 fe80::1:2:3:4/64 scope link 
       valid_lft forever preferred_lft forever
```

result from ``ubus call network.interface dump`` (anonymized + shortened):

```
{
	"interface": [
		...
		{
			"interface": "wanb6",
			"up": true,
			"pending": false,
			"available": true,
			"autostart": true,
			"dynamic": false,
			"uptime": 3946,
			"l3_device": "ethX",
			"proto": "dhcpv6",
			"device": "ethX",
			"updated": [
				"prefixes"
			],
			"metric": 2,
			"dns_metric": 0,
			"delegation": false,
			"ipv4-address": [
				
			],
			"ipv6-address": [
				{
					"address": "2000:2001:200a:200b:1:2:3:4",
					"mask": 64,
					"preferred": 0,
					"valid": 3714
				},
				{
					"address": "2000:2002:2003:2004:1:2:3:4",
					"mask": 64,
					"preferred": 3462,
					"valid": 7062
				}
			],
			"ipv6-prefix": [
				{
					"address": ...,
					"mask": 60,
					"preferred": 3248,
					"valid": 6848,
					"class": "wanb6",
					"assigned": {
						
					}
				}
			],
			"ipv6-prefix-assignment": [
				
			],
			"route": [
				...
			],
			"dns-server": [
				...
			],
			"dns-search": [
				
			],
			"neighbors": [
				
			],
			"inactive": {
				"ipv4-address": [
					
				],
				"ipv6-address": [
					
				],
				"route": [
					
				],
				"dns-server": [
					
				],
				"dns-search": [
					
				],
				"neighbors": [
					
				]
			},
			"data": ...
		}
	]
}
```

Conclusion: there is the attribute ``preferred`` in seconds. The highest value indicates the address that is expected to be valid for the longest duration.

brada4, 2023-08-17:
Sort by preferred number?

brada4, 2023-08-17:
Youf json has it.

l-jonas, 2023-08-17:
The part of the JSON that I posted only contains IPv6 addresses and no IPv4 ones. The IPv4 addresses (in the part that I did not include in the post) only have these two attributes: If you want an example:

```
		{
			"interface": "wanb4",
			"up": true,
			"pending": false,
			"available": true,
			"autostart": true,
			"dynamic": false,
			"uptime": ...,
			"l3_device": "ethX",
			"proto": "dhcp",
			"device": "ethX",
			"updated": [
				"addresses",
				"routes",
				"data"
			],
			"metric": 2,
			"dns_metric": 0,
			"delegation": true,
			"ipv4-address": [
				{
					"address": ...,
					"mask": 24
				}
			],
			"ipv6-address": [
				
			],
			"ipv6-prefix": [
				
			],
			"ipv6-prefix-assignment": [
				
			],
			"route": [
				{
					"target": "0.0.0.0",
					"mask": 0,
					"nexthop": ...,
					"source": ...
				}
			],
			"dns-server": [
				...
			],
			"dns-search": [
				...
			],
			"neighbors": [
				
			],
			"inactive": {
				"ipv4-address": [
					
				],
				"ipv6-address": [
					
				],
				"route": [
					
				],
				"dns-server": [
					
				],
				"dns-search": [
					
				],
				"neighbors": [
					
				]
			},
			"data": {
				"dhcpserver": ...,
				"leasetime": ...,
				"ntpserver": ...
			}
		},
```

closed 2024-02-05 by neheb