openwrt/packages #21820: banip: change variable {ban_rdapfile} to ${ban_rdapfile}

qosmio, 2023-08-13:
The variable `$ban_rdapfile` is missing an `$` which ends up writing it as a literal to the root directory `"/{ban_rdapfile}"`.

```patch
diff --git a/net/banip/files/banip-functions.sh b/net/banip/files/banip-functions.sh
index 545753635..cc3a00974 100644
--- a/net/banip/files/banip-functions.sh
+++ b/net/banip/files/banip-functions.sh
@@ -1443,7 +1443,7 @@ f_monitor() {

                "${ban_logreadcmd}" -fe "${ban_logterm%%??}" 2>/dev/null |
                        while read -r line; do
-                               : >"{ban_rdapfile}"
+                               : >"${ban_rdapfile}"
                                proto=""
                                ip="$(printf "%s" "${line}" | "${ban_awkcmd}" 'BEGIN{RS="(([0-9]{1,3}\\.){3}[0-9]{1,3})+"}{if(!seen[RT]++)printf "%s ",RT}')"
                                ip="$(f_trim "${ip}")"
```

BKPepe, 2023-08-18:
@dibdot

dibdot, 2023-10-03:
fixed with https://github.com/openwrt/packages/commit/00cad2980cc7707f662acb1fa2a51c4e4fc331d9

closed 2023-10-03 by dibdot