openwrt/packages #21821: mwan3: interface is error (16) and tracking is active

jsalatiel, 2023-08-13:
I have been using mwan3 for almost 4 years flawless, but recently after upgrading openwrt from 19.07 to 22.03(fw4) my configuration stopped working. 
I have 3 interfaces, wan1, wan2 are "normal" interfaces and are working just fine.
The problem is the third interface, it is an openvpn interface ( tun7 ), this interface is configured as  unmanaged on openwrt and the tun7 device is created by the openvpn startup script:

[image]

This interface is used only by one desktop  and all traffic from this desktop should go out using the tun7 interface or not go out at all ( blackhole ). So this is my config 
```
config globals 'globals'  
        option mmx_mask '0x3F00'
        option local_source 'lan'
...
                                    
config interface 'work'
        option enabled '1'
        list track_ip '8.8.8.8'     
        list track_ip '208.67.220.220'
        option family 'ipv4'
        option reliability '1'        
        option count '3'
        option timeout '5' 
        option interval '5'            
        option down '3'   
        option up '3'               

config member 'work_m1_w1'            
        option interface 'work'       
        option metric '1'             
        option weight '1' 

config policy 'vpn_work'            
        list use_member 'work_m1_w1'                                                                                  
        option sticky  1            
        option timeout 60             
        option last_resort 'blackhole'

config rule 'bound_to_work'           
        option src_ip '192.168.254.169'
        option proto 'all'             
        option use_policy 'vpn_work'   
```

This is the logs from openwrt where the interface is up:
```
Sun Aug 13 19:50:17 2023 daemon.notice openvpn(work)[27451]: TUN/TAP device tun7 opened
Sun Aug 13 19:50:17 2023 daemon.notice openvpn(work)[27451]: net_iface_mtu_set: mtu 1500 for tun7
Sun Aug 13 19:50:17 2023 daemon.notice openvpn(work)[27451]: net_iface_up: set tun7 up
Sun Aug 13 19:50:17 2023 daemon.notice netifd: Network device 'tun7' link is up
Sun Aug 13 19:50:17 2023 daemon.notice openvpn(work)[27451]: net_addr_ptp_v4_add: 10.103.0.234 peer 10.103.0.233 dev tun7
Sun Aug 13 19:50:17 2023 daemon.notice openvpn(work)[27451]: /usr/libexec/openvpn-hotplug up work tun7 1500 1557 10.103.0.234 10.103.0.233 init
Sun Aug 13 19:50:17 2023 user.notice mwan3-hotplug[27482]: Execute ifup event on interface work (tun7)
Sun Aug 13 19:50:19 2023 user.info mwan3track[19797]: Detect ifup event on interface work (tun7)
Sun Aug 13 19:50:19 2023 user.warn mwan3track[19797]: no src ipv4 address found from netifd for interface 'work' dev 'tun7' guessing 10.103.0.234
Sun Aug 13 19:50:19 2023 user.notice mwan3track[19797]: Interface work (tun7) is online
Sun Aug 13 19:50:20 2023 user.notice firewall: Reloading firewall due to ifup of work (tun7)
```
mwan3 status
```
mwan3  status
Interface status:
 interface wan1 is online 00h:00m:09s, uptime 27h:29m:11s and tracking is active
 interface wan2 is online 00h:00m:10s, uptime 27h:29m:19s and tracking is active
 interface work is error (16) and tracking is active

Current ipv4 policies:
balanced:
 wan2 (50%)
 wan1 (50%)
prefer_wan1:
 wan1 (100%)
prefer_wan2:
 wan2 (100%)
vpn_work:
 work (100%)
wan1_only:
 wan1 (100%)
wan2_only:
 wan2 (100%)

```

The connectivity on openwrt itself is working :
```
# curl --interface tun7 checkip.dyndns.com
<html><head><title>Current IP Check</title></head><body>Current IP Address: REDACTED</body></html>
``` 
ip route on the openwrt
```
default via 100.65.16.1 dev pppoe-wan1 proto static metric 10 
default via 179.156.178.1 dev lan4 proto static src 179.156.178.231 metric 20 
10.39.0.1 via 10.39.1.17 dev tun7 
10.39.1.17 dev tun7 proto kernel scope link src 10.39.1.18 
100.65.16.1 dev pppoe-wan1 proto kernel scope link src REDACTED
172.16.0.0/24 dev br-lan.102 proto kernel scope link src 172.16.0.1 
172.16.31.0/24 dev br-lan.101 proto kernel scope link src 172.16.31.1
179.156.178.0/24 dev lan4 proto static scope link metric 20 
192.168.254.0/24 dev br-lan.1 proto kernel scope link src 192.168.254.1
```
But the desktop 192.168.154.169 only gets no route to host to everywhere.
```
[desktop] curl checkip.dyndns.com  -v
*   Trying 193.122.130.0:80...
* connect to 193.122.130.0 port 80 failed: Network is unreachable
*   Trying 132.226.8.169:80...
* connect to 132.226.8.169 port 80 failed: Network is unreachable
*   Trying 132.226.247.73:80...
* connect to 132.226.247.73 port 80 failed: Network is unreachable
*   Trying 158.101.44.242:80...
* connect to 158.101.44.242 port 80 failed: Network is unreachable
*   Trying 193.122.6.168:80...
* connect to 193.122.6.168 port 80 failed: Network is unreachable
* Failed to connect to checkip.dyndns.com port 80 after 12308 ms: Network is unreachable
* Closing connection 0
curl: (7) Failed to connect to checkip.dyndns.com port 80 after 12308 ms: Network is unreachable
```
Any idea what may be causing this? The same config used to work on openwrt 19.07

jsalatiel, 2023-08-14:
Found the problem. Apparently default routes for the openvpn are not being pulled.

closed 2023-08-14 by jsalatiel

vincejv, 2023-11-23:
For anyone experiencing it with WG VPNs, don't forget to tick Route Allowed IPs in Luci or add `option route_allowed_ips '1'` in `/etc/config/network`