openwrt/packages #21841: miniupnpd-nftables: STUN does not work because it does not open local ports properly

LGA1150, 2023-08-17:
Maintainer: @stintel @stangri @jow- 
Environment: OpenWrt SNAPSHOT, r23763-46ed38adeb, mt7622

Description:
My ISP provides CGNAT IP address with fullcone (including TCP). However, when I try to set up STUN for miniupnpd, it failed to detect the fullcone type.

I think OpenWrt's firewall blocked the incoming STUN packets, because the same had happened when I tested the NAT type on Windows: it does not detect the fullcone type unless I disabled the firewall or add the detection tool to the firewall allow list.

I set `option log_output '1'` in upnpd config to enable verbose logs, then I noticed it tried to add firewall rules to `upnp_forward`.
```
daemon.info miniupnpd[8842]: STUN: Performing with host=stun.miwifi.com and port=3478 ...
daemon.debug miniupnpd[8842]: resolve_stun_host: stun.miwifi.com:3478 => x.x.x.x:3478
daemon.info miniupnpd[8842]: perform_stun: local ports 40225 33021 33632 52450
daemon.debug miniupnpd[8842]: wait_for_stun_responses: waiting 3 secs and 0 usecs
daemon.debug miniupnpd[8842]: wait_for_stun_responses: received responses: 1
daemon.debug miniupnpd[8842]: wait_for_stun_responses: waiting 3 secs and 0 usecs
daemon.debug miniupnpd[8842]: wait_for_stun_responses: select(): no more responses
daemon.debug miniupnpd[8842]: wait_for_stun_responses: waiting 3 secs and 0 usecs
daemon.debug miniupnpd[8842]: wait_for_stun_responses: select(): no more responses
daemon.debug miniupnpd[8842]: wait_for_stun_responses: waiting 3 secs and 0 usecs
daemon.debug miniupnpd[8842]: wait_for_stun_responses: select(): no more responses
daemon.debug miniupnpd[8842]: table_cb(0x7fdaf0ad90, 0x7fdaf0bdf0) fw4 upnp_forward 2
daemon.debug miniupnpd[8842]: table_cb(0x7fdaf0af80, 0x7fdaf0bdf0) fw4 upnp_forward 2
daemon.debug miniupnpd[8842]: table_cb(0x7fdaf0b17c, 0x7fdaf0bdf0) fw4 upnp_forward 2
daemon.debug miniupnpd[8842]: table_cb(0x7fdaf0b378, 0x7fdaf0bdf0) fw4 upnp_forward 2
daemon.debug miniupnpd[8842]: table_cb(0x7fdaf0ad90, 0x7fdaf0bdf0) fw4 upnp_forward 2
daemon.debug miniupnpd[8842]: table_cb(0x7fdaf0af80, 0x7fdaf0bdf0) fw4 upnp_forward 2
daemon.debug miniupnpd[8842]: table_cb(0x7fdaf0b17c, 0x7fdaf0bdf0) fw4 upnp_forward 2
daemon.debug miniupnpd[8842]: table_cb(0x7fdaf0ad90, 0x7fdaf0bdf0) fw4 upnp_forward 2
daemon.debug miniupnpd[8842]: table_cb(0x7fdaf0af80, 0x7fdaf0bdf0) fw4 upnp_forward 2
daemon.debug miniupnpd[8842]: table_cb(0x7fdaf0ad90, 0x7fdaf0bdf0) fw4 upnp_forward 2
daemon.debug miniupnpd[8842]: parse_stun_response: Type 0x0101, Length 48, Magic Cookie 2112a442
daemon.debug miniupnpd[8842]: parse_stun_response: MAPPED-ADDRESS x.x.x.x:62527
daemon.debug miniupnpd[8842]: parse_stun_response: RESPONSE-ORIGIN x.x.x.x:3478
daemon.debug miniupnpd[8842]: parse_stun_response: OTHER-ADDRESS x.x.x.x:3479
daemon.debug miniupnpd[8842]: parse_stun_response: XOR-MAPPED-ADDRESS x.x.x.x:62527
daemon.notice miniupnpd[8842]: perform_stun: 1 response out of 4 received
daemon.warn miniupnpd[8842]: STUN: ext interface wan with private IP address 192.168.1.101 is now behind restrictive or symmetric NAT with public IP address 101.86.230.102 which does not support port forwarding
daemon.warn miniupnpd[8842]: NAT on upstream router blocks incoming connections set by miniupnpd
daemon.warn miniupnpd[8842]: Turn off NAT on upstream router or change it to full-cone NAT 1:1 type
```

When miniupnpd is restarting, 4 firewall rules are added to `upnp_forward` chain:
```
root@OpenWrt:~# nft list chain inet fw4 upnp_forward
table inet fw4 {
        chain upnp_forward {
                iif "wan" th dport 50826 @nh,128,32 0xc0a80165 @nh,72,8 0x11 accept
                iif "wan" th dport 45510 @nh,128,32 0xc0a80165 @nh,72,8 0x11 accept
                iif "wan" th dport 50952 @nh,128,32 0xc0a80165 @nh,72,8 0x11 accept
                iif "wan" th dport 55067 @nh,128,32 0xc0a80165 @nh,72,8 0x11 accept
        }
}
```
As the name suggests, this chain is only used in `forward` chain, not `input` chain, so miniupnpd does not open the ports properly.
If I manually call `upnp_forward` in `input`, everything works:
`nft insert rule inet fw4 input jump upnp_forward`

```
daemon.info miniupnpd[9316]: STUN: Performing with host=stun.miwifi.com and port=3478 ...
daemon.debug miniupnpd[9316]: resolve_stun_host: stun.miwifi.com:3478 => x.x.x.x:3478
daemon.info miniupnpd[9316]: perform_stun: local ports 34390 44751 37992 52958
daemon.debug miniupnpd[9316]: wait_for_stun_responses: waiting 3 secs and 0 usecs
daemon.debug miniupnpd[9316]: wait_for_stun_responses: received responses: 1
daemon.debug miniupnpd[9316]: wait_for_stun_responses: waiting 3 secs and 0 usecs
daemon.debug miniupnpd[9316]: wait_for_stun_responses: received responses: 2
daemon.debug miniupnpd[9316]: wait_for_stun_responses: waiting 3 secs and 0 usecs
daemon.debug miniupnpd[9316]: wait_for_stun_responses: received responses: 3
daemon.debug miniupnpd[9316]: wait_for_stun_responses: waiting 3 secs and 0 usecs
daemon.debug miniupnpd[9316]: wait_for_stun_responses: received responses: 4
daemon.debug miniupnpd[9316]: table_cb(0x7fc8ba0a40, 0x7fc8ba1aa0) fw4 upnp_forward 2
daemon.debug miniupnpd[9316]: table_cb(0x7fc8ba0c30, 0x7fc8ba1aa0) fw4 upnp_forward 2
daemon.debug miniupnpd[9316]: table_cb(0x7fc8ba0e2c, 0x7fc8ba1aa0) fw4 upnp_forward 2
daemon.debug miniupnpd[9316]: table_cb(0x7fc8ba1028, 0x7fc8ba1aa0) fw4 upnp_forward 2
daemon.debug miniupnpd[9316]: table_cb(0x7fc8ba0a40, 0x7fc8ba1aa0) fw4 upnp_forward 2
daemon.debug miniupnpd[9316]: table_cb(0x7fc8ba0c30, 0x7fc8ba1aa0) fw4 upnp_forward 2
daemon.debug miniupnpd[9316]: table_cb(0x7fc8ba0e2c, 0x7fc8ba1aa0) fw4 upnp_forward 2
daemon.debug miniupnpd[9316]: table_cb(0x7fc8ba0a40, 0x7fc8ba1aa0) fw4 upnp_forward 2
daemon.debug miniupnpd[9316]: table_cb(0x7fc8ba0c30, 0x7fc8ba1aa0) fw4 upnp_forward 2
daemon.debug miniupnpd[9316]: table_cb(0x7fc8ba0a40, 0x7fc8ba1aa0) fw4 upnp_forward 2
daemon.debug miniupnpd[9316]: parse_stun_response: Type 0x0101, Length 48, Magic Cookie 2112a442
daemon.debug miniupnpd[9316]: parse_stun_response: MAPPED-ADDRESS x.x.x.x:63246
daemon.debug miniupnpd[9316]: parse_stun_response: RESPONSE-ORIGIN x.x.x.x:3478
daemon.debug miniupnpd[9316]: parse_stun_response: OTHER-ADDRESS x.x.x.x:3479
daemon.debug miniupnpd[9316]: parse_stun_response: XOR-MAPPED-ADDRESS x.x.x.x:63246
daemon.debug miniupnpd[9316]: parse_stun_response: Type 0x0101, Length 48, Magic Cookie 2112a442
daemon.debug miniupnpd[9316]: parse_stun_response: MAPPED-ADDRESS x.x.x.x:63247
daemon.debug miniupnpd[9316]: parse_stun_response: RESPONSE-ORIGIN x.x.x.x:3479
daemon.debug miniupnpd[9316]: parse_stun_response: OTHER-ADDRESS x.x.x.x:3479
daemon.debug miniupnpd[9316]: parse_stun_response: XOR-MAPPED-ADDRESS x.x.x.x:63247
daemon.debug miniupnpd[9316]: parse_stun_response: Type 0x0101, Length 48, Magic Cookie 2112a442
daemon.debug miniupnpd[9316]: parse_stun_response: MAPPED-ADDRESS x.x.x.x:63248
daemon.debug miniupnpd[9316]: parse_stun_response: RESPONSE-ORIGIN x.x.x.x:3478
daemon.debug miniupnpd[9316]: parse_stun_response: OTHER-ADDRESS x.x.x.x:3479
daemon.debug miniupnpd[9316]: parse_stun_response: XOR-MAPPED-ADDRESS x.x.x.x:63248
daemon.debug miniupnpd[9316]: parse_stun_response: Type 0x0101, Length 48, Magic Cookie 2112a442
daemon.debug miniupnpd[9316]: parse_stun_response: MAPPED-ADDRESS x.x.x.x:63249
daemon.debug miniupnpd[9316]: parse_stun_response: RESPONSE-ORIGIN x.x.x.x:3479
daemon.debug miniupnpd[9316]: parse_stun_response: OTHER-ADDRESS x.x.x.x:3479
daemon.debug miniupnpd[9316]: parse_stun_response: XOR-MAPPED-ADDRESS x.x.x.x:63249
daemon.info miniupnpd[9316]: STUN: ext interface wan with IP address 192.168.1.101 is now behind unrestricted full-cone NAT 1:1 with public IP address x.x.x.x and firewall does not block incoming connections set by miniunnpd
daemon.info miniupnpd[9316]: Port forwarding is now enabled
```

kiligone, 2023-12-14:
移动大内网内用这个方法吗，大佬

daiaji, 2024-01-07:
Same situation.

Beaverfffan, 2024-01-11:
how can we fix it temporarily

daiaji, 2024-01-13:
```
venv/bin/pystun3 -H stun.qq.com
NAT Type: Full Cone
External IP: *.*.*.*
External Port: 34561
Press any key to continue
```
Even in Full Cone NAT mode, the STUN of miniupnpd does not work for me.

funigna, 2024-04-02:
u right.
after change input policy to accept, then stun works without problem.

the problem is miniupnpd mistakenly put this rule to chain nft_forward
you can inspect its act following this path

perform_stun -> add_filter_rule2 -> rule_set_filter -> rule_set_filter_common
`nftnl_rule_set_str(r, NFTNL_RULE_CHAIN, nft_forward_chain);`
nft_forward_chain is a char *, which points the chain miniupnpd will use.

solution is straightforward but tough for me.
it involves the process to add the rule into right chain. assign the rule for top priority.
and remove it after stun is tested.

funigna, 2024-04-06:
use miniupnpd-iptables and it work for me.

QZAiXH, 2024-04-29:
Is there any progress so far? I encountered the same situation and couldn't solve it with `nft insert rule inet fw4 input jump upnp_forward`. I'm using ImmortalWrt 23.05.2.

The output of `nft list chain inet fw4 upnp_forward` is as follows:

```
root@ImmortalWrt:~# nft list chain inet fw4 upnp_forward
table inet fw4 {
	chain upnp_forward {
	}
}
```

purofle, 2024-10-03:
> Is there any progress so far? I encountered the same situation and couldn't solve it with `nft insert rule inet fw4 input jump upnp_forward`. I'm using ImmortalWrt 23.05.2.目前有进展吗？我遇到了同样的情况，无法解决 `nft insert rule inet fw4 input jump upnp_forward` 。我正在使用 ImmortalWrt 23.05.2。
> 
> The output of `nft list chain inet fw4 upnp_forward` is as follows:的输出 `nft list chain inet fw4 upnp_forward` 如下：
> 
> ```
> root@ImmortalWrt:~# nft list chain inet fw4 upnp_forward
> table inet fw4 {
> 	chain upnp_forward {
> 	}
> }
> ```

Same problem.

purofle, 2024-10-03:
> Is there any progress so far? I encountered the same situation and couldn't solve it with `nft insert rule inet fw4 input jump upnp_forward`. I'm using ImmortalWrt 23.05.2.
>
> The output of `nft list [...]

The same problem.