openwrt/packages #21865: strongswan: All strongswan-related packages is missed for x86_64 architecture

Pristavkin, 2023-08-19:
Maintainer: @pprindeville @Thermi
Environment: x86_64  OpenWrt 22.03.5, r20134-5f15225c1e 

Description:
All strongswan-related packages are missed for x86_64 architecture. Not available with opkg, not listed on [package status page](https://openwrt.org/packages/pkgdata/strongswan)

```
ssh root@10.154.7.32

BusyBox v1.35.0 (2023-04-27 20:28:15 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 22.03.5, r20134-5f15225c1e
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@lab-gw:~# opkg update
Downloading https://downloads.openwrt.org/releases/22.03.5/targets/x86/64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/releases/22.03.5/targets/x86/64/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/x86_64/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/x86_64/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/x86_64/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/x86_64/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/x86_64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/x86_64/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/x86_64/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/x86_64/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/x86_64/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/x86_64/telephony/Packages.sig
Signature check passed.
root@lab-gw:~# opkg install strongswan
Unknown package 'strongswan'.
Collected errors:
 * opkg_install_cmd: Cannot install package strongswan.
root@lab-gw:~#
```

luizluca, 2023-08-24:
https://downloads.openwrt.org/releases/faillogs-22.03/x86_64/packages/strongswan/compile.txt

pprindeville, 2023-08-29:
> https://downloads.openwrt.org/releases/faillogs-22.03/x86_64/packages/strongswan/compile.txt

Not found.

luizluca, 2023-08-29:
So it is fixed ;-)

```
# grep VERSION= /etc/os-release 
VERSION="22.03.5"
# opkg list | grep strongswan
crconf - 2018-03-02-8bd99640-1 - Linux crypto layer configuraton tool. Use this to manipulate kernel Crypto API/framework stuff, like drivers, templates, etc. Example: https://wiki.strongswan.org/projects/strongswan/wiki/Pcrypt Most interesting stuff you can find on https://wiki.strongswan.org
strongswan - 5.9.5-13.1 - StrongSwan is an OpenSource IPsec implementation for the Linux operating system. This package contains shared libraries and scripts.
strongswan-charon - 5.9.5-13.1 - StrongSwan is an OpenSource IPsec implementation for the Linux operating system. This package contains charon, an IKEv2 keying daemon.
strongswan-charon-cmd - 5.9.5-13.1 - StrongSwan is an OpenSource IPsec implementation for the Linux operating system. This package contains the charon-cmd utility.
strongswan-default - 5.9.5-13.1 - StrongSwan is an OpenSource IPsec implementation for the Linux operating system. This meta-package contains only dependencies to match upstream defaults.
```

pprindeville, 2023-08-29:
No idea.  I've been seeing a lot of CI/CD failures only on x86_64 even when it builds on all other platforms.

Don't know who maintains CI/CD these days if not @champtar or who to reach out to to help investigate.

Thermi, 2023-08-29:
I took a look at that log file before it was deleted. The problem with building was with a redefinition of symbols related to wolfssl. I presume it was resolved when something in wolfssl headers changed that resolved the collision.

closed 2023-10-17 by Pristavkin