openwrt/packages #21866: adguardhome: build failure after go version bumped to 1.21

hayzamjs, 2023-08-19:
Maintainer: @dobo90 
Environment: x86_64 OpenWRT master branch

Description:

Unable to build net/adguardhome on the master branch after the go1.21 upgrade that happened a few days ago, here's the error I'm getting:

```
github.com/quic-go/quic-go/internal/qtls                                                                                                                                                                                  
# github.com/quic-go/quic-go/internal/qtls                                                                                                                                                                                
../../../../../dl/go-mod-cache/github.com/quic-go/quic-go@v0.36.2/internal/qtls/go121.go:5:13: cannot use "The version of quic-go you're using can't be built on Go 1.21 yet. For more details, please see https://github.
com/quic-go/quic-go/wiki/quic-go-and-Go-versions." (untyped string constant "The version of quic-go you're using can't be built on Go 1.21 yet. F...) as int value in variable declaration
```

hnyman, 2023-08-19:
Apparently AdGuard Home v0.107.36 is needed for go lang 1.21

hnyman, 2023-08-19:
I made a PR for version bump:
https://github.com/openwrt/packages/pull/21869

hayzamjs, 2023-08-21:
@hnyman so is this an issue upstream?