openwrt/packages #21868: adguardhome 0.107.21-1 doesn't wait for ip6 address and fails to start after reboot

iljah, 2023-08-19:
Environment: 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.5'
DISTRIB_REVISION='r20134-5f15225c1e'
DISTRIB_TARGET='rockchip/armv8'
DISTRIB_ARCH='aarch64_generic'

Description:

adguardhome doesn't start after reboot: `[fatal] couldn't start forwarding DNS server: listening to udp socket: list en udp [fda3:ca1f:2ac2::1]:53: bind: cannot assign requested address`

Moving adguard init from S25 to S97 doesn't help. Adding respawn 300 3 3 doesn't help giving same result several times. Starting manually after logging in with ssh works. Above address is bridge interface, does it not get created fast enough? It's using static ip4 but I don't see any options for ip6 when editing it in web ui.

iljah, 2023-08-19:
@jamesmacwhite seemed to active previously: https://github.com/openwrt/packages/issues/18585

CallMeR, 2023-08-27:
https://github.com/AdguardTeam/AdGuardHome/releases/tag/v0.107.36

The official latest version of Adguard Home is already ahead of the OpenWRT's repository .

iljah, 2023-08-27:
Is this issue fixed there?

PR #22211 (2024-03-18): adguardhome: wait for interfaces to be up at boot
Maintainer: @dobo90 
Compile tested: n/a
Run tested: @iljah and @mangkoran 

Description:

This should allow the service to be activated even earlier during the boot process and also avoids race condition against network.

~~I've not tested this change yet since I'm not affected by this issue.
@iljah if you can compile your own packages, please test this.~~

Fixes #21868.
closed 2024-03-18 by stangri