openwrt/packages #21875: usteer: avoid kicking wds ap-s/relays

brada4, 2023-08-20:
Maintainer: @blocktrron 
Environment: Various models v22.03.5 ath79/mt76

Description:

WDS gear is not expected to change channel while connected in WDS mode.
When wds is enabled on AP then WDS clients establish one connection to primary wlan0 interface and create addidional wlan0.staN interface for each connected relay/AP so far so good. 
Once usteer finds channel loaded it tries to roam clients to upper band, also including the wlan0 (2.4ghz) associations that back .staX interfaces causing random glitches in faraway AP-s which have to wait sometimes minutes to regain connectivity.
It all works non-stop as long as no usteer is activated on main gateway.

ptpt52, 2023-08-24:
mark.

brada4, 2024-11-18:
imlressive core packave maintenance work without breaking sweat