openwrt/packages #21881: dnscrypt-proxy2 out of date (2.1.4 vs 2.1.5 latest)

ascendbeing, 2023-08-21:
hopefully it's an easy bump but this release I feel is worth bumping to cause there were some serious problems with the current version regarding HTTP3 or HTTP2 (I forget exactly) etc.
2.1.4 I think failed to build on some snapshot targets too. just saying, maybe 2.1.5 would fall less.

brada4, 2023-08-21:
Probably you refer to this patch closing idle UDP sockets:
https://github.com/DNSCrypt/dnscrypt-proxy/commit/24a301b1af83bf613bbb967e6a90de2a14fc6709
Or you mean to disable h3 support until it can reuse sockets for multiple requests?

ascendbeing, 2023-08-21:
> Probably you refer to this patch closing idle UDP sockets: [DNSCrypt/dnscrypt-proxy@24a301b](https://github.com/DNSCrypt/dnscrypt-proxy/commit/24a301b1af83bf613bbb967e6a90de2a14fc6709) Or you mean to disable h3 support until it can reuse sockets for multiple requests?

nah I'm not saying to disable it outright if this gets bumped. but yeah that was the issue I was having w/ 2.1.4, I would be interested in trying the new version

brada4, 2023-08-21:
I mean as a short-term measure until h3 support is stabilised, i.e each UDP socket serving as quasi-permanent communication channel to each upstream server pooling all requests in same "connection"

PR #22032 (2023-09-15): dnscrypt-proxy2: update to version 2.1.5
Maintainer: me
Compile tested: Nope, I was not able to compile it on MacBook A1990, while using OpenWrt 22.03. The previous version was just fine. Preparing currently env with OpenWrt master to give it another try until then draft.

Description:
- Fixes compilation with Go 1.21+.

Fixes #21881
closed 2023-09-15 by BKPepe