openwrt/packages #21887: mwan3: mwan3_get_mwan3track_status() can't find PID and incorrectly labels interface status

jlvosloo, 2023-08-23:
I'm currently using openwrt 22.03.05 release.

Coming from openwrt v18, I've now noticed a lot of the following error messages:

cat: can't open '/proc/8140/cmdline': No such file or directory

I've traced it down to the following function in the common.sh of mwan3.

```
mwan3_get_mwan3track_status()
{
	local track_ips pid
	mwan3_list_track_ips()
	{
		track_ips="$1 $track_ips"
	}
	config_list_foreach "$1" track_ip mwan3_list_track_ips

	if [ -n "$track_ips" ]; then
		pid="$(pgrep -f "mwan3track $1$")"
		if [ -n "$pid" ]; then
			if [ "$(cat /proc/"$(pgrep -P $pid)"/cmdline)" = "sleep${MAX_SLEEP}" ]; then
				tracking="paused"
			else
				tracking="active"
			fi
		else
			tracking="down"
		fi
	else
		tracking="not enabled"
	fi
	echo "$tracking"
}
```

I updated this function with some debug commands to double check that is was the origin of the issue:

```
mwan3_get_mwan3track_status() {
    local track_ips pid parent_pid

    mwan3_list_track_ips() {
        track_ips="$1 $track_ips"
    }
    config_list_foreach "$1" track_ip mwan3_list_track_ips

    if [ -n "$track_ips" ]; then
        pid="$(pgrep -f "mwan3track $1$")"
        if [ -n "$pid" ]; then
            parent_pid="$(pgrep -P $pid)"
            if [ -z "$parent_pid" ]; then
                echo "Debug: No parent PID found for mwan3track PID: $pid" >> /tmp/mwan3_debug.log
                tracking="down" # Default to 'down' when parent PID is not found
            else
                if [ ! -f /proc/$parent_pid/cmdline ]; then
                    echo "Debug: Cmdline file not found for parent PID $parent_pid" >> /tmp/mwan3_debug.log
                    tracking="down" # Default to 'down' when cmdline file for parent PID is not found
                else
                    local cmdline_content=$(cat /proc/$parent_pid/cmdline)
                    if [ "$cmdline_content" = "sleep${MAX_SLEEP}" ]; then
                        tracking="paused"
                    else
                        tracking="active"
                    fi
                fi
            fi
        else
            tracking="down"
        fi
    else
        tracking="not enabled"
    fi
    echo "$tracking"
}

```

This is a snippet from my mwan3_debug.log

Debug: No parent PID found for mwan3track PID: 1593
30082
Debug: No parent PID found for mwan3track PID: 1592
Debug: No parent PID found for mwan3track PID: 1593
30186
Debug: No parent PID found for mwan3track PID: 1593
30470
Debug: No parent PID found for mwan3track PID: 1593
Debug: No parent PID found for mwan3track PID: 1592
Debug: No parent PID found for mwan3track PID: 1593
31378
Debug: No parent PID found for mwan3track PID: 1593
31775
Debug: No parent PID found for mwan3track PID: 1593
32021
Debug: Cmdline file not found for parent PID 32416
Debug: No parent PID found for mwan3track PID: 1593
32435
Debug: No parent PID found for mwan3track PID: 1592
Debug: No parent PID found for mwan3track PID: 922
1593
Debug: No parent PID found for mwan3track PID: 1037
1593
Debug: Cmdline file not found for parent PID 1083
Debug: No parent PID found for mwan3track PID: 1528
1593
Debug: No parent PID found for mwan3track PID: 1593
1780
Debug: No parent PID found for mwan3track PID: 1592
Debug: No parent PID found for mwan3track PID: 1592
Debug: Cmdline file not found for parent PID 3308
Debug: No parent PID found for mwan3track PID: 1593
3460
Debug: No parent PID found for mwan3track PID: 1593
3558
Debug: No parent PID found for mwan3track PID: 1592

This is causing mwan3 to incorrectly label an interface as DOWN and causing network stability issues.

Could someone please help me understand the root cause of this issue?

Thanks!

feckert, 2023-09-20:
Cann you try this? https://github.com/openwrt/packages/pull/22158

feckert, 2023-10-04:
No reaction from the creator

closed 2023-10-04 by feckert

jlvosloo, 2023-10-05:
@feckert sorry for late reply. I have tested your latest commit to master and it has solved this issue. Thank you very much.

feckert, 2023-10-06:
@jlvosloo No problem, I was also late.