openwrt/packages #21932: banip: Dependency on logd breaks syslog-ng installation (again)

kesslerm, 2023-08-26:
Maintainer: @dibdot 

Since commit 82a491bac8 the `banip` package depends on `logd` again. This dependency was previously removed in PR https://github.com/openwrt/packages/pull/13848 after some discussion on issue https://github.com/openwrt/packages/issues/13820.

Is this simply an oversight, or is there some new functionality that can only be supported with `logd` but not `syslog-ng`?

kuretru, 2023-09-10:
@kesslerm Hi, did your banip with syslog-ng working now? I test it, but banIP seems could not parse logs correctly.

kesslerm, 2023-09-18:
I have only compile tested, using the turris-omnia toolchain (hbd branch).

brada4, 2023-09-20:
logread is used only for asterisk as per script. Rest should be totally compatible down to removing logread as dependency even having it in scripts.

dibdot, 2023-10-03:
Nope, the log detection depends on logread ATM and is not compatible with other log systems.

kesslerm, 2023-10-03:
What about `logread` compatibility layers such as the [one provided](https://github.com/openwrt/packages/blob/master/admin/syslog-ng/files/logread) by `syslog-ng`?

dibdot, 2023-10-03:
That does not work - I'll provide a suitable wrapper in a future update.

commit c4e81407 (2023-11-05): banip: release 0.9.2-1

* the log file monitor now supports standard log files used by other log daemons like syslog-ng
  Set 'ban_logreadfile' accordingly, by default it points to /var/log/messages
* removed logd dependency, closes #21932
commit 4c197c1a (2023-11-10): banip: release 0.9.2-1

* the log file monitor now supports standard log files used by other log daemons like syslog-ng
  Set 'ban_logreadfile' accordingly, by default it points to /var/log/messages
* removed logd dependency, closes #21932

(cherry picked from commit c4e814074003a1d45bb583e98cac435575e09ca4)
closed 2023-11-05 by dibdot