openwrt/packages #21938: acme-acmesh + acme-acmesh-dnsapi + luci-app-acme, can't use dns authentication method.

guidozualdi, 2023-08-28:
Maintainer: @\tohojo
Environment: ARMv7 Processor rev 5 (v7l), AVM FRITZ!Box 7530, pq40xx/generic, OpenWrt 23.05.0-rc3

Description:
If I attempt to create an ACME configuration with dns authentication, it seems to be ignored and acme.sh is launched without necessary parameters:

With luci interface I generated following config file:
```
config acme
	option account_email 'notify@example.com'
	option debug '1'

config cert 'Router'
	option enabled '1'
	option staging '1'
	option use_staging '0'
	option keylength '2048'
	option update_uhttpd '1'
	option validation_method 'dns'
	option dns 'dns_ovh'
	list credentials 'OVH_AK="xxxxxxxxxx"'
	list credentials 'OVH_AS="xxxxxxxxxx"'
	list credentials 'OVH_CK="xxxxxxxxxx"'
	list domains 'domain.example.com'
```
options **validation_method** and **dns** seems to be consistent with acme package documentation https://openwrt.org/docs/guide-user/services/tls/acmesh .

By launching /etc/init.d/acme start I obtain:
```
acme-acmesh: Running ACME for avm.comunifvg.ovh
acme-acmesh: /usr/lib/acme/client/acme.sh --debug --renew --home /etc/acme -d domain.example.com
```

If I launch acme.sh manually with correct parameters, as for example
```
#!/bin/bash
export OVH_AK="xxxxxxxxxx"
export OVH_AS="xxxxxxxxxx"
export OVH_CK="xxxxxxxxxx"
/usr/lib/acme/client/acme.sh --issue -i -d domain.example.com --dns dns_ovh --server letsencrypt

```
certificates are created correctly.

guidozualdi, 2023-08-28:
acme packages version is 3.0.6-1.

ashtonian, 2023-09-05:
I too have this issue. It looks like its ignoring the config file and sending "myemail@example.com" even though the config file has all the details. acme, acme-dns, and acme-luci are all installed.

cat /etc/config/acme
```
config acme
	option account_email 'myemail'
	option debug '1'

config cert 'wildcard2'
	option enabled '1'
	option use_staging '1'
	option keylength '2048'
	list domains 'mydomain'
	list domains '*.mydomain'
	option update_uhttpd '1'
	option update_nginx '1'
	option validation_method 'dns'
	option dns 'cloudflare'
	list credentials 'CF_Token="mytoken"'
	list credentials 'CF_Account_ID="myaccount"'
```
System Log
```
Tue Sep  5 16:31:59 2023 daemon.info acme-acmesh: Running ACME for mydomain
Tue Sep  5 16:31:59 2023 daemon.info acme-acmesh: /usr/lib/acme/client/acme.sh --debug --renew --home /etc/acme -d mydomain
Tue Sep  5 16:32:01 2023 daemon.debug acme: cleaning up
```

// root@wrt:/etc/ssl/acme# /usr/lib/acme/client/acme.sh --debug --renew --home /etc/acme -d mydomain.erg
```
[Tue Sep  5 16:32:53 UTC 2023] Lets find script dir.
[Tue Sep  5 16:32:53 UTC 2023] _SCRIPT_='/usr/lib/acme/client/acme.sh'
[Tue Sep  5 16:32:53 UTC 2023] _script='/usr/lib/acme/client/acme.sh'
[Tue Sep  5 16:32:53 UTC 2023] _script_home='/usr/lib/acme/client'
[Tue Sep  5 16:32:53 UTC 2023] Using config home:/etc/acme
https://github.com/acmesh-official/acme.sh
v3.0.6
[Tue Sep  5 16:32:53 UTC 2023] Running cmd: renew
[Tue Sep  5 16:32:53 UTC 2023] _renewServer
[Tue Sep  5 16:32:53 UTC 2023] Using config home:/etc/acme
[Tue Sep  5 16:32:53 UTC 2023] config file is empty, can not read DEFAULT_ACME_SERVER
[Tue Sep  5 16:32:53 UTC 2023] default_acme_server
[Tue Sep  5 16:32:53 UTC 2023] ACME_DIRECTORY='https://acme.zerossl.com/v2/DV90'
[Tue Sep  5 16:32:53 UTC 2023] DOMAIN_PATH='/etc/acme/mydomain.org'
[Tue Sep  5 16:32:53 UTC 2023] Renew: 'mydomain.org'
[Tue Sep  5 16:32:53 UTC 2023] Le_API='https://acme-v02.api.letsencrypt.org/directory'
[Tue Sep  5 16:32:53 UTC 2023] Renew to Le_API=https://acme-v02.api.letsencrypt.org/directory
[Tue Sep  5 16:32:53 UTC 2023] Using config home:/etc/acme
[Tue Sep  5 16:32:53 UTC 2023] ACME_DIRECTORY='https://acme-v02.api.letsencrypt.org/directory'
[Tue Sep  5 16:32:53 UTC 2023] _main_domain='mydomain.org'
[Tue Sep  5 16:32:53 UTC 2023] _alt_domains='*.mydomain.org'
[Tue Sep  5 16:32:53 UTC 2023] Le_NextRenewTime
[Tue Sep  5 16:32:53 UTC 2023] Using ACME_DIRECTORY: https://acme-v02.api.letsencrypt.org/directory
[Tue Sep  5 16:32:53 UTC 2023] _init api for server: https://acme-v02.api.letsencrypt.org/directory
[Tue Sep  5 16:32:53 UTC 2023] GET
[Tue Sep  5 16:32:53 UTC 2023] url='https://acme-v02.api.letsencrypt.org/directory'
[Tue Sep  5 16:32:53 UTC 2023] timeout=
[Tue Sep  5 16:32:53 UTC 2023] _CURL='curl --silent --dump-header /etc/acme/http.header  -L  -g '
[Tue Sep  5 16:32:53 UTC 2023] ret='0'
[Tue Sep  5 16:32:53 UTC 2023] ACME_KEY_CHANGE='https://acme-v02.api.letsencrypt.org/acme/key-change'
[Tue Sep  5 16:32:53 UTC 2023] ACME_NEW_AUTHZ
[Tue Sep  5 16:32:53 UTC 2023] ACME_NEW_ORDER='https://acme-v02.api.letsencrypt.org/acme/new-order'
[Tue Sep  5 16:32:53 UTC 2023] ACME_NEW_ACCOUNT='https://acme-v02.api.letsencrypt.org/acme/new-acct'
[Tue Sep  5 16:32:53 UTC 2023] ACME_REVOKE_CERT='https://acme-v02.api.letsencrypt.org/acme/revoke-cert'
[Tue Sep  5 16:32:53 UTC 2023] ACME_AGREEMENT='https://letsencrypt.org/documents/LE-SA-v1.3-September-21-2022.pdf'
[Tue Sep  5 16:32:53 UTC 2023] ACME_NEW_NONCE='https://acme-v02.api.letsencrypt.org/acme/new-nonce'
[Tue Sep  5 16:32:53 UTC 2023] Using CA: https://acme-v02.api.letsencrypt.org/directory
[Tue Sep  5 16:32:53 UTC 2023] _on_before_issue
[Tue Sep  5 16:32:53 UTC 2023] _chk_main_domain='mydomain.org'
[Tue Sep  5 16:32:53 UTC 2023] _chk_alt_domains='*.mydomain.org'
[... 19 lines cut ...]
[Tue Sep  5 16:32:54 UTC 2023] _ret='0'
[Tue Sep  5 16:32:54 UTC 2023] POST
[Tue Sep  5 16:32:54 UTC 2023] _post_url='https://acme-v02.api.letsencrypt.org/acme/new-acct'
[Tue Sep  5 16:32:54 UTC 2023] _CURL='curl --silent --dump-header /etc/acme/http.header  -L  -g '
[Tue Sep  5 16:32:54 UTC 2023] _ret='0'
[Tue Sep  5 16:32:54 UTC 2023] code='400'
[Tue Sep  5 16:32:54 UTC 2023] Register account Error: {
  "type": "urn:ietf:params:acme:error:invalidContact",
  "detail": "Error creating new account :: invalid contact domain. Contact emails @example.org are forbidden",
  "status": 400
}
[Tue Sep  5 16:32:54 UTC 2023] _on_issue_err
[Tue Sep  5 16:32:54 UTC 2023] Please add '--debug' or '--log' to check more details.
[Tue Sep  5 16:32:54 UTC 2023] See: https://github.com/acmesh-official/acme.sh/wiki/How-to-debug-acme.sh
[Tue Sep  5 16:32:54 UTC 2023] Diagnosis versions:
openssl:openssl
OpenSSL 3.0.10 1 Aug 2023 (Library: OpenSSL 3.0.10 1 Aug 2023)
apache:
apache doesn't exist.
nginx:
nginx version: nginx/1.25.0 (x86_64-pc-linux-gnu)
built with OpenSSL 3.0.10 1 Aug 2023
TLS SNI support enabled
...
```

stokito, 2023-10-01:
@guidozualdi  please run with debug mode.
Also please note that you have staging enabled e.g. real certs won't be issued and deployed. 
The `option keylength '2048'` should be replaced with `option keysize 'rsa2048'`.
The `option update_uhttpd '1'` is ignored and can be removed. But instead you need to create a symlink in the www .well-known folder to `/var/run/acme`. It should be on Wiki.

Also please ensure that you installed the acme-dns.

@ashtonian you need to specify your real email. You have specified `email@example.org` and the LetsEncrypt doesn't allow them.

stokito, 2023-10-01:
@ashtonian sorry, I missed that you configured the correct email.
The acmesh saves account details to its own config. Check the /etc/acme folder and change the email manually.

stangri, 2023-10-08:
> option dns 'cloudflare'

Works for me with `option dns 'dns_cf'`.

diepeterpan, 2026-01-06:
I have configured and am using a local ACME server when running the command **/etc/init.d/acme renew** the script errors, and it is not using the ACME URL defined in the Luci GUI and reverts to **zerossl**.

```
Running cmd: renew
_renewServer
Using config home: /etc/acme
Config file is empty, cannot read DEFAULT_ACME_SERVER
default_acme_server
ACME_DIRECTORY='https://acme.zerossl.com/v2/DV90'
```