openwrt/packages #21989: strongswan: dependencies hell

timur-davletshin, 2023-09-01:
Maintainer: @pprindeville
Environment: (OpenWrt 23.05.0-rc3)

Description:

**1.** I believe strongswan packages (both full and default) should depend on kmod-xfrm-interface package.
Without it I get error in logread:

```
daemon.info ipsec: 00[KNL] received netlink error: Unknown device type (95)
daemon.info ipsec: 00[KNL] failed to create XFRM interface 'xfrmi-test-6147'
```

**2.** Same can be said about strongswan-mod-drbg (results in non-functional setup if strongswan-default is installed)

**3.** strongswan-full should include strongswan-mod-gmpdh in its dependencies too or user will get:

`plugin 'gmpdh': failed to load - gmpdh_plugin_create not found and no plugin file available`

**4.** I believe strongswan-mod-wolfssl is broken because of wolfSSL version mismatch.
I get:

`plugin 'wolfssl' failed to load: Error relocating /usr/lib/ipsec/plugins/libstrongswan-wolfssl.so: wolfssl_ec_diffie_hellman_create: symbol not found`

tobiasbrunner, 2023-09-04:
Some comments on the points you brought up.

>  I believe strongswan packages (both full and default) should depend on kmod-xfrm-interface package.
Without it I get error in logread:

Not a fatal error and only an problem if the user actually want's to use XFRM interfaces, in which case the `kmod-xfrm-interface` is required anyway.

> Same can be said about strongswan-mod-drbg (results in non-functional setup if strongswan-default is installed)

I don't see why that would be the case because the _drbg_ plugin is really only used by the legacy _ntru_ plugin and when the _gmp_ plugin generates private keys. So how does it result in a "non-functional setup"?

> strongswan-full should include strongswan-mod-gmpdh in its dependencies too or user will get:

Not a significant error. And that plugin is completely useless when the `strongswan-mod-gmp` package is installed and the _gmp_ plugin is loaded (`full` and `default` both depend on it). The _gmpdh_  plugin is basically a trimmed down version of the _gmp_ plugin that only provides DH (i.e. no RSA) and links libgmp statically.

In `full`, the `strongswan-mod-gmp` package is also completely redundant as it also depends on the `strongswan-mod-openssl` package (that's actually true for a lot of packages `full` installs as the _openssl_ plugin pretty much provides all the crypto operations the built-in crypto plugins like _aes_, _ccm_, _gcm_, _hmac_, _sha1_ or _sha2_ do). Also, installing `strongswan-mod-wolfssl` or `strongswan-mod-gcrypt` doesn't really serve a purpose if `strongswan-mod-openssl` is installed. So `full` seems pretty bloated.

> I believe strongswan-mod-wolfssl is broken because of wolfSSL version mismatch. I get:

`wolfssl_ec_diffie_hellman_create` is a function provided by the _wolfssl_ plugin itself. In both `wolfssl_plugin.c` (where the function is referenced) and `wolfssl_ec_diffie_hellmn.c` (where it is defined) we depend on `HAVE_ECC_DHE`, so it's weird that it would be undefined only in one location but not the other.

timur-davletshin, 2023-09-04:
I have impression that people who maintain packages are not using it.

pprindeville, 2023-09-05:
> I have impression that people who maintain packages are not using it.

I do not use wolfssl, you’re correct.

vincejv, 2023-10-17:
@pprindeville 
i'm getting the ff error on `strongswan-full`, the log files are now gone, but something like this
```
failed to create tun device 
```
and only way to fix it is to install `kmod-tun`

timur-davletshin, 2023-10-17:
It is what I was talking about - dependencies are not properly resolved.

vincejv, 2023-10-17:
@timur-davletshin i'm giving this package the benefit of the doubt, if i can't get it working, i'll switch to softethervpn and crossing my fingers 🤞 that it works

ericclose, 2024-04-06:
>  I believe strongswan-mod-wolfssl is broken because of wolfSSL version mismatch.
I get:
`plugin 'wolfssl' failed to load: Error relocating /usr/lib/ipsec/plugins/libstrongswan-wolfssl.so: wolfssl_ec_diffie_hellman_create: symbol not found`

_There are other (redundant) crypto plugins installed._

* souce: [error message of swanctl on openwrt](https://github.com/strongswan/strongswan/discussions/2033)

Neustradamus, 2025-10-22:
@ffainelli, @xwang1498: This ticket is cited in "Turris OS 9.0 is in RC!":
- https://forum.turris.cz/t/turris-os-9-0-is-in-rc/22087/32

Can you solve this ticket?

cc: @peci1.

vincejv, 2026-08-19:
hello bumping this issue. because currently `strongswan-full` installs both `strongswan-swanctl` and `strongswan-ipsec` we do not need both and it will actually cause issues as both services listen on the same port and serve the same purpose, one is the modern implementation, one is legacy. can we change it so that it's installed as `strongswan-swanctl | strongswan-ipsec`??

the workaround currently is to disable `/etc/init.d/ipsec` service

feckert, 2026-08-19:
Time to clean things up?

strongswan-full I think was meant for building the entire StrongSwan system, not for target installations. And let’s be real: the stroke backend (strongswan-ipsec) has been deprecated for some time and will eventually be removed.
We should just rely on swanctl (strongswan-swanctl). The only real question is: policy-based or xfrm?
I recently updated the UCI IPsec config to align with swanctl, but that broke compatibility with the old stroke setup.
I’m currently using strongswan-default (so strongswan-swanctl).

dhrm1k, 2026-08-19:
@feckert I'm looking into this.

As a first step, would it make sense to remove `strongswan-ipsec` and
`strongswan-mod-stroke` from `strongswan-full`, leaving
`strongswan-swanctl` as the supported frontend?

That would avoid installing both init scripts on new installations while
leaving the legacy packages available for users who explicitly need
them. I would keep the policy-based/XFRM choice separate unless it
requires a dependency change here.