openwrt/packages #21990: luci-app-wireguard: QR code feature request

timur-davletshin, 2023-09-01:
Maintainer: @Environment: (put here arch, model, OpenWrt version)

Description:

(In order to provide compatibility with the official WG app for Android) export peer IP address in QR code if mask `/32` is used in `list allowed_ips` (IPv4) or `/128` for IPv6.

For example:

```
config wireguard_vpn 'wgclient'
	option preshared_key 'longkeygoeshere='
	option private_key 'longkeygoeshere='
	option public_key 'longkeygoeshere='
	list allowed_ips '192.168.10.128/32'
	list allowed_ips 'fd00:10::128/128'
```

Same about DNS server setup.

```
config interface 'vpn'
	option proto 'wireguard'
	option listen_port '51820'
	option private_key 'longkeygoeshere='
	list addresses '192.168.10.1/24'
	list addresses 'fd00:10::1/64'
```

DNS server address can be deducted from that `list-adresses`. Additional check via uci can be made to make sure that dnsmasq is up and running (same).

aleks-mariusz, 2023-09-29:
@Entropy512 looks like there's a simple small [qrencode utility package](https://openwrt.org/packages/pkgdata/qrencode) with a [small lib dependency](https://openwrt.org/packages/pkgdata/libqrencode).. altogether about 30kb so not too unreasonable.

I found [this procedure on how to use the tool](https://www.cyberciti.biz/faq/how-to-generate-wireguard-qr-code-on-linux-for-mobile/#:~:text=Now%20create%20the%20QR%20code), and [this example site](https://www.wireguardconfig.com/qrcode) which probably uses this under the hood.

would be curious to see to see what you come up with (even though i'm not currently a wireguard user, i might be in the future)

egc112, 2023-10-10:
WireGuard already has a QR package available you 'only' have to add the required items in LuCi e.g. address, DNS and MTU