openwrt/packages #21991: openvswitch: compilation error

patrykk, 2023-09-01:
Maintainer: @stintel  @paulfertser @dhewg 
Environment: mediatek_filogic , Asus TUF AX4200, main commit 560965d5826626e3be8c1f1db194db43cc7002cf

Description:
Hi, 
I tried to build openwrt from branch main ( commit 560965d5826626e3be8c1f1db194db43cc7002cf - Thu Aug 31 ).
I got errors like below for package openvswitch.

```
touch /usr/build/builds/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/openvswitch-2.17.0/.built_check
CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -ffile-prefix-map=/usr/build/builds/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/openvswitch-2.17.0=openvswitch-2.17.0 -flto=auto -fno-fat-lto-objects -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -std=gnu99  -I/usr/build/builds/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/include -I/usr/build/builds/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include/fortify -I/usr/build/builds/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include " CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -ffile-prefix-map=/usr/build/builds/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/openvswitch-2.17.0=openvswitch-2.17.0 -flto=auto -fno-fat-lto-objects -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -std=gnu99 -flto=auto -fno-fat-lto-objects  -I/usr/build/builds/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/include -I/usr/build/builds/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include/fortify -I/usr/build/builds/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include " LDFLAGS="-L/usr/build/builds/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/lib -L/usr/build/builds/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/lib -fuse-ld=bfd -flto=auto -fuse-linker-plugin -znow -zrelro " PYTHONPATH="/usr/build/builds/openwrt/staging_dir/hostpkg/lib/python3.11:/usr/build/builds/openwrt/staging_dir/hostpkg/lib/python3.11/site-packages" make  -C /usr/build/builds/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/openvswitch-2.17.0/. AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -ffile-prefix-map=/usr/build/builds/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/openvswitch-2.17.0=openvswitch-2.17.0 -flto=auto -fno-fat-lto-objects -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -std=gnu99" LD="aarch64-openwrt-linux-musl-ld.bfd" NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CROSS="aarch64-openwrt-linux-musl-" ARCH="aarch64" ;
make[4]: Entering directory '/usr/build/builds/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/openvswitch-2.17.0'
make  all-recursive
make[5]: Entering directory '/usr/build/builds/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/openvswitch-2.17.0'
Making all in datapath
make[6]: Entering directory '/usr/build/builds/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/openvswitch-2.17.0/datapath'
make[7]: Entering directory '/usr/build/builds/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/openvswitch-2.17.0/datapath'
make[7]: Leaving directory '/usr/build/builds/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/openvswitch-2.17.0/datapath'
make[6]: Leaving directory '/usr/build/builds/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/openvswitch-2.17.0/datapath'
make[6]: Entering directory '/usr/build/builds/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/openvswitch-2.17.0'
an.tmac:lib/ovs-fields.7:762: warning: tbl preprocessor failed, or it or soelim was not run; table(s) likely not rendered (TE macro called with TW register undefined)
troff:vswitchd/ovs-vswitchd.8:1290: warning: cannot select font 'CW'
make[6]: *** [Makefile:5684: manpage-check] Error 1
make[6]: Leaving directory '/usr/build/builds/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/openvswitch-2.17.0'
make[5]: *** [Makefile:4431: all-recursive] Error 1
make[5]: Leaving directory '/usr/build/builds/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/openvswitch-2.17.0'
make[4]: *** [Makefile:2333: all] Error 2
make[4]: Leaving directory '/usr/build/builds/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/openvswitch-2.17.0'
make[3]: *** [Makefile:280: /usr/build/builds/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/openvswitch-2.17.0/.built] Error 2
make[3]: Leaving directory '/usr/build/builds/openwrt/feeds/packages/net/openvswitch'
time: package/feeds/packages/openvswitch/compile#1.44#0.37#1.68
    ERROR: package/feeds/packages/openvswitch failed to build.
make[2]: *** [package/Makefile:120: package/feeds/packages/openvswitch/compile] Error 1

```

best regards,
Patryk

brada4, 2023-09-03:
Disable building man pages that do not reach final packages?

g-branden-robinson, 2023-09-03:
Hi there,

I work on `groff` upstream and cloned and built `ovs` to have a look at this issue.

Downgrading `groff` might be an overreaction.  The `man` package and the formatter are trying to tell you about actual problems with the man pages.

```
an.tmac:lib/ovs-fields.7:762: warning: tbl preprocessor failed, or it or soelim was not run; table(s) likely not rendered (TE macro called with TW register undefined)
```

The usual cause of this diagnostic is someone forgetting to run the `tbl` preprocessor, or equivalently forgetting to pass `groff` the `-t` option so that it will do so itself.  I see that [the `ovs-fields.7` man page is constructed via an elaborate process](https://github.com/openvswitch/ovs/blob/bc79a7bf033fa4cda8ccfc5481db3cfccd72650c/lib/automake.mk#L683) and it is possible that the Python tooling involved is neglecting to do something that it should.

```
troff:vswitchd/ovs-vswitchd.8:1290: warning: cannot select font 'CW'
```

This problem is similar.  `groff` 1.23.0 is [throwing a diagnostic now where it did not before, but not formatting things any differently](https://git.savannah.gnu.org/cgit/groff.git/commit/?id=1986da1d4bb11dc0421e004b153729b3d2a2a3ca).  This man page [uses `so` requests extensively](https://github.com/openvswitch/ovs/blob/master/vswitchd/ovs-vswitchd.8.in#L98), and if the "sourced" man page fragments were to contain `tbl` tables, the `soelim` preprocessor *must* be run or the page will not be formatted properly and as of `groff` 1.23.0, you'd get warnings like the first one above (about the `TW` register).

From `groff`'s perspective, these are only warnings.  It does not and will not exit with an error status because they are thrown.  My surmise is that whatever is trying to validate these man pages is exiting with an error status if `groff` (or, possibly, GNU `troff` if run directly by something managing the formatting pipeline itself as was done with AT&T `troff` in the 1980s) writes anything to the standard error stream.  If that is the case, I am not sure it is the right choice, just as the `-Werror` option to GCC is not always warranted.

My advice is to try to get these problems into the hands of the people who maintain the man page sources.  [`groff` 1.23.0 features over 400 bug fixes](https://savannah.gnu.org/bugs/index.php?go_report=Apply&group=groff&func=&set=custom&msort=0&report_id=225&advsrch=0&bug_id=&summary=&submitted_by=0&resolution_id=1&assigned_to=0&bug_group_id=0&status_id=3&severity=0&category_id=0&plan_release_id=103&history_search=0&history_field=0&history_event=modified&history_date_dayfd=3&history_date_monthfd=9&history_date_yearfd=2023&chunksz=50&spamscore=5&boxoptionwanted=1#options), several of these affecting man page rendering.

g-branden-robinson, 2023-09-03:
[This commit to openvswitch](https://github.com/openvswitch/ovs/commit/6180fefa835c7cad36e89f77f3d9de13c680fb88) might be of interest.

trippleflux, 2023-09-11:
Managed to compiled openvswitch 2.17.7 for x86_64 target with  @g-branden-robinson pointer and my modified previous openvswitch patch : 
**_0007-build-only-link-libopenvswitch-with-libunwind-libunb.patch_** 
```
diff -Naur a/lib/automake.mk b/lib/automake.mk
--- a/lib/automake.mk
+++ b/lib/automake.mk
@@ -10,6 +10,8 @@
 lib_libopenvswitch_la_LIBADD = $(SSL_LIBS)
 lib_libopenvswitch_la_LIBADD += $(CAPNG_LDADD)
 lib_libopenvswitch_la_LIBADD += $(LIBBPF_LDADD)
+lib_libopenvswitch_la_LIBADD += $(LIBUNBOUND_LDADD)
+lib_libopenvswitch_la_LIBADD += $(LIBUNWIND_LDADD)
 
 
 if WIN32
diff -Naur a/lib/libopenvswitch.pc.in b/lib/libopenvswitch.pc.in
--- a/lib/libopenvswitch.pc.in
+++ b/lib/libopenvswitch.pc.in
@@ -7,5 +7,5 @@
 Description: Open vSwitch library
 Version: @VERSION@
 Libs: -L${libdir} -lopenvswitch
-Libs.private: @LIBS@ @SSL_LIBS@ @CAPNG_LDADD@ @LIBBPF_LDADD@
+Libs.private: @LIBS@ @SSL_LIBS@ @CAPNG_LDADD@ @LIBBPF_LDADD@ @LIBUNBOUND_LDADD@ @LIBUNWIND_LDADD@
 Cflags: -I${includedir}
diff -Naur a/m4/openvswitch.m4 b/m4/openvswitch.m4
--- a/m4/openvswitch.m4
+++ b/m4/openvswitch.m4
@@ -735,7 +735,8 @@
   [AC_CHECK_LIB(unbound, ub_ctx_create, [HAVE_UNBOUND=yes], [HAVE_UNBOUND=no])
    if test "$HAVE_UNBOUND" = yes; then
      AC_DEFINE([HAVE_UNBOUND], [1], [Define to 1 if unbound is detected.])
-     LIBS="$LIBS -lunbound"
+     LIBUNBOUND_LDADD="-lunbound"
+	 AC_SUBST(LIBUNBOUND_LDADD)
    fi
    AM_CONDITIONAL([HAVE_UNBOUND], [test "$HAVE_UNBOUND" = yes])
    AC_SUBST([HAVE_UNBOUND])])
@@ -747,7 +748,8 @@
    [HAVE_UNWIND=no])
    if test "$HAVE_UNWIND" = yes; then
      AC_DEFINE([HAVE_UNWIND], [1], [Define to 1 if unwind is detected.])
-     LIBS="$LIBS -lunwind"
+     LIBUNWIND_LDADD="-lunwind"
+     AC_SUBST(LIBUNWIND_LDADD)
    fi
    AM_CONDITIONAL([HAVE_UNWIND], [test "$HAVE_UNWIND" = yes])
    AC_SUBST([HAVE_UNWIND])])
```

**_0008-groff-fix.patch_**
```
diff -Naur a/Makefile.am b/Makefile.am
--- a/Makefile.am
+++ b/Makefile.am
@@ -368,7 +368,7 @@
 manpage-check: $(man_MANS) $(dist_man_MANS) $(noinst_man_MANS)
 	@error=false; \
 	for manpage in $?; do \
-	  LANG=en_US.UTF-8 groff -w mac -w delim -w escape -w input -w missing -w tab -T utf8 -man -p -z $$manpage >$@.tmp 2>&1; \
+	  LANG=en_US.UTF-8 groff -t -w mac -w delim -w escape -w input -w missing -w tab -T utf8 -man -p -z $$manpage >$@.tmp 2>&1; \
 	  if grep warning: $@.tmp; then error=:; fi; \
 	  rm -f $@.tmp; \
 	done; \
```

Feel free to open a PR based on above.

neocturne, 2023-09-21:
I don't know about the
```
an.tmac:lib/ovs-fields.7:762: warning: tbl preprocessor failed, or it or soelim was not run; table(s) likely not rendered (TE macro called with TW register undefined)
```
error, I haven't seen that one myself (Arch Linux, groff 1.23.0-5).

```
troff:vswitchd/ovs-vswitchd.8:1290: warning: cannot select font 'CW'
```
has been fixed upstream in https://github.com/openvswitch/ovs/commit/0945e1a5fa3d66749facd24f8c60f732f7220f11. I've opened #22161, #22162 and #22163 to backport the commit to all maintained branches of openwrt/packages.

g-branden-robinson, 2023-09-22:
> I don't know about the
> 
> ```
> an.tmac:lib/ovs-fields.7:762: warning: tbl preprocessor failed, or it or soelim was not run; table(s) likely not rendered (TE macro called with TW register undefined)
> ```
> 
> error, I haven't seen that one myself (Arch Linux, groff 1.23.0-5).

[Colin Watson provided a patch for this issue.](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1042358#37)

Here's a [link directly to the relevant diff](https://bugs.debian.org/cgi-bin/bugreport.cgi?att=6;bug=1042358;filename=v2-0005-docs-Run-tbl-preprocessor-in-manpage-check-rule.patch;msg=37).

neocturne, 2023-09-23:
> [Colin Watson provided a patch for this issue.](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1042358#37)
>
> Here's a [link directly to the relevant [...]

I've now also tried on Debian Testing, still couldn't reproduce it... I guess we could still just include that patch, as it's both in OVS master and in Debian?

neocturne, 2023-09-23:
> I've now also tried on Debian Testing, still couldn't reproduce it... I guess we could still just include that patch, as it's both in OVS master and in Debian?

Ah, my Debian image didn't even contain groff, so the failing check was skipped... testing again.

neocturne, 2023-09-23:
I was able to reproduce the issue on Debian Testing after all, and it turned out that with the patch that @g-branden-robinson linked, the build would then fail with another manpage check warning (which is why Debian added 4 more patches I assume).

In the end I decided that adding several patches just to fix the manpages we don't even install was getting a bit silly, so I replaced my PRs with this one: #22194

neocturne, 2023-09-26:
@patrykk Do you want to give the PR a try, or should we just merge it?

patrykk, 2023-09-26:
> @patrykk Do you want to give the PR a try, or should we just merge it?

Hi, If you already have solution please go with it and supply patch and merge it. I haven't had time to test it,

PR #22194 (2023-09-27): openvswitch: disable groff manpage check
Maintainer: @yousong 
Compile tested: x86-64, OpenWrt master (d1521889066ce9698187ae56111cf591b09c10c5)
Run tested: Not run tested. This is a build fix that does not affect runtime.

Description:

The openvswitch build trips over a number of warnings during the manpage-check step if groff 1.23 is installed on the build host, resulting in a failed build.

As this check is optional, and we don't even install the manpages, simply override the groff configure check to never detect groff.

Supersedes #22161, as it turned out that several additional patches w [... cut]
closed 2023-09-27 by neocturne