openwrt/packages #22003: Tailscale: 23.05 will be out with an old version

avrutg, 2023-09-02:
Maintainer: @ja-pa 

Description: Tailscale in the 23.05 branch is outdated. I got the impression, that it can't be updated further, because it requires a newer golang version. That's really bad. 23.05 will be out with an already outdated version and by the time the next openwrt version comes out, it will be heavily outdated. Unlike some other projects, tailscale has no LTS version, so it means that it won't get even minor updates. The situation is really bad. It seems that a program which requires open WAN ports should not be left outdated. Isn't there something that can be done?

brada4, 2023-09-03:
Last 3 versions have QA issues with ubuntu. Maybe old version is better to avoid upstream fast  forward quality issues?

BKPepe, 2023-09-03:
What are the reasons to have it updated to the latest version except the urge to have the need not to have an outdated version? Because from your post, we are not doing a great job and it is our fault that tailscale in OpenWrt 23.05 does not have the version you need. 

Since you pointed out those LTS versions, you can ask Tailscale developers if they want to distinguish their stable versions and LTS as well. Currently, from our side, we can not update Golang (in techy terms: backport) in 23.05 since many packages in the main branch were not updated to support that recent update of Go, so they are failing right now. We appreciate any help you can give us regarding that, and as always, pull requests are welcome. :)

1715173329, 2023-09-03:
As a commercial software, their developers don't even consider compatibility issues, so who else cares ┑(￣Д ￣)┍
I'd say there's nothing fault on OpenWrt side, ask Tailscale for update, or just use Zerotier.
Of course they also offer static build binaries, just get from there: https://pkgs.tailscale.com/stable/#static

avrutg, 2023-09-03:
> Because from your post, we are not doing a great job and it is our fault that tailscale in OpenWrt 23.05 does not have the version you need.

I surely didn't mean that, I'm sorry if it came that way

> What are the reasons to have it updated to the latest version except the urge to have the need not to have an outdated version?

Currently the biggest thing is that newer versions have native nftables support, whereas OpenWRT has to resort to patching. I'm also afraid of breaking changes occuring on their side, preventing the use of older versions, since I plan on leaving my OpenWRT installation unattended for some months

> Since you pointed out those LTS versions, you can ask Tailscale developers if they want to distinguish their stable versions and LTS as well.

I would be glad if they did :(

closed 2023-11-04 by BKPepe

qral, 2024-03-04:
BTW you can update via app itself `# tailscale update`.