openwrt/packages #22006: Migration from PCRE to PCRE2

BKPepe, 2023-09-03:
Due to recent findings described in https://github.com/openwrt/packages/pull/21800, we should track somewhere (= here) packages which are still using PCRE instead of PCRE2 to propose their developers to look into PCRE2 and prepare the support (Assignee: Here maintainers of relevant packages).
_____
**Reasons to switch to PCRE - Perl Compatible Regular Expressions 2:**
- PCRE2 was first released in 2015 to replace the API in the original PCRE library, which is now obsolete and no longer maintained.   [^1] [^2]
- Since OpenWrt targets to be minimalistic, there is no reason to have installed these two libraries side by side on the device, which results to be bigger and it may not fit to devices with small storage.  [^3]

[^1]: https://github.com/PCRE2Project/pcre2#pcre2---perl-compatible-regular-expressions
[^2]: https://www.pcre.org
[^3]: https://openwrt.org/toh/recommended_routers
______
**Packages, which needs to be taken care of:**
- [ ] aircrack-ng
**Maintainer:** @ZeroChaos- even though I though this package is rather not maintained
**Steps to be done**: 
1. Ask aircrack-ng developer to release a new version with PCRE2 as those changes were merged to the master branch (See PR: https://github.com/aircrack-ng/aircrack-ng/pull/2391 ) 
2. Seek for new maintainer for this package in OpenWrt or if not found, we should decide if we want to keep this package here or not.
- [ ] atftp
**Maintainer:** @dddaniel
**Steps to be done:**
1. Backport patch from https://sourceforge.net/p/atftp/support-requests/11/
2. Someone should step it and take maintainership of this package here.
- [ ] net-snmp
**Maintainer:** @stintel 
**Steps to be done:**
1. Needs to be investigated further
- [ ] shadowsocks-libev
**Maintainer:** @yousong 
**Steps to be done:**
Remove this package, because there were efforts to port it to PCRE2 and it is not merged and as well this package is getting to be removed from Debian. See: https://github.com/shadowsocks/shadowsocks-libev/issues/1788
- [ ] uwsgi
**Maintainer:** @Ansuel 
**Steps to be done:**
_Oh this time, its different! There is pending pull request:_
https://github.com/unbit/uwsgi/pull/2543
- [ ] nmap
**Maintainer:** @nunojpg 
**Steps to be done:**
There is support since commit https://github.com/nmap/nmap/commit/828ab48764b82d0226e860c73c5dac5b11f77385 , no release was tagged since that time
- [ ] apache
**Maintainer:** @heil
**Steps to be done:**
1. Is anyone interested to take maintainership of this package?
2. Support is already there https://svn.apache.org/viewvc?view=revision&revision=1898399 , so we need to bump this package and switch the library in the Makefile
- [ ] kismet
**Maintainer:** @padre-lacroix and @sourceindex
**Steps to be done:**
This package is not maintained in our repository since it was introduced here even though upstream are releasing new versions
- [ ] mg
**Maintainer:** @nxhack 
**Steps to be done:**
1. Needs to be investigated further
- [ ] bandwidthd
**Maintainer:** @padre-lacroix 
**Steps to be done:**
Upstream repository is archived and read only since February 2022. See: https://github.com/NethServer/bandwidthd
1. **Remove this package.**
- [ ] zsh
**Maintainer:** @msva
**Steps to be done:** 
There is support in upstream repository based on https://sourceforge.net/p/zsh/code/ci/10bdbd8b5b0b43445aff23dcd412f25cf6aa328a/
- [ ] slang2
**Maintainer:** @jefferyto 
**Steps to be done:**
1. Needs to be investigated further
- [ ] czmq
**Maintainer:** @ja-pa 
**Steps to be done:**
1. Needs to be investigated further

- [ ] micropython-lib
**Maintainer:** @jefferyto 
**Steps to be done:**
1. Needs to be investigated further
- [ ] nginx
**Maintainer:** @Ansuel and @heil 
**Steps to be done:** 
1. Since version 1.12.5 is using pcre2 by default according to https://nginx.org/en/CHANGES , but still in the Makefile, there is legacy pcre being used.

I ask anyone to be responsible and look into it, if possible and as well rather sooner than later. Otherwise, if there is no respond to the package, which you are maintaing the package will be removed in the beginning of the next month - **1st October 2023, so we have soft deadline.** I am fully aware that if there is no deadline, then no further actions will be taken care. You are not doing it just for me, but also for others including OpenWrt users.

Thank you for your cooperation guys and make sure that it is run tested on the router! ;)

1715173329, 2023-09-03:
>  shadowsocks-libev

I guess it's time to switch to [shadowsocks-rust](https://github.com/shadowsocks/shadowsocks-rust), which has much better performance (10x faster on aarch64 platform).

stintel, 2023-09-03:
https://github.com/net-snmp/net-snmp/issues/420

BKPepe, 2023-09-03:
Thanks, @stintel, I included your referenced issue to OP.

padre-lacroix, 2023-09-06:
For the package kismet, I originally become the maintainer to prevent this package to be abandoned.
But, I do not wish to maintain it anymore, mostly because this is a package that I do not use and also because I have no way to test it.
If somebody wants to become the maintainer, then I am OK with that, otherwise it can be dropped.

jefferyto, 2023-09-10:
slang2 is done in #22020.

neheb, 2023-09-21:
I looked into kismet before. seems to now be a convoluted project with protocol buffers and a bunch of other dependencies.

Ansuel, 2023-09-21:
@BKPepe is it ok to include patch while we wait for a release? (for pcre2 support)

nxhack, 2023-09-22:
@BKPepe mg editor done.

BKPepe, 2023-09-22:
> I looked into kismet before.

@neheb The question is do we care about quality over quantity?
Many packages here are not maintained for various reasons. Almost everyone can add a new package to this repository, no matter what. No votes, if it has purpose or if it makes any sense. I know it will sound bad, but I would rather remove those packages, which could be updated by upstream but not here. This should hopefully result in end users or someone else complaining that the package is missing within OpenWrt and that they can help us. I'm not saying that there are not enough maintainers. We (existing ones) are doing it for some purpose. We care about it. We are trying to update it, we want to have the latest versions on our routers, etc.

I might be strict, but I would simply remove those packages. You know, it is very pleasant of you that you will take into kismet, but how many users are using it?

I fail to see such unmaintained packages here: https://downloads.openwrt.org/stats/awstats.downloads.openwrt.org.allextra2.html

Is anyone really using them? If we are doing that for a minority <5-10 users then it is probably not worth all the efforts we are putting into it. They can compile themselves. They can use GitHub Actions (simply by forking this repository), if they are not familiar with the OpenWrt build system and from the CI, download the package and install it or even prepare their own feed.

With the removing these packages, we will as someone from OpenWrt core team is saying: "Less using compute resources, faster build time, happy users"

BKPepe, 2023-09-22:
> @BKPepe is it ok to include patch while we wait for a release? (for pcre2 support)

You have there two packages, where you are listed as maintainer. We can probably backport uwsgi pending pull request as it seems that for 3 weeks, there has been no activity. Regarding nginx, it can be done immediately because PCRE2 support is there for many weeks.

Ansuel, 2023-09-22:
Was talking about other packages since the deadline is 1 October. For nginx
I will check today what are the required thing to do. The blocker was lua
support

neheb, 2023-09-22:
kismet can go away. I don’t use it.

padre-lacroix, 2023-09-22:
I am the maintainer of kismet and I do not use it myself: as I mentioned in a previous message, I took it to prevent at the time (2015-2016) that it will disappear.

So, kismet can disappear as far as I am concerned: I cannot and will not maintain it as I cannot even test it or use it myself.

oskarirauta, 2023-09-23:
@msva

Proposal for zsh libpcre2 usage: #22197

BKPepe, 2023-09-28:
It seems that we do have a lot of progress here! Glad to see it, and thanks anyone, who looked into it.

- Unfortunately, I am not sure what to do with ``Aircrack-ng``.
  - While looking at the link, which I already put here (https://downloads.openwrt.org/stats/awstats.downloads.openwrt.org.allextra2.html), it seems that ``Aircrack-ng`` was downloaded 146 times. It wouldn't be great to remove it. I asked the developers if they could release a new version because it would help us instead of backporting patches, using the master branch, and giving it to users, etc.
     - We still need to seek for a new maintainer here.

trippleflux, 2023-09-28:
@BKPepe 

  [Trying ](https://github.com/openwrt/packages/issues/22240)my best about Aircrack-ng, hmm not to be sound rude but the bad news is I don't yet feel the like or knowledgeable enough in aircrack-ng to maintain or use it.

Ansuel, 2023-09-28:
@BKPepe https://github.com/micropython/micropython-lib/pull/737 wasted way too much time on this but eheheheh

Ansuel, 2023-09-30:
Well tomorrow is the deadline. I would drop pcre library and mark the remaining package as `@BROKEN ` while we wait for the remaining package to be update. For the one where we don't have any info I would move them to package-abbandoned
https://github.com/openwrt/packages-abandoned

stintel, 2023-09-30:
> Well tomorrow is the deadline. I would drop pcre library and mark the remaining package as `@BROKEN` while we wait for the remaining package to be update. For the one where we don't have any info I would move them to package-abbandoned https://github.com/openwrt/packages-abandoned

Please don't. You can't seriously consider dropping net-snmp from a distribution that mainly targets network devices.

stintel, 2023-09-30:
> @stintel if you want I can waste also today on migrating another package to pcre2.

The upstream issue hasn't seen any updates since January. Don't see any references to a branch where this might be worked on, and no idea how much work it's going to be. I currently don't have the time to dig in myself.

But dropping net-snmp would be a bad thing. I would actually consider finding an alternative to OpenWrt if that happens.

Ansuel, 2023-09-30:
@stintel well this is main/master so it's expected to have some package broken for some time. These changes won't affect stable...

Aside from this... I'm checking the package with a quick grep and I can see there is actual support for pcre2...
https://github.com/net-snmp/net-snmp/commit/d3e95c87b32397815f6d5bcfc844259f2552697a

Let me prepare a pr and see how it goes.

stintel, 2023-09-30:
> Aside from this... I'm checking the package with a quick grep and I can see there is actual support for pcre2...
> [net-snmp/net-snmp@d3e95c8](https://github.com/net-snmp/net-snmp/commit/d3e95c87b32397815f6d5bcfc844259f2552697a)

Nice find! Tag me in a PR and I can make some time to do some thinking. Thanks for looking at this!

BKPepe, 2023-10-08:
As you can see, I put in a few extra days over the deadline. I'm just joking. I didn't look at it recently. I see that everyone did a great job over here. Mostly @Ansuel, who was working on it. Appreciated! 👏 

Maybe some of you wouldn't like me, that's understandable, but most of those switches to PCRE2 were backported to the stable branches. You know, it is quite simple on this one. We would not get rid of the deprecated PCRE really fast, and it would be quite boring once the master is ready, but stable releases are not. This helps us that users can report it to us and we can really fast investigate it and fix it. OpenWrt does not release quite often as I was used to do with Turris OS (yep... they are still based on OpenWrt 21.02 so far with Turris OS 6.x. Thus some fixes landed there as well)

I am not using legacy PCRE on my router anymore, but some of you might. We are getting to the end. 🤞 There are just a few packages remaining to be moved/migrated to PCRE2. 

I haven't heard anything about ``shadowsocks-libev``, I will create the pull request to remove it from the master branch very soon.

Ansuel, 2023-10-08:
@BKPepe
- micropython is getting investigated (why I can't repro errors other reports...)
- zsh is ready
- nmap we need to wait for a new point release since adding a 4mb patch is too much

with shadowsocks-libev getting dropped i think pcre can be dropped.

Also consider that uwsgi is currently broken due to a problem with python and PATH but another member is working on fixing that so that should be handled indirectly. (also wasn't catch by CI test as they don't compile every sub package)

yousong, 2023-10-09:
@BKPepe @Ansuel Though not a user of the package myself, I tend to believe that shadowsocks-libev seems to be still used by quite a few people ([stats](https://downloads.openwrt.org/stats/awstats.downloads.openwrt.org.allextra2.html)).

Before an alternative is available, I will try to keep it here by modifying the source code to drop dependency on pcre.  Hopefully it's just acl support only recently enabled by @fertkir in #20647

Ansuel, 2023-10-09:
@yousong or... https://github.com/openwrt/packages/pull/22356

BKPepe, 2023-10-14:
Oh no, this did not make it to the final stable version of 23.05.0, never mind. So, it be 23.05.1.

Ansuel, 2023-10-14:
@BKPepe mhhh in theory we can still continue to backport packages and continue this. Just people will update their packages

crza, 2023-10-28:
Hi,

there still seem to be some packages with missing migration to pcre2, which makes my build for openwrt-23.05 branches fail:

- freeradius3-common
- snort3
- postfix with PCRE enabled

Will post / debug further tomorrow and update this post or create separate issues

stintel, 2023-10-29:
In addition to the comment above, http://lists.openwrt.org/pipermail/openwrt-devel/2023-October/041689.html suggests some more packages were missed.
* fdm
* haproxy
* libndpi
* privoxy
* tvheadend
* zabbix

Ansuel, 2023-10-29:
Oh no... More package to convert... Already converted postfix

crza, 2023-10-29:
> […](#)

Can someone backport to openwrt-23.05 after PR is pushed / merged?

crza, 2023-10-29:
@stintel according to https://www.zabbix.com/documentation/current/en/manual/installation/requirements#:~:text=libpcre3%27%20or%20%27libpcre1%27.-,PCRE%20v8.x%20and%20PCRE2%20v10.x%20(from%20Zabbix%206.0.0)%20are%20supported.,-libevent  zabbix 6.0+ already has pcre2 support, so maybe no patches are needed?

Ansuel, 2023-10-29:
Yep I will check each missing package. I just need the list

BKPepe, 2023-10-29:
Hmm, I think what I did wrong, uhh! Yep, it's just a few more packages.
Guys, I am so sorry for this! I am feeling bad, because of that. I will add
those packages in OP in a short manner.

Ansuel, 2023-10-29:
@BKPepe taking care of checking each suggested package and moving them... for now only 2 require special care (aka me having to convert the package I think...)

BKPepe, 2023-10-29:
> Can someone backport to openwrt-23.05 after PR is pushed / merged?

Hmm, I think this question is answered here in this thread already. ;) You might want to check also the stable branches. Of course, it will be backported over the time, no need to worry about it.

Ansuel, 2023-10-30:
@BKPepe did you find other package that still use pcre?

systemcrash, 2023-10-30:
freeswitch
kamailio
rtpengine
sipgrep
snort (not v3)

http://lists.openwrt.org/pipermail/openwrt-devel/2023-October/041689.html

BKPepe, 2023-10-30:
Dont confuse it with other feeds. They are on their own. This is only for
packages feed, where are the most packages. For other feeds something like
this needs to be created or they will notice that once libpcre is gone.

Ansuel, 2023-11-03:
@BKPepe @systemcrash Is the list now complete? only 3 package to do... freeswitch is ""easy"" and I will manually convert snor and snort3 is HELL ON EARTH.

systemcrash, 2023-11-03:
Those are the ones that pop up when I run make. Nothing further to add.

Ansuel, 2023-11-05:
Welp only snort3 is missing... might take a while as I won't be at pc in the next few days :(

Ansuel, 2023-11-07:
And with also snort3 manually converted, there isn't any more package that still require pcre.... Now it's really a matter of testing and merge the pending pr...

systemcrash, 2023-11-18:
How is this migration coming along? Any objections to merging these?

Ansuel, 2023-11-18:
@systemcrash nmap are backport for the other 3 package.. pr is open but no feedback from mainatiners

Ansuel, 2024-04-28:
One left :(

We should really spam or find a good way to make freeswitch guy aware of the deprecation.

BKPepe, 2024-04-28:
Yeah... only left.

Hmm, maybe we can convience telephony maintainers to merge your pending patch. :)
- Most of it, most of your work 👏 , was backported to OpenWrt 23.05 and as well OpenWrt 22.03, if it was possible.
   - I am thinking... we can force it by removing libpcre library. :P :D

Ansuel, 2024-04-28:
Honestly I totally get why he is a bit hesitant... they looks like big changes but reality is that it's the very basic handling of pcre2 (nothing fancy) I would wait a bit, I don't like forcing stuff especially when we are in the same org :D

systemcrash, 2024-04-28:
>       * I am thinking... we can force it by removing libpcre library. :P :D

Evil genius 😈

BKPepe, 2024-04-30:
Naaah, I am not. 😇 We have been sitting on PCRE for a longer period than I thought. It looked like an easy job in the beginning, but several open-source projects were not prepared, and it took some time for developers to look into it, review @Ansuel's work, and then, if possible, release a new version, so we don't need to carry patches, which were quite huge.

Keeping this only for one remaining software... I am kinda convinced to drop it; it's the master branch after all, issues are expected, and I am quite sure that some hidden issues will be revealed once PCRE is removed.

closed 2024-05-11 by BKPepe