openwrt/packages #22046: modemmanager: why are we shipping udev rules?

pprindeville, 2023-09-05:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @mips171 
Environment: x86_64, generic, HEAD

Description:

Looking at the `net/modemmanager/Makefile` I'm seeing:

```
        $(INSTALL_DIR) $(1)/lib/udev/rules.d
        $(INSTALL_DATA) $(PKG_INSTALL_DIR)/lib/udev/rules.d/*.rules $(1)/lib/udev/rules.d
```

and I'm wondering, what use do the `udev` rules do?

feckert, 2023-09-05:
Modemmanager could run without udev. But the rules are still needed to register the right modem with the right interface in the modem manager. .
Please have a look at the following [changes](https://gitlab.freedesktop.org/aleksm/ModemManager/-/commit/58c955f5f23e874e4f8c2a4b389e46c0775e7f07) made in the modemmanger to work without udev service.

feckert, 2023-09-13:
I have found additional information about your question on them Modemmanager page.
https://modemmanager.org/docs/modemmanager/port-and-device-detection/#hotplug-in-openwrt

closed 2023-09-13 by feckert