openwrt/packages #22052: LuaJIT build fail

zcy85611, 2023-09-05:
I did nothing to the code, but fail on compilation. it was fine on yesterday's compilation, would anyone help to fix it please?
```
make[3]: Entering directory '/home/openwrt/feeds/packages/lang/luajit'
rm -f /home/openwrt/build_dir/target-mipsel_24kc_musl/LuaJIT-2.1.0-beta3/.built
touch /home/openwrt/build_dir/target-mipsel_24kc_musl/LuaJIT-2.1.0-beta3/.built_check
make -j1 -C /home/openwrt/build_dir/target-mipsel_24kc_musl/LuaJIT-2.1.0-beta3 HOST_CC="/home/openwrt/staging_dir/host/bin/gcc -O2 -I/home/openwrt/staging_dir/host/include -I/home/openwrt/staging_dir/hostpkg/include -I/home/openwrt/staging_dir/target-mipsel_24kc_musl/host/include -m32" CROSS="mipsel-openwrt-linux-musl-" DPREFIX=/home/openwrt/build_dir/target-mipsel_24kc_musl/LuaJIT-2.1.0-beta3/ipkg-install/usr PREFIX=/usr TARGET_SYS=Linux TARGET_CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/home/openwrt/build_dir/target-mipsel_24kc_musl/LuaJIT-2.1.0-beta3=LuaJIT-2.1.0-beta3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" BUILDMODE=dynamic
make[4]: Entering directory '/home/openwrt/build_dir/target-mipsel_24kc_musl/LuaJIT-2.1.0-beta3'
==== Building LuaJIT 2.1.0-beta3 ====
make -C src
make[5]: Entering directory '/home/openwrt/build_dir/target-mipsel_24kc_musl/LuaJIT-2.1.0-beta3/src'
HOSTCC    host/minilua.o
In file included from /usr/lib/gcc/x86_64-linux-gnu/11/include/limits.h:203,
                 from /usr/lib/gcc/x86_64-linux-gnu/11/include/syslimits.h:7,
                 from /usr/lib/gcc/x86_64-linux-gnu/11/include/limits.h:34,
                 from host/minilua.c:33:
/usr/include/limits.h:26:10: fatal error: bits/libc-header-start.h: No such file or directory
   26 | #include <bits/libc-header-start.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

BKPepe, 2023-09-05:
I think, you are missing on the host this dependency - ``gcc-multilib``

zcy85611, 2023-09-05:
> I think, you are missing on the host this dependency - `gcc-multilib`

thanks for reply, gcc-multilib g++-multilib, I have both installed on the host

closed 2023-09-05 by zcy85611