openwrt/packages #22058: SoftetherVPN: Unable to get connection while using Azure VPN

solomonricky, 2023-09-06:
Maintainer: @fededim @neheb 
Environment: 
 **Server Side**
- NanoPi R4s (OpenWrt SNAPSHOT r23018-3b76f6eee4)
- Xiaomi CR6609 (OpenWrt SNAPSHOT r23859-28bd578178)
- Ubuntu 22.04 

 **Client Side**

- Windows 10 22H2 (SoftetherVPN Adapter 4.25.0.9658, SoftetherVPN VPN Client Build 9787)

Description:
Unable to get connection while using Azure VPN, which does not need Port Forwarding and Public IP Address. I tried to use Static IP Address but failed to connect, too. 

What did I set:
Basically, normal setup for SoftetherVPN Server when comes to the Local Bridge Setting Part, I tried to use **Bridge with Physical Existing Network Adapter** and **Bridge with New Tap Device**, bridge it with **br-lan** in OpenWrt, but no luck both way unable to connect. With the same method, I set up in Ubuntu 22.04 it works, successfully getting IP Address from DHCP. 

For reference, the Wireshark log is provided in comments.

solomonricky, 2023-09-06:
Seems like DHCP from OpenWrt does not OFFER any IP Address
```
No.     Time           Source                Destination           Protocol Length Info
      1 0.000000       5e:f0:ee:dd:22:e6     LLDP_Multicast        LLDP     58     MA/5e:f0:ee:dd:22:e6 MA/5e:f0:ee:dd:22:e6 3601 

Frame 1: 58 bytes on wire (464 bits), 58 bytes captured (464 bits) on interface \Device\NPF_{075465C9-F25A-47CA-8D68-0DFB05127844}, id 0
Ethernet II, Src: 5e:f0:ee:dd:22:e6 (5e:f0:ee:dd:22:e6), Dst: LLDP_Multicast (01:80:c2:00:00:0e)
Link Layer Discovery Protocol

No.     Time           Source                Destination           Protocol Length Info
      2 0.010592       0.0.0.0               255.255.255.255       DHCP     342    DHCP Discover - Transaction ID 0x9aa0e663

Frame 2: 342 bytes on wire (2736 bits), 342 bytes captured (2736 bits) on interface \Device\NPF_{075465C9-F25A-47CA-8D68-0DFB05127844}, id 0
Ethernet II, Src: 5e:f0:ee:dd:22:e6 (5e:f0:ee:dd:22:e6), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
Internet Protocol Version 4, Src: 0.0.0.0, Dst: 255.255.255.255
User Datagram Protocol, Src Port: 68, Dst Port: 67
Dynamic Host Configuration Protocol (Discover)

No.     Time           Source                Destination           Protocol Length Info
      3 0.038074       fe80::a6ce:2cf9:230e:23e8 ff02::1:2             DHCPv6   145    Solicit XID: 0x333ae1 CID: 000100012c0709812cf05d659004 

Frame 3: 145 bytes on wire (1160 bits), 145 bytes captured (1160 bits) on interface \Device\NPF_{075465C9-F25A-47CA-8D68-0DFB05127844}, id 0
Ethernet II, Src: 5e:f0:ee:dd:22:e6 (5e:f0:ee:dd:22:e6), Dst: IPv6mcast_01:00:02 (33:33:00:01:00:02)
Internet Protocol Version 6, Src: fe80::a6ce:2cf9:230e:23e8, Dst: ff02::1:2
User Datagram Protocol, Src Port: 546, Dst Port: 547
DHCPv6

No.     Time           Source                Destination           Protocol Length Info
      4 0.242304       5e:f0:ee:dd:22:e6     Broadcast             ARP      42     Who has 169.254.104.195? (ARP Probe)

Frame 4: 42 bytes on wire (336 bits), 42 bytes captured (336 bits) on interface \Device\NPF_{075465C9-F25A-47CA-8D68-0DFB05127844}, id 0
Ethernet II, Src: 5e:f0:ee:dd:22:e6 (5e:f0:ee:dd:22:e6), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
Address Resolution Protocol (ARP Probe)

No.     Time           Source                Destination           Protocol Length Info
      5 0.242376       169.254.104.195       224.0.0.22            IGMPv3   62     Membership Report / Join group 224.0.0.251 for any sources / Join group 239.255.255.250 for any sources

Frame 5: 62 bytes on wire (496 bits), 62 bytes captured (496 bits) on interface \Device\NPF_{075465C9-F25A-47CA-8D68-0DFB05127844}, id 0
Ethernet II, Src: 5e:f0:ee:dd:22:e6 (5e:f0:ee:dd:22:e6), Dst: IPv4mcast_16 (01:00:5e:00:00:16)
Internet Protocol Version 4, Src: 169.254.104.195, Dst: 224.0.0.22
Internet Group Management Protocol

No.     Time           Source                Destination           Protocol Length Info
      6 0.242447       ::                    ff02::1:ff0e:23e8     ICMPv6   78     Neighbor Solicitation for fe80::a6ce:2cf9:230e:23e8

Frame 6: 78 bytes on wire (624 bits), 78 bytes captured (624 bits) on interface \Device\NPF_{075465C9-F25A-47CA-8D68-0DFB05127844}, id 0
Ethernet II, Src: 5e:f0:ee:dd:22:e6 (5e:f0:ee:dd:22:e6), Dst: IPv6mcast_ff:0e:23:e8 (33:33:ff:0e:23:e8)
Internet Protocol Version 6, Src: ::, Dst: ff02::1:ff0e:23e8
Internet Control Message Protocol v6

No.     Time           Source                Destination           Protocol Length Info
      7 0.242476       fe80::a6ce:2cf9:230e:23e8 ff02::2               ICMPv6   62     Router Solicitation

Frame 7: 62 bytes on wire (496 bits), 62 bytes captured (496 bits) on interface \Device\NPF_{075465C9-F25A-47CA-8D68-0DFB05127844}, id 0
Ethernet II, Src: 5e:f0:ee:dd:22:e6 (5e:f0:ee:dd:22:e6), Dst: IPv6mcast_02 (33:33:00:00:00:02)
Internet Protocol Version 6, Src: fe80::a6ce:2cf9:230e:23e8, Dst: ff02::2
Internet Control Message Protocol v6

No.     Time           Source                Destination           Protocol Length Info
      8 0.242545       fe80::a6ce:2cf9:230e:23e8 ff02::16              ICMPv6   110    Multicast Listener Report Message v2

Frame 8: 110 bytes on wire (880 bits), 110 bytes captured (880 bits) on interface \Device\NPF_{075465C9-F25A-47CA-8D68-0DFB05127844}, id 0
Ethernet II, Src: 5e:f0:ee:dd:22:e6 (5e:f0:ee:dd:22:e6), Dst: IPv6mcast_16 (33:33:00:00:00:16)
Internet Protocol Version 6, Src: fe80::a6ce:2cf9:230e:23e8, Dst: ff02::16
Internet Control Message Protocol v6

No.     Time           Source                Destination           Protocol Length Info
      9 0.336942       0.0.0.0               255.255.255.255       DHCP     342    DHCP Discover - Transaction ID 0xe87ed7ff

Frame 9: 342 bytes on wire (2736 bits), 342 bytes captured (2736 bits) on interface \Device\NPF_{075465C9-F25A-47CA-8D68-0DFB05127844}, id 0
[... 1266 lines cut ...]
Multicast Domain Name System (query)

No.     Time           Source                Destination           Protocol Length Info
    153 44.001202      169.254.104.195       239.255.255.250       SSDP     167    M-SEARCH * HTTP/1.1 

Frame 153: 167 bytes on wire (1336 bits), 167 bytes captured (1336 bits) on interface \Device\NPF_{075465C9-F25A-47CA-8D68-0DFB05127844}, id 0
Ethernet II, Src: 5e:f0:ee:dd:22:e6 (5e:f0:ee:dd:22:e6), Dst: IPv4mcast_7f:ff:fa (01:00:5e:7f:ff:fa)
Internet Protocol Version 4, Src: 169.254.104.195, Dst: 239.255.255.250
[... cut ...]
Frame 154: 86 bytes on wire (688 bits), 86 bytes captured (688 bits) on interface \Device\NPF_{075465C9-F25A-47CA-8D68-0DFB05127844}, id 0
Ethernet II, Src: 5e:f0:ee:dd:22:e6 (5e:f0:ee:dd:22:e6), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
Internet Protocol Version 4, Src: 169.254.104.195, Dst: 169.254.255.255
User Datagram Protocol, Src Port: 57621, Dst Port: 57621
Data (44 bytes)

0000  53 70 6f 74 55 64 70 30 71 db 12 f5 d9 79 5b 04   SpotUdp0q....y[.
0010  00 01 00 04 48 95 c2 03 6e 69 91 af 38 dc bf a2   ....H...ni..8...
0020  1d 68 04 e0 79 70 2b 67 4f ad 16 6f               .h..yp+gO..o

```

brada4, 2023-09-14:
Do you receive DHCP solicitation in tun on Openwrt? On the br-lan? On the dnsmasq on that br-lan?
like `tcpdump -p -i potat0`

solomonricky, 2023-09-16:
Unfortunately no. The only log that I get is `ICMP6, neighbor solicitation, who has fe80::2ae:e0ff:fe00:aee0, length 32`. 

By right should get `ARP, Request who-has 192.168.5.247 tell CR6608.lan, length 28` and `IP CR6608.lan.67 > 192.168.5.247.68: BOOTP/DHCP, Reply, length 300` this right?

brada4, 2023-09-16:
Your assumption on how it shold function is correct. Anything windows send out to VPN tunnel must reach other end aka openwrt.