openwrt/packages #22059: openwrt 23.05 nginx webdav can't work

deneyson, 2023-09-06:
root@openwrt:~# nginx -T -c '/etc/nginx/uci.conf'
2023/09/06 20:14:29 [emerg] 9352#0: unknown directive "dav_ext_methods" in /etc/                                                                             nginx/conf.d/webdav.conf:26
nginx: configuration file /etc/nginx/uci.conf test fail
nginx-full installed,but did not work

sbwml, 2023-09-22:
nginx webdav works well.

/etc/nginx/conf.d/webdav-test.conf

```nginx
server {
    listen 5005;
    server_name _;
    root /home;

    location / {
        dav_methods PUT DELETE MKCOL COPY MOVE;
        dav_ext_methods PROPFIND OPTIONS;
        create_full_put_path on;
        client_max_body_size 0;
        dav_access user:rw group:rw all:r;
        #auth_basic "Restricted";
        #auth_basic_user_file /etc/nginx/.htpasswd;
    }
}
```

```log
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 23.05.0-rc3, r23389-5deed175a5
 -----------------------------------------------------
[root@OpenWrt ~]# nginx -t -c /etc/nginx/uci.conf
nginx: the configuration file /etc/nginx/uci.conf syntax is ok
nginx: configuration file /etc/nginx/uci.conf test is successful
[root@OpenWrt ~]# service nginx reload
[root@OpenWrt ~]#
```

[image]

sbwml, 2023-09-22:
you need to install nginx-mod-dav-ext package, starting from openwrt 23.05, nginx-full enables more extensions in the form of modules.

deneyson, 2023-09-28:
> you need to install `nginx-mod-dav-ext` package, starting from openwrt 23.05, nginx-full enables more extensions in the form of modules.

Thank you for your help.I had installed nginx-full nginx-mod-dav-ext nginx-ssl-util nginx-util and used your config to reload service,but still not working,So can you show me what packages you install.Thank you

sbwml, 2023-09-28:
☹ Sorry, I just realized a problem now, the nginx I'm using is not from openwrt-23.05. For some reason, the nginx I'm currently using belongs to the bump of the openwrt-22.03 branch.

So my answer above cannot be used as a reference for your situation.

[image]

my nginx-mod-dav-ext module belongs to builtin.

Sergey1560, 2023-10-03:
Install nginx-mod-dav-ext and add ngx_http_dav_ext_module.so to config:

root@OpenWrt:/etc/nginx/module.d$ cat luci.module 
load_module /usr/lib/nginx/modules/ngx_http_ubus_module.so;
load_module /usr/lib/nginx/modules/ngx_http_dav_ext_module.so;

deneyson, 2023-10-16:
root@Router-J4125:~# nginx -V
nginx version: nginx/1.25.0 (x86_64-pc-linux-gnu)
built with OpenSSL 3.0.11 19 Sep 2023
TLS SNI support enabled
configure arguments: --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --disable-dependency-tracking --program-prefix= --program-suffix= --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls --crossbuild=Linux::x86_64 --prefix=/usr --conf-path=/etc/nginx/nginx.conf --modules-path=/usr/lib/nginx/modules --with-compat --with-http_ssl_module --with-ipv6 --with-http_stub_status_module --with-http_flv_module --with-http_dav_module --with-http_auth_request_module --with-http_v2_module --with-http_realip_module --with-http_secure_link_module --with-http_sub_module --add-dynamic-module=/home/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.25.0/nginx-dav-ext-module --add-dynamic-module=/home/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.25.0/nginx-ubus-module --error-log-path=stderr --pid-path=/var/run/nginx.pid --lock-path=/var/lock/nginx.lock --http-log-path=/var/log/nginx/access.log --http-client-body-temp-path=/var/lib/nginx/body --http-proxy-temp-path=/var/lib/nginx/proxy --http-fastcgi-temp-path=/var/lib/nginx/fastcgi --with-cc=x86_64-openwrt-linux-musl-gcc --with-cc-opt='-I/home/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/usr/include -I/home/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/include/fortify -I/home/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.25.0=nginx-1.25.0 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNGX_LUA_NO_BY_LUA_BLOCK' --with-ld-opt='-L/home/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/usr/lib -L/home/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib -Wl,--gc-sections -znow -zrelro' --without-http_upstream_zone_module --without-pcre2
root@openwrt:~# nginx -T -c '/etc/nginx/uci.conf'
2023/09/06 20:14:29 [emerg] 9352#0: unknown directive "dav_ext_methods" in /etc/ nginx/conf.d/webdav.conf:26
This is my webdav config.
server {
    listen 81  ssl default_server;
    listen [::]:81  ssl default_server;
    ssl_certificate /etc/uhttpd.crt;
    ssl_certificate_key /etc/uhttpd.key;
    server_name webdav;
    charset utf-8;
    client_max_body_size 0;
    root /mnt/sda1;
    client_body_temp_path /mnt/sda1/temp/tmp;

    location /{
        index index.html index.htm;
        autoindex on;
        dav_methods PUT DELETE MKCOL COPY MOVE;
        dav_ext_methods PROPFIND OPTIONS;
        create_full_put_path on;
        dav_access user:rw group:rw all:r;
        auth_basic "Authorized Users Only";
        auth_basic_user_file /etc/nginx/webdavpasswd;
    }
    # return 302 https://$host$request_uri;
}
I use the new openwrt 23.05.0,and I install nginx-full1.25.0-4 nginx-mod-dav-ext1.25.0-4 nginx-mod-ubus1.25.0-4
nginx-ssl-util1.6-18 nginx-util1.6-18.
when I use chrome to visist the nas.It work perfect,But some app just like nPlayer a IOS'sAPP, It show the wrong code and said "method not allowed (Status 405).when I add the config "dav_ext_methods PROPFIND OPTIONS;"
It said unknown directive "dav_ext_methods" in /etc/ nginx/conf.d/webdav.conf:26.
I think the nginx-mod-dav-ext1.25.0-4 had a bug,but I am not the Developer I hope some one can fix because I love OPENWRT,THANK YOU.

closed 2023-12-12 by deneyson