openwrt/packages #22061: nebula-proto: Tunnel device in config file (nebula1 ) doesn't match interface name (nebula1)!

AveoCross, 2023-09-06:
Maintainer: @stangri 
Environment: ARMv8 Processor rev 4 / OpenWrt 22.03.5 r20134-5f15225c1e / LuCI openwrt-22.03 branch git-23.093.57104-ce20b4a
Package version: 1.6.1-1

Description:
```
Wed Sep  6 22:49:05 2023 user.notice nebula-proto: Killed interface nebula1.
Wed Sep  6 22:49:05 2023 daemon.notice netifd: Interface 'nebula1' is now down
Wed Sep  6 22:49:05 2023 daemon.notice netifd: Interface 'nebula1' is setting up now
Wed Sep  6 22:49:05 2023 user.notice nebula-proto: Tunnel device in config file (nebula1
) doesn't match interface name (nebula1)!
```

Created interface in LuCI WebUI named "nebula1", config.yml parsing is correct (verified with online validators). After restarting router, only these logs as above are shown. 
When removing nebula-proto and installing nebula-service, using the same config.yml, nebula tunnel and interface start up correctly.

stangri, 2023-09-16:
@AveoCross Can you post your yml file?

AveoCross, 2023-09-16:
> @AveoCross Can you post your yml file?

[config.zip](https://github.com/openwrt/packages/files/12641057/config.zip)

stangri, 2023-09-16:
I can't reproduce, however I'm on a later version available on 23.05 and later.

Can you try the proto file from this commit: https://github.com/stangri/source.openwrt.melmac.net/commit/3e8a0e5489d34a3ee9d298eae53954972010def2 and let me know if it fixes your issue?

It needs to go to `/lib/netifd/proto/nebula.sh`.

AveoCross, 2023-09-22:
Tried now and tested for a while and:
1. It now properly parses yml file and creates and links up nebula1 interface
2. If there is any non-working host in static_host_map in yml file, service links up and then kills the nebula1 interface for some reason (even if there are other working static hosts)
3. Errors like these 
`daemon.notice netifd: nebula1 (18565): time="2023-09-22T19:36:21Z" level=error msg="Failed to write to tun" error="input/output error"`
appear and nebula network is dead at the same time. Restarting nebula1 interface fixes the issue
4. It appears there are no statistics in "Network > Interfaces > RX/TX". They are always reporting 0B (0 Pkts.)
5.  There are logs like these
`user.notice nebula-proto: Running nebula1 from config.yml with addresses: .`
Even if interface works, i dont know if it should report IP as a dot. For a while i thought it can't set up a proper IP adress for nebula1 interface but after some time i figured it just works.

stangri, 2023-09-25:
> 2\. If there is any non-working host in static_host_map in yml file, service links up and then instantly kills the nebula1 interface for some reason (even if there are other working static hosts)

Is this not consistent with behaviour on other platforms? I believe I've had a similar experience on macOS a while back until I switched to using IP addresses rather than FQDNs for `static_host_map`.

> 3\. Errors like these
>     `daemon.notice netifd: nebula1 (18565): time="2023-09-22T19:36:21Z" level=error msg="Failed to write to tun" error="input/output error"`
>     appear and nebula network is dead at the same time. Restarting nebula1 interface fixes the issue

I'll keep checking logs every now and then, but I haven't had any problem like that. Is `nebula1` the only tun-based interface on your system?

> 4\. It appears there are no statistics in "Network > Interfaces > RX/TX". They are always reporting 0B (0 Pkts.)

I'm not sure I'll be able to overcome this with the limitations of the tun-based interfaces.

> 5.\ There are logs like these
>   `user.notice nebula-proto: Running nebula1 from config.yml with addresses: .`
>   Even if interface works, i dont know if it should report IP as a dot. For a while i [...]

I believe I have this fixed in [this commit](https://github.com/stangri/source.openwrt.melmac.net/commit/cc8160f9c1a6517138a768e2e39eab4568d95fa5). I'll send PR/merge it if you could test it.

AveoCross, 2023-09-25:
Thank for your replies.

> Is this not consistent with behaviour on other platforms? I believe I've had a similar experience on macOS a while back until I switched to using IP addresses rather than FQDNs for `static_host_map`.

I didnt used nebula on my other systems so unfortunately i dont have anything to compare.

> I'll keep checking logs every now and then, but I haven't had any problem like that. Is `nebula1` the only tun-based interface on your system?

Yes.

> I'm not sure I'll be able to overcome this with the limitations of the tun-based interfaces.

Small mostly visual issue for me but i think it worked on original package.

> I believe I have this fixed in [this commit](https://github.com/stangri/source.openwrt.melmac.net/commit/cc8160f9c1a6517138a768e2e39eab4568d95fa5). I'll send PR/merge it if you could test it.

Will give it a try soon. Thanks.

stangri, 2024-01-29:
@AveoCross any updates?

AveoCross, 2024-02-04:
> @AveoCross any updates?

No, sorry. Didn't had much time to tinker, and on top of that, our wan mesh network kinda died for now. 
When i'll have some more time for that, i will give updates.

stangri, 2024-02-04:
I'm closing this then, I couldn't reproduce.

closed 2024-02-04 by stangri