openwrt/packages #22070: ddns-scripts:  error message during installation with configuration

luizluca, 2023-09-09:
Maintainer: @maxberger
Environment: OpenWrt 23.05-rc3

Description:

If you install the package with a configuration already in place (after an upgrade), it looks like the scripts tries to do something (restart?) that generates these errors:

```
cat: can't open '/var/run/ddns/afraidv4.dat': No such file or directory
cat: can't open '/var/run/ddns/nowdnsv4.dat': No such file or directory
cat: can't open '/var/run/ddns/afraidv6.dat': No such file or directory
```

It might be harmless but it should be fixed.

maxberger, 2023-09-10:
Hi @luizluca ,

is it possible that you have nowdns or afraid.org configured on your system? 

Also, are those the only warning lines that show up, or do you see more of them?

Thanks

Max

luizluca, 2023-09-10:
Yes, I do. I actually have all these:

        option service_name 'afraid.org-v2-token'
        option service_name 'changeip.com'
        option service_name 'now-dns.com'
        option service_name 'dynv6.com'

But only those I listed gave me a warning