openwrt/packages #22071: transmission-remote: missing seccomp syscalls crashes the process, corrupted downloads

luizluca, 2023-09-10:
Maintainer: @dangowrt
Environment: 23.05.0-rc3 mediatek/filogic aarch64_cortex-a53

Description:

transmission was very unstable. It was constantly crashing, resulting in corrupted downloads. If the download speed was at 5MBp/s or more, it was crashing so frequently that the service was kept down:

```
daemon.info procd: Instance transmission::instance1 s in a crash loop 6 crashes, 18 seconds since last crash
```

I didn't find any error logs, even after increasing the transmission message_log. So, I focused on seccomp. I straced the daemon and noticed some crashes like:

```
24098 ftruncate(46, 55707)              = 46
24098 +++ killed by SIGSYS +++
```

```
[pid 25985] restart_syscall(<... resuming interrupted io_setup ...> <unfinished ...>
[pid 25985] +++ killed by SIGSYS +++
```

I also executed the process with trace and got "ftruncate" instead of "ftruncate64" present in /etc/seccomp/transmission-daemon.json (BTW, ftruncate should be after fstat,fstat64. ). I added the io_setup as well but it was still crashing very frequently (when not in trace mode). I also didn't get the io_setup during a trace. I use `transmission.@transmission[0].peer_socket_tos='lowcost'` that might have caused that.

At least the truncate did improve the stability. I'll leave it running in trace more for a couple of days to see if I get something new.

I wonder why I got no seccomp violations in log. Is there anything missing that prevents that? Does it need auditd or a more verbose dmesg?

luizluca, 2023-09-17:
I did a long trace. Only ftruncate appeared. I'll remove io_setup from my side and check if it crashes again

neheb, 2024-06-09:
This looks fixed.

commit 4a94edb0 (2023-09-17): transmission: add ftruncate syscall to seccomp

Fixes many crashes.

Fixes #22071.

(cherry picked from commit a0372545887a2f16329be56949465e13af0d04c7)
closed 2024-06-09 by neheb