openwrt/packages #22135: ddns-scripts-nsupdate: Zone support missing

LorbusChris, 2023-09-18:
Maintainer: @feckert @chris5560
Environment: any

Description:

RFC2136/nsupdate-capable servers support zoning. This e.g. allows for multi-tenant use of the DNS server, with each user only having edit-access to records in specific zones.

From https://linux.die.net/man/8/nsupdate:
> **zone zonename**
>     Specifies that all updates are to be made to the zone zonename. If no zone statement is provided, nsupdate will attempt determine the correct zone to update based on the rest of the input. 

Example command file:
```
zone myzone.example.com
update delete host.myzone.example.com. AAAA
update add host.myzone.example.com. 600 AAAA 203.0.113.0
```

Support for zones is currently missing from the ddns-scripts-nsupdate package.

brada4, 2023-09-18:
nsupdate splits off domain and does proper request to dns server.
users are distinguished by authentication keys nowadays.
If you inted to use dots in hostnames where current nsupdate would fail feel free to submit PR

LorbusChris, 2023-09-18:
Thanks, removing the zone statement does indeed work for me with current nsupdate.

I'll close this one.

RFE for keyfile support is https://github.com/openwrt/packages/issues/22136

closed 2023-09-18 by LorbusChris