openwrt/packages #22136: ddns-scripts-nsupdate: Keyfile support missing

LorbusChris, 2023-09-18:
Maintainer: @feckert @chris5560
Environment: any

Description:

RFC2136/nsupdate-capable servers support auth via keyfile in lieu of username/password.

From https://linux.die.net/man/8/nsupdate:
> **nsupdate** [ **-d** ] [ [ **-y keyname:secret** ] [ **-k keyfile** ] ] [ **-v** ] [ **filename** ] 

> nsupdate uses the -y or -k option to provide the shared secret needed to generate a TSIG record for authenticating Dynamic DNS update requests. These options are mutually exclusive. With the -k option, nsupdate reads the shared secret from the file keyfile, whose name is of the form K{name}.+157.+{random}.private. For historical reasons, the file K{name}.+157.+{random}.key must also be present.

Example `nsupdate` invocation:
```
nsupdate -k /root/myzone.example.com.private
```

Support for auth via keyfile is currently missing from the ddns-scripts-nsupdate package.

LorbusChris, 2023-09-18:
I'd be happy to work on this myself, but I'll need some guidance.

brada4, 2023-09-18:
The file is here
https://github.com/openwrt/packages/blob/master/net/ddns-scripts/files/usr/lib/ddns/update_nsupdate.sh
You can type hmac:user for username and b64 key for password? 
No keyfile possible in that scripted context.
Other manual here : https://manpages.ubuntu.com/manpages/impish/man1/knsupdate.1.html
/eg short ttl is not needed at all/

LorbusChris, 2023-09-19:
> You can type hmac:user for username and b64 key for password?

I only have a keyfile and no username, so no.

I'm aware of the script, I'm just not familiar with OpenWrt package scripting in general.

Really my question is: How do I add a new parameter to `/etc/config/ddns`?

Is any `parameter` I add in the config file automagically parsed to `$parameter` in the shell script?
 
It should be rather trivial to change the script itself to consume a `keyfile` path as a paramter:
If `$keyfile` is non-zero, ignore the `username` and `password` parameters, omit the line `key $username $password` from $DATFILE and add `-k $keyfile` to the nsupdate invocation (else behave as before)

brada4, 2023-09-19:
If you print that key file it is in common DNS RR format where domain name without dot is username and authentication key is first of base64 values in that record.

LorbusChris, 2023-09-19:
That works indeed! Thank you. Closing this as not needed.

closed 2023-09-19 by LorbusChris