openwrt/packages #22170: ffmpeg: build failure with binutils 2.41

oskarirauta, 2023-09-21:
Maintainer: @thess
Environment: x86_64, latest git

Description:

Build failed, following error repeated numerous times:

```
Error: operand type mismatch for `shr'
```

I found a patch for it from here: [https://www.linuxquestions.org/questions/slackware-14/regression-on-current-with-ffmpeg-4175727691/](https://www.linuxquestions.org/questions/slackware-14/regression-on-current-with-ffmpeg-4175727691/) file: [https://git.videolan.org/?p=ffmpeg.git;a=patch;h=effadce6c756247ea8bae32dc13bb3e6f464f0eb](https://git.videolan.org/?p=ffmpeg.git;a=patch;h=effadce6c756247ea8bae32dc13bb3e6f464f0eb)

I can verify that this patch fixes the issue.

BKPepe, 2023-09-21:
I can not see that binutils 2.41 is in the master branch. I double-checked it and in the master branch, there is 2.40. It seems that you are using the latest git also with something else.

You can see that the latest ffmpeg update was successfully compiled on all targets: https://github.com/openwrt/packages/pull/22128

In this case as you seems to be using something else then the latest git, I suggest to create pull request  with that patch or wait until a new version ffmpeg is released. Also, you can usd binutils 2.40

oskarirauta, 2023-09-22:
what? I am using plain version.
Check here: https://github.com/openwrt/openwrt/tree/main/toolchain/binutils

There's 2.41

BKPepe, 2023-09-22:
Not used as default version - https://github.com/openwrt/openwrt/blob/587f94eb1adb2eb65ff7b87a78c4641d2d15dc2c/toolchain/binutils/Config.in#L5.

thess, 2023-09-22:
@oskarirauta - Thanks for flagging this issue. Apparently binutils introduced this change in:
https://sourceware.org/git/?p=binutils-gdb.git;a=commit;h=c34d1cc9200ae24dc7572aaf77d80276c0490e9b
and, the corresponding trac item: https://trac.ffmpeg.org/ticket/10405 documents a patch for mathops.h in libavcodec.

ffmpeg (5.1 branch) fixed on 16-Jul-2023 by:
https://git.ffmpeg.org/gitweb/ffmpeg.git/commit/25cd95a9dc3510c3cc0d7aad6f9d83f6a1078c7e
not yet released but should be in 5.1.4. If you think this is important for now, then submit a PR with the upstream patch with a remark that it can be reverted at the next update to ffmpeg.

oskarirauta, 2023-09-23:
@thess

There is a reason why I made this issue instead of PR - I only verified that this fixes build with binutils-2.41, impact with older versions of binutils is un-tested by me. Surely it's important, at least for me, who constantly re-builds images to participate and provide fixes; aim always is to be able to build everything; but I am not using ffmpeg for anything at the moment. Yet, importance exists for those who do. But sure, I'll submit a PR as requested.

BKPepe, 2023-09-24:
> Sure- but so isn't 6.x branch kernel and yet preparation to develop on that branch is on focus... 

I don't dispute it, of course. Kernel 6.1 is slowly getting integrated into OpenWrt and several targets already have it as testing kernel. There's a difference. In your OP, there were no details on how to reproduce it. You did not say that you weren't used defaults. That's important to know. You just said platform and latest git even though the ffmpeg update was tested and as I pointed out, it was compile tested by CI, which is using the latest commit from each branch (my PR was against master branch, thus it was using the master branch) and buildbots do not report it.

**Usually these testing things are not meant to be for end users, but for someone who will provide as much details as possible to be able to reproduce it and as well for developers, who can fix that.**

> Whole point of this is issue is to maintain compatibility with master- even when defaults are not chosen as choosing alternating options shouldn't result in issues, or if it does; someone could attempt to fix it.

Since you already notice it with all these force push, it is not so easy to fix it instead of creating the issue and complaining that someone else did the job pretty badly, right?

___

In any case, this issue should not be closed as it is not fixed. It will be.

PR #22191 (2023-09-24): ffmpeg: fix assembling with binutils v2.41
fix addresses issue when building ffmpeg with binutils v2.41 that is documented [here](https://trac.ffmpeg.org/ticket/10405)

Fixes: #22170

Remarks:
Patch is merged to ffmpeg upstream and will be included in next release of ffmpeg, so patch applies only to current version and is removed when package is updated to next version of ffmpeg.

Upstream patch refreshed.

Maintainer: @thess
Compile tested: x86_64, latest git
closed 2023-09-24 by thess