openwrt/packages #22192: new `luajit2` package Lead to Recursive loop!!!

zxlhhyccc, 2023-09-23:
````
tmp/.config-package.in:106579:error: recursive dependency detected!
tmp/.config-package.in:106579:	symbol PACKAGE_dnsdist depends on PACKAGE_dnsdist
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:106611:error: recursive dependency detected!
tmp/.config-package.in:106611:	symbol PACKAGE_dnsdist-full depends on PACKAGE_dnsdist-full
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:105249:error: recursive dependency detected!
tmp/.config-package.in:105249:	symbol PACKAGE_snort depends on PACKAGE_snort
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:105280:error: recursive dependency detected!
tmp/.config-package.in:105280:	symbol PACKAGE_snort3 depends on PACKAGE_snort3
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:107124:error: recursive dependency detected!
tmp/.config-package.in:107124:	symbol PACKAGE_knot-resolver depends on PACKAGE_knot-resolver
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:105113:error: recursive dependency detected!
tmp/.config-package.in:105113:	symbol PACKAGE_openappid depends on PACKAGE_openappid
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"
``````

BKPepe, 2023-09-23:
You dont need to spam it everywhere. You posted it already on three places:
1. Pull request, where it was merged
2. Commit, which was merged
and now here.

BKPepe, 2023-09-23:
Next time, please fill the proper issue template.

zxlhhyccc, 2023-09-23:
> Next time, please fill the proper issue template.

OK!

ptpt52, 2023-09-24:
just to say, same issue for my build.

Ansuel, 2023-09-25:
Fixed

closed 2023-09-25 by Ansuel