openwrt/packages #22202: unbound: exclude local networks from private-address rebind_protection configuration

brianjmurrell, 2023-09-24:
Maintainer: @EricLuehrsen
Environment: 22.03.5 r20134-5f15225c1e

### Description:

When using `private-address` configuration items to implement `rebind_protection` the local network ought to be excluded in at least some circumstances.  One use-case I can think of for example is that if the unbound server is on an RFC-1918 network and has a forwarding zone to a server on that network, it would be reasonable to expect that queries to that server would return valid responses with RFC-1918 addresses in them.

There are perhaps other conditions that ought to be considered for this exception.

Having to completely disable `rebind_protection` for such use-cases is a pretty big hammer and opens up attack vectors that are much bigger than they need to be.

brianjmurrell, 2023-09-24:
If it could be relied-upon that such networks would properly configure the DNS search domains for their network in their DHCP (and/or SLAAC) responses, `private-domain` could be used to resolve this issue, however I suspect more of such networks don't properly set their DNS search domains than do.  Sadly since `private-domain` would be such an elegant solution.

brada4, 2023-09-25:
It is default upstream behaviour. You can make 8-16 complements to full 1918 network and make it work like YOU intend.

brada4, 2023-09-25:
Disable default private box and make complement networks....

brianjmurrell, 2023-09-25:
> Disable default private box and make complement networks....

I'm afraid I am not parsing what you are trying to say.  What is "private box" and how do you propose I make complement networks?  What kind of `/etc/config` changes are you proposing for example?

Let say my router is on a 192.168.10.0/24 network.  What do you propose the relevant `/etc/config` changes are for that, including having rebind protection?

I am not at all being lazy here and wanting you to do the work for me.  I am quite skeptical of your proposed solution as I cannot see how it can be done.

brada4, 2023-09-25:
`option rebind_protection '0'`

Then add private-address: statements to unbound-ext.conf

brianjmurrell, 2023-09-25:
Thanks for your effort @brada4 but your workaround is **exactly** what this ticket is NOT about.  What you are suggesting is a non-scaling solution that every person with this use-case should not have to re-invent for themselves.

`option rebind_protection` exists.  It just has use-cases that it does not cover.  This ticket was not to solicit workarounds but to suggest/request an enhancement to the rebind protection implementation to cover implementation gaps.

brada4, 2023-09-26:
You can override tebind_protection for a list of domain names for individual forward zones?
If you want some blacklist to point to 1918 lan you have to re-implement rebind protection to exclude that netblock.It is exactly the case rebind protection is ought to protect you from otherwise.

brianjmurrell, 2023-09-26:
> You can override tebind_protection for a list of domain names for individual forward zones?

Yes, and [as I said](https://github.com/openwrt/packages/issues/22202#issuecomment-1732647883), if one could rely on such networks populating their DHCP replies with a search list of their intranet domains, this could be a solution.  I suspect however that the success rate of such intranets populating their search domains is low.

This all has to be _automatic_ as the use-case for all of this is for a travel router which will be plugged into different network after different network as one travels from hotel to hotel.  Having to spend a half hour on every network you plug it into figuring out what their intranet domains are and what their RFC-1918 use is and then re-configuring unbound for those things is simply not an acceptable solution.

brada4, 2023-09-26:
It is private-domain setting 
```
private-domain: "lan"
private-domain: "1.168.192.in-addr.arpa"
```
Ps certainly missing from [Documentation](https://github.com/openwrt/packages/blob/master/net/unbound/files/README.md)

brianjmurrell, 2023-09-26:
Hrm.  This might be more complicated than I had originally considered.

Just so it's clear what the problem is, let's consider such an intranet without the complication of unbound first.

You connect your device (i.e. phone, tablet, computer, etc.) to a hotel network.  It does a DHCP request and gets an IP address in an RFC-1918 network along with a DNS server address and no DNS search list.  You then open a web-page to (say) www.google.com (but it can be any webpage).  That causes your phone to ask the hotel DHCP provided DNS server for an address for www.google.com.  That DNS server, because it wants you to go through a captive portal to enter authentication details and/or agree to terms and conditions, etc. gives you an (RFC-1918) address on it's local network because it wants you to load their captive portal website instead of www.google.com.

Now substitute a router running unbound for the above device.  It is a wifi client to the hotel network and wifi AP to it's own clients (your phone, laptop, etc.) using NAT.  It requests an IP address and gets all of the same information as above.  Unbound is configured to forward all queries to the hotel DHCP provided DNS server because typically it cannot access **anything** outside that intranet until the captive portal has been satisfied.

This is where the _one size fits all_ rebind protection of OpenWRT breaks.  With your device connected to the OpenWRT router running unbound and OpenWRT's rebind protection enabled, your device doesn't get any answer for www.google.com (or anything else) from unbound on the router because all of the answers are the same intranet captive portal address which the rebind protections `private-address` settings are causing to be dropped.

My original suggestion was that this is where rebind protection needs to have an exception for a query being made to a forwarded DNS server that is in the same RFC-1918 network that the router is on returning responses in the same RFC-1918 network.

But the problem with that is that unbound doesn't know if the answer it is receiving is for a query of something that is legitimately inside the intranet or is a answer for something external that the intranet DNS server is just forwarding back to unbound, and subject to the rebinding attack.

Ultimately, it seems it is impossible to prevent rebinding attacks in such a scenario.

I suppose the solution here is to switch unbound to being a full resolver and making external queries directly (bypassing the intranet DNS server) with rebind protection enabled once the captive portal has been satisfied.  This has two problems.

1. How does OpenWRT know when the captive portal has been satisfied so that it can update unbound's configuration to stop forwarding DNS requests and start resolving them itself.
2. How does OpenWRT know when the access granted by satisfying the captive portal is revoked and the captive portal needs to be satisfied again, and switching unbound back to forwarding to the intranet DNS server.

I suppose I could also question the value (to me at least) of rebinding protection on a network (i.e. the hotel intranet) where no devices are my own.  All of my devices are on my own network behind the travelrouter, where I could implement `private-address` for my own network behind the OpenWRT router).  It's a selfish perspective to be sure, but on the other hand, it's the hotel DNS's responsibility to provide that protection for their users.

brada4, 2023-09-26:
So you need to set rebind protection for static interface in LAN, and permit lan dns zone through it imitating dnsmasq.
private-address: 192.168.1.0/24
private-domain: "lan"

In place of total rebind protection checkbox.

closed 2023-09-26 by brianjmurrell

EricLuehrsen, 2023-10-14:
Sorry, other priorities lately. This parallels dnsmasq rebind protection when I implemented a few years ago. Repeating above, the problem with trying to do this in a travel router is the walled garden at a hotel. Each floor in your hotel maybe isolated  networks to limit issues for guests. (maybe 172.16.0.0...172.31.0.0). This holds the private address of your router. But the walled garden doesn't live there. The walled garden is on hotel business operation network and path to the edge/ISP. (maybe different 172.16.0.0/16 or 10.0.0.0/8). How would you automatically figure that out? Every website goes to 10.10.8.101 until you agree to T&C, but your router has 172.18.58.93?

EricLuehrsen, 2023-10-14:
If the hotel network isn't virtually chopped up something like that, then a guest can find the network printer in the managers office and cause trouble.

brianjmurrell, 2023-10-19:
@EricLuehrsen Thanks for your response here.  My ultimate goal was to be a good netizen to my "neighbors" in the hotel and make sure that my hosts could be used as a rebind attack vector against any of them.  Through ongoing analysis, it's pretty clear that that is a potentially unachievable goal for the reasons you suggest as well as others.  Sadly.

In the end I am going to have to be happy with just making sure the devices on my side of the travel router are protected and hope for the best (but certainly not hold my breath) that the hotel is doing it's job of adding rebind protection to it's network.