openwrt/packages #22209: Unbound 1.18.0-1 apk broken on x86/64

xxxx81, 2023-09-25:
The unbound app packages for version 1.18-1 on x86/64 are buggy and contain invalid symbols which causes the unbound daemon to crash. It seems to be the gcc error that debug symbols are added to the build.

BKPepe, 2023-09-25:
In this repository, we do have a proper issue template, which you decided to remove. We would like to help you more, but without providing any logs or outputs, it is hard to know what is going on. 

Did you compile the package yourself?

xxxx81, 2023-09-25:
I can't say much about it because the log spits out nothing except that the unbound service crashed. I installed/updated it normally via opkg for the RC3 but opkg brought some symbol errors after that my dns was paralyzed. The symbol errors I already know that is a GCC compiler problem is not the first time that unbound has the problem last time but it was quickly fixed with new built packages.

brada4, 2023-09-26:
You need to describe your environment more closely. [Unbound package you implicate](https://downloads.openwrt.org/releases/23.05.0-rc3/packages/x86_64/packages/unbound-daemon_1.18.0-1_x86_64.ipk) does not exhibit any problems for me.

xxxx81, 2023-09-26:
Ok here the the opkg output:

`Error relocating /usr/sbin/unbound: extended_error_encode: symbol not found
Error relocating /usr/sbin/unbound: timeval_divide: symbol not found
Error relocating /usr/sbin/unbound: get_slabhash_stats: symbol not found
Error relocating /usr/sbin/unbound: timeval_subtract: symbol not found
Error relocating /usr/sbin/unbound: extended_error_encode: symbol not found
Error relocating /usr/sbin/unbound: timeval_divide: symbol not found
Error relocating /usr/sbin/unbound: get_slabhash_stats: symbol not found
Error relocating /usr/sbin/unbound: timeval_subtract: symbol not found
Collected errors:
 * resolve_conffiles: Existing conffile /etc/config/unbound is different from the conffile in the new package. The new conffile will be placed at /etc/config/unbound-opkg.`

Here the output if I would start from command line:

`Error relocating /usr/sbin/unbound: extended_error_encode: symbol not found
Error relocating /usr/sbin/unbound: timeval_divide: symbol not found
Error relocating /usr/sbin/unbound: get_slabhash_stats: symbol not found
Error relocating /usr/sbin/unbound: timeval_subtract: symbol not found
`

brada4, 2023-09-26:
Likely you need to upgrade existing libunbound package to be same version as unbound

brada4, 2023-09-27:
The dependency should be enforced with strict version like kmod-s do? (it is actually any 1.18.x being compatible. like 1.17 used to be so nobody needed the libunbound upgrade)
??? @EricLuehrsen 

Try to search forums for wifi issue. If you are convinced about the defect even better if you know approximate solution, please report to ../openwrt as all wpad is core packages.

xxxx81, 2023-09-27:
@brada unfortunately this doesn't seem to work where I had noted it on Sunday, had updated the update script what I use libunbound but not the daemon (for whatever reason) the result was the same so the 1.17 daemon is not compatible with the 1.8 libunbound and vice versa.

With the wifi problem it gets very confusing the fix seems to be around for a while https://github.com/openwrt/openwrt/commit/ed0ad7759c6ff823f3d43c5189cf6c2d59529244 why it didn't arrive with the new packages is a mystery to me, well it could be because it's a kernel package but the hostapd update for rc3 should have been held back. One can then only hope that it does not happen with the stable series. I have now done it manually because it ran.

brada4, 2023-09-27:
Yes, different version unbound components are not compatible, if maintainer can fix that in an elegant way - he is already called in.
unbound has no relation to wpad. As seen in your commit link - wpad got fixed in different repo.

xxxx81, 2023-09-27:
Ok if the maintainer already knows I can close the bug report especially since the problem is different than I thought. 
I only added the Wpad problem because I didn't want to open a new bug report. But since it is probably gone until the release of the final 23.05 should not be a problem otherwise I will create there again a bug report if no one else does it. Thanks for the help.

brada4, 2023-09-27:
Please re-open obviously... I tagged the maintainer here: https://github.com/openwrt/packages/issues/22209#issuecomment-1736888452 .

xxxx81, 2023-09-27:
Ok I reopen the issue.

BKPepe, 2023-09-27:
No, no, no, guys. This has been wrong since the beginning. 

This is, let's say, behavior, which could probably be narrowed down, but it won't be in the near future.
- There are threads that should discourage you from using ``opkg upgrade``. 
References:

  - https://forum.openwrt.org/t/opkg-upgrade-package-upgrade-warning/63370
  - https://forum.openwrt.org/t/opkg-upgrade-good-or-bad-idea/2342/2?u=pepe

What can be done, but it brings other issues and overall, I don't like it is something that is done in Knot DNS package, what we do have here. 

You should use sysupgrade or you need to know what you are doing, because you are root and somehow you are responsible for things, which you are doing otherwise, you can shoot your own leg.

At least, I know what Turris was trying to solve with Updater-ng and it works there. This issue is somehow inside opkg.

Because of this, I am closing this.

closed 2023-09-27 by BKPepe

xxxx81, 2023-09-27:
@BKPepe There would be the possibility to merge the unbound daemon and libunbound into one package so the opkg bug would not be a problem anymore. Or is there a reason why they are separated?

EricLuehrsen, 2023-10-14:
Sorry, other priorities lately. An integrated unbound release is not recommended. libunbound is used without daemon for a few packages. Debian already has some problems from a situation where Unbound is compiled through-and-through with libunbound integrated, and  libunbound is provided uniquely.