openwrt/packages #22223: compilation error on latest nginx ssl

zcy85611, 2023-09-26:
compilation error on nginx after recent commits, please help to look into this, thanks

```
make[3]: Entering directory '/home/openwrt/feeds/packages/net/nginx'
rm -f /home/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.2/.built
touch /home/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.2/.built_check
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/home/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.2=nginx-1.25.2 -mips16 -minterlink-mips16 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -DNGX_LUA_NO_BY_LUA_BLOCK  -I/home/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/usr/include -I/home/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/include/fortify -I/home/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/include   " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/home/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.2=nginx-1.25.2 -mips16 -minterlink-mips16 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -DNGX_LUA_NO_BY_LUA_BLOCK -mips16 -minterlink-mips16 -ffunction-sections -fdata-sections  -I/home/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/usr/include -I/home/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/include/fortify -I/home/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/include   " LDFLAGS="-L/home/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/usr/lib -L/home/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/lib -fuse-ld=bfd -Wl,--gc-sections -znow -zrelro   " make  -C /home/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.2/. AR="mipsel-openwrt-linux-musl-gcc-ar" AS="mipsel-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/home/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.2=nginx-1.25.2 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNGX_LUA_NO_BY_LUA_BLOCK" LD="mipsel-openwrt-linux-musl-ld.bfd" NM="mipsel-openwrt-linux-musl-gcc-nm" CC="mipsel-openwrt-linux-musl-gcc" GCC="mipsel-openwrt-linux-musl-gcc" CXX="mipsel-openwrt-linux-musl-g++" RANLIB="mipsel-openwrt-linux-musl-gcc-ranlib" STRIP=mipsel-openwrt-linux-musl-strip OBJCOPY=mipsel-openwrt-linux-musl-objcopy OBJDUMP=mipsel-openwrt-linux-musl-objdump SIZE=mipsel-openwrt-linux-musl-size CROSS="mipsel-openwrt-linux-musl-" ARCH="mipsel" ;
make[4]: Entering directory '/home/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.2'
make -f objs/Makefile
make[5]: Entering directory '/home/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.2'
mipsel-openwrt-linux-musl-gcc -c -fPIC -I/home/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/luajit-* -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/home/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.2=nginx-1.25.2 -mips16 -minterlink-mips16 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -DNGX_LUA_NO_BY_LUA_BLOCK  -I/home/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/usr/include -I/home/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/include/fortify -I/home/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/include   -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/home/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.2=nginx-1.25.2 -mips16 -minterlink-mips16 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -DNGX_LUA_NO_BY_LUA_BLOCK -DNDK_SET_VAR -I src/core -I src/event -I src/event/modules -I src/event/quic -I src/os/unix -I /home/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/luajit-2.1 -I /home/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.2/nginx-mod-lua/src/api -I /home/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/libxml2 -I /home/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.2/nginx-mod-ubus/src -I objs -I src/http -I src/http/modules -I src/http/v2 \
	-o objs/addon/src/ngx_http_lua_script.o \
	/home/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.2/nginx-mod-lua/src/ngx_http_lua_script.c
cc1: note: someone does not honour COPTS correctly, passed 2 times
In file included from /home/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.2/nginx-mod-lua/src/ngx_http_lua_script.h:11,
                 from /home/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.2/nginx-mod-lua/src/ngx_http_lua_script.c:13:
/home/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.2/nginx-mod-lua/src/ngx_http_lua_common.h:58:10: fatal error: pcre.h: No such file or directory
   58 | #include <pcre.h>
      |          ^~~~~~~~
compilation terminated.
```

brada4, 2023-09-27:
It was upgraded to pcre2 here:
https://github.com/openwrt/packages/commit/f0754531c4d85ac301c95d7dc6c83acdad6b24e7
Please update your source tree completely in sync with SDK.

zcy85611, 2023-09-27:
I'm using github action, every time it compiles, it downloads the latest source code, I just tried again, still no luck.

it seems to be the problem

/home/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.2/nginx-mod-lua/src/ngx_http_lua_common.h:58:10: fatal error: pcre.h: No such file or directory
   58 | #include <pcre.h>
      |          ^~~~~~~~

BKPepe, 2023-09-27:
You need to provide more details about your environment. There is a lack of details and because of that, it is very tempting to close this issue. We don't have a crystal ball to know exactly what is going on. There is a reason why we have an issue template, which was removed.

Most likely, you are missing the host package : https://github.com/openwrt/openwrt/blob/0c43a48735959245f18d79a6e908d3b45cff0a94/package/libs/pcre2/Makefile

Anyway, we are using Github Actions, too, and it is working for us: https://github.com/openwrt/packages/pull/21018

zcy85611, 2023-09-28:
> You need to provide more details about your environment. There is a lack of details and because of that, it is very tempting to close this issue. We don't have a crystal ball to know exactly what is [...]

I'm using Ubuntu 20.02，compiled for MT7621, Newifi-D2 and  installed followings.

I also tried replacing libpcre3-dev to libpcre2-dev, it doesn't make any difference.
```
sudo -E apt-get -qq install build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git \
                              libncurses5-dev libssl-dev python3-distutils rsync unzip file wget zlib1g-dev \
                              libpcre3-dev libfuse-dev 
```

in the log, it shows pcre2 has been compiled

 ```
 make[3] -C feeds/packages/net/acme-common compile
 make[3] -C feeds/packages/net/socat compile
 make[3] -C package/libs/openssl compile
 make[3] -C package/libs/pcre2 compile
 make[3] -C package/system/ca-certificates compile
 make[3] -C feeds/packages/libs/libuv compile
```

```
make[3]: Entering directory '/home/runner/work/Auto2/Auto2/openwrt/package/libs/pcre2'
mkdir -p /home/runner/work/Auto2/Auto2/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/runner/work/Auto2/Auto2/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/Auto2/Auto2/openwrt/build_dir/target-mipsel_24kc_musl/pcre2-10.42/.pkgdir/libpcre2/. /home/runner/work/Auto2/Auto2/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/runner/work/Auto2/Auto2/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.libpcre2_installed
echo "libpcre2" >> /home/runner/work/Auto2/Auto2/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/pcre2.default.install
make[3]: Leaving directory '/home/runner/work/Auto2/Auto2/openwrt/package/libs/pcre2'
time: package/libs/pcre2/compile#0.22#0.09#0.29
```

brada4, 2023-09-28:
Why you do not show REPO_URL and REPO_BRANCH variables entering your script?
What in the world is Ubuntu 20.02 compiled for Newifi-D2?

zcy85611, 2023-09-28:
> Why you do not show REPO_URL and REPO_BRANCH variables entering your script? What in the world is Ubuntu 20.02 compiled for Newifi-D2?

OS：Ubuntu 20.02
Router Model: mt7621_d-team_newifi-d2

REPO_URL and REPO_BRANCH 
```
env:
  REPO_URL: https://github.com/openwrt/openwrt
  REPO_BRANCH: master
```

brada4, 2023-09-28:
Such ubuntu version was never released.
Plese install ubuntu in a virtual machine, do the apt activities then in a new user account do the build. 
Hope that helps you to fix your container.

closed 2023-09-28 by zcy85611