openwrt/packages #22227: travelmate: RFE to have defaults for auto-added STAs

brianjmurrell, 2023-09-27:
Maintainer: @didbot
Environment: 22.03.5

Description:
I believe this is an RFE as I have found no way to achieve my desired result.

What I would like is to be able to set default options for auto-added open STAs.

For example, I would like all auto-added open STAs to have `vpnservice` and `vpniface` to be set to some preset/default values so that I don't have to go configure those manually on every new auto-added open network.

brianjmurrell, 2023-09-27:
For anyone that stops by here, a Q&D hack around this is to add the following to a file in `/etc/hotplug.d/iface/`:

```bash
#!/bin/bash
# Note that this doesn't actually run with bash as it's called with /bin/sh

if ! command -v wg || [ "$INTERFACE" != "trm_wwan" ]; then
    exit 0
fi

# bring up wireguard on trm_wwan up
if [ "$INTERFACE" = "trm_wwan" ]; then
    case $ACTION in
        ifup) ifup wg0
    ;;
        ifdown) ifdown wg0
    ;;
    esac
    exit 0
fi
```

brada4, 2023-09-28:
You forgot to mention the result you want to achieve?

brianjmurrell, 2023-09-28:
Given everything I have written here, including both previous messages, I think it's pretty obvious.  I want a/my VPN service to start on connection of any/every auto-added STA without having to go configure each auto-added STA with the VPN parameters.

brada4, 2023-09-28:
You could `wg syncconf` without tearing down wg0 and all tcp crossing it.. That is clearly missing in scripts here: https://github.com/openwrt/openwrt/tree/main/package/network/utils/wireguard-tools/files

brada4, 2023-09-28:
I think it is more of enhancing those scripts to do better with all multiwan ups downs and failovers... But I agree physical interface up/down is the right trigger for btinging wg0 in order.

commit 6db87db0 (2023-10-30): travelmate: release 2.1.1

* various vpn/wireguard improvements & fixes
* improved compatibility with new netifd
* added open STA improvements by @brianjmurrell
* closes #22227 #22288 #22357

(cherry picked from commit 81658c58236a050b22dbf08309b637576db1a5c9)