openwrt/packages #22231: dnsmasq-full_2.89-6 not work

ProphSerg, 2023-09-27:
Maintainer: 
Environment: 22.03.5 mipsel_24kc
Description:

Add package dnsmasq-full_2.89-6_mipsel_24kc.ipk in ImageBuilder (openwrt-imagebuilder-22.03.5-ramips-mt7621.Linux-x86_64)
When build image have error:
```
Finalizing root filesystem...
..../openwrt-imagebuilder-22.03.5-ramips-mt7621.Linux-x86_64/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/dnsmasq: line 4: /lib/functions.sh: No such file or directory
Enabling adblock-lean
Enabling boot
Enabling bootcount
Enabling cron
Enabling ddns
Enabling dnscrypt-proxy
./etc/init.d/dnsmasq: line 4: /lib/functions.sh: No such file or directory
```

And when restart service
```
root@router:~# /etc/init.d/dnsmasq restart
udhcpc: started, v1.35.0
udhcpc: broadcasting discover
udhcpc: no lease, failing
/etc/rc.common: eval: line 587: ipcalc: not found
```

brada4, 2023-09-28:
You should report your confusion with base packages  to ../openwrt with complete details asked over there.

dnsmasq version on 22.03 is 2.86-16 which likely means you need to sync your package feeds before trying again.

REF: https://github.com/openwrt/openwrt/blob/0a1dc007e49cf20a961de52c36dbf35dd7ef74dd/package/network/services/dnsmasq/Makefile#L11

closed 2023-09-28 by BKPepe

bahner, 2023-09-28:
AOL. MIne just "died" and won't start again. I hadn't changed anything in weeks, but I run upgrade automatically each night."

bahner, 2023-09-28:
FYI: I hadn't upgraded my firmware in a while. After I did things started working again.

risa2000, 2023-10-14:
@BKPepe
I am seeing the exactly same error when doing a complete build of ImageBuilder from OpenWrt master branch, `script/feeds` updated feeds, and then building the image from the image builder.
It seems that the current `dnsmasq` package `Makefile` is at `2.89` version. This has been changed in January, but I do not remember seeing this error before, until now (I did rebuilds in June and May).
Where should I report this?

brada4, 2023-10-14:
Certainly not in old thread

risa2000, 2023-10-14:
@brada4
Well, the reason why I was asking where to report it, was exactly this - I did not want to report it in the old thread. So you were telling me, what I already know, and was totally useless.
Since you were the one who suggested to the OP where to report his issue, maybe you actually could know if mine is in package or in openwrt?

brada4, 2023-10-14:
Read this COMPLETELY and try to keep rude comments on your side of keyboard.
https://openwrt.org/bugs